arendst/Tasmota #5773: Questions/issues about Shelly 2.5

calcipher49, 2019-05-09:
Hi,

I 'am testing Shelly 2.5 with last Tasmota dev firmware 6.5.0.9 (cfd48f3)
 As Shelly 2.5 is not in device list, I use this template talken from #5592 :
{"NAME":"Shelly 2.5","GPIO":[52,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}

Power monitoring is ok but temperature doesn't appear.

What is the correct setup/template to show internal temperature?

Also, both internal entries (SW1 and SW2) as configured as SwitchXn. Is it possible to configure them as (monostable) buttons instead?
I did some tests but it doesn't work.

Thanks for your assistance.

Jason2866, 2019-05-09:
You have to use https://github.com/arendst/Sonoff-Tasmota/tree/development-adc0
It is not in develop path because it isnt tested a lot.

calcipher49, 2019-05-09:
Thanks for answer Jason.
Temperature is ok with firmware 6.5.0.10 from your link and tamplate above.

However, I still can not configure SW1 and SW2 entries as buttons (button1 is ok with small pushbutton gpio2).
If someone can help

calcipher49, 2019-05-09:
Ok found it:
- In template set GPIO2 to none, then GPIO13 to Button1n ang GPIO5 to Button2n
- Then Setoption62 1

closed 2019-05-09 by calcipher49

curzon01, 2019-05-10:
For correct Shelly 2.5 template see https://blakadder.github.io/templates/shelly25.html