arendst/Tasmota #5777: OBI socket v2

nagyrobi, 2019-05-10:
Hello!
Using two OBI Sockets with 6.5.0. 
With the original factory firmware the socket used to lit up one of the LEDs (there are two, a green and a red one) while the socket was ON. Now with Tasmota the greed LED flashes as it's supposed to, but the red LED does nothing. It would be nice to have a visual confirmation as it was originally, that the socket outputs power - and the red LED would be just perfect for that.
Any way to accomplish this?

Jason2866, 2019-05-10:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

nagyrobi, 2019-05-10:
Addressed there too, but unfortunately no success.

In the template code,  GPIO_LED2,        // GPIO13 Red LED - Power status  - is definitely wrong.

Tried all the GPIOs according to the [procedure](https://github.com/arendst/Sonoff-Tasmota/wiki/Configuration-Procedure-for-New-Devices) but none of them trigger the red LED.

meingraham, 2019-05-10:
User (255) means "not used" in essence. Ignore it.

I couple of things at play...

1. Instead of "wrong", I would say "different". It is not uncommon with these devices to have what appears to be exactly the same device on the exterior, same marking, same model, everything, yet internally a change to the electronics. So, a module or template that work before can easily not apply to a newer device.

2. The [procedure](https://github.com/arendst/Sonoff-Tasmota/wiki/Configuration-Procedure-for-New-Devices) can take a few times in order to master. Before giving up, put it aside and come back to it with a new mindset and give it another go... or tow... or three. At some point, if the previous firmware was able to control the LEDs, it's likely, for this "simple" smart device, to get it to work... with the proper configuration. The fact that selecting a different module selection makes a different LED light up is a good indication that your GPIO configuration attempts need more refinement ;-)

closed 2019-05-10 by ascillato2