arendst/Tasmota #5780: Add a colorwheel to the user interface

Def3nder, 2019-05-10:
**Have you look for this feature in other issues and in the wiki?**
I did not find this feature mentioned anywhere in the wiki.

**Is your feature request related to a problem? Please describe.**
Using Tasmota as new firmware for existing devices is great.
In the last year there are more and more devices with integrated RGB leds and there are devices that controll RGB led stripes.

Using Tasmota for these devices does result in only having a brightness slider and an On/Off switch, but you cannot control the color itself.

**Describe the solution you'd like**
It would be great to have a color wheel included in the user interface when PWMs are assigned to GPIO ports.
There is a great open source library for this (MIT License):  
`https://github.com/fujaru/jquery-wheelcolorpicker`
[image]

**Describe alternatives you've considered**
Alternatives are _ESPurna_ (that has included just this color wheel in the UI: https://github.com/xoseperez/espurna), but I would like to stay with Tasmota as I do like the MQTT imlementation a lot more and the console is a really great feature only in Tasmota, too.

**Additional context**
_Add any other context or screenshots about the feature request here._
Here is a dome of the features of the color wheel: http://files.jar2.net/jquery/wheelcolorpicker/example-v3/example.html

ascillato2, 2019-05-10:
Hi, 

Sorry, Tasmota is mainly mqtt. This is not going to be added due to the flash and RAM are limited in the esp8266, so the remaining space are reserved for features and sensors support. 

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-10 by ascillato2

Def3nder, 2019-05-10:
okay - thx for the answer