arendst/Tasmota #5781: Visual Studio Code Settings for Shelly

lobocobra, 2019-05-11:
Hi all

I am moving to Visual Studio code from Arduino and followed the wiki page to complie the arendst firmware. 

My issue:
Shelly has 2mb (which I can easily switch to in Arduino), but for Visual-Studio-Code I need to change the config in platform.ini.
=> I assume that I have to switch this value: 
`[common]
framework                 = arduino
board                     = esp01_1m
board_build.flash_mode    = dout`

But to what I do neet to switch esp01_1m? 

Many thanks for a short hint. I found nothing in google and wiki.

Jason2866, 2019-05-11:
Leave everything unchanged in platformio.ini  Shelly works fine with the standard settings

SargonofAssyria, 2019-05-11:
I have a OBI IP44 with a esp-wroom-2M also 2mb.
It would be nice to have a compile setting for 1.5 mb Sketch and 512kb spiff, so OTA would be easy.
Is that possible?

ascillato2, 2019-05-11:
For backwards compatibility with Tasmota users, the precompiled bins will continue be with the standard memory map.
If you have a different memory map in your devices, you need to full erase with esptool.py as explained in the wiki and flash it again. 

If you need further assistant, please, address this to the Tasmota support chat. Thanks

closed 2019-05-11 by ascillato2

SargonofAssyria, 2019-05-13:
Found my solution: https://groups.google.com/forum/#!topic/sonoffusers/wCquqDdfAnA