arendst/Tasmota #5782: Rule Trigger and command

uviniciusc, 2019-05-11:
Hi everyone. 

It's possible to trigger do 2 or more comand's?

Like this: 
Rule1 on time#minute=507 do var1 55 var2 40 var3 10 var4 05 var5 FF endon

My result is {"Var1":"55 var2 4"}

Somebody could help me?
I dont no if it's really possible.
I tried for 2 days and nothing.

Thank you so much.

calcipher49, 2019-05-11:
Hi,

Use backlog:
Rule1 on time#minute=507 do backlog var1 55 ; var2 40 ; var3 10 ; var4 05 ; var5 FF endon

uviniciusc, 2019-05-11:
It's amazing. 
Works fine.
Thank you so much. Now my weekend begin, i was prepared to stay a lot of hours on the chair but now you saved my saturday.

Sharing my scenery:
I have one rule that every minute read this 5 var and set color. Now i can set value of this var's in other rule to 5 variable at the same time in one especific minute with only one command, it's really nice, if i dont need to change all var's this rulle works too, is only necessary keep out the var that i dont wanna change. 

One more thank you so much calcipher49 and GitHuber's
Bye 
It was really one fast answer.

closed 2019-05-11 by uviniciusc