arendst/Tasmota #5786: Sensor34 6 command not working as intended?

thucar, 2019-05-12:
### BUG DESCRIPTION
It seems to me that the Sensor34 6 command is not functioning at all, or is not functioning as expected. If I issue a sensor34 6 xxxx command, then the value red by the sensor never changes.
For instance:
I have my scale calibrated and it has a known weight of 5kg sitting on it. It shows 5kg. I issue a Sensor34 6 xxxx command to change WeightItem value. The sensor still displays 5kg

### REQUESTED INFORMATION
- [*] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [*] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [*] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [*] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [*] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [*] Device used (i.e. Sonoff Basic) : Sonoff TH16
- [*] Tasmota binary firmware version number used : 6.5.0.7 / self-compiled
- [*] Development IDE - Compiler / Upload tools used : Arduino IDE
- [*] Provide the output of command ``status 0`` :
```
08:33:01 MQT: stat/sonoffk/STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff"],"Topic":"sonoffk","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:33:01 MQT: stat/sonoffk/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-minimal.bin","RestartReason":"Exception","Uptime":"0T01:14:51","StartupUTC":"2019-05-12T04:18:10","Sleep":50,"CfgHolder":4617,"BootCount":33,"SaveCount":91,"SaveAddress":"FB000"}}
08:33:01 MQT: stat/sonoffk/STATUS2 = {"StatusFWR":{"Version":"6.5.0.7(sonoff)","BuildDateTime":"2019-05-10T14:03:14","Boot":6,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
08:33:01 MQT: stat/sonoffk/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Indome Office","Kristalli"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000500000000000000000000000000","00000000"]}}
08:33:01 MQT: stat/sonoffk/STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":504,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0F082384","000003A0","211617C6","000008C0"]}}
08:33:01 MQT: stat/sonoffk/STATUS5 = {"StatusNET":{"Hostname":"sonoffk-3870","IPAddress":"192.168.1.117","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"194.204.0.208","Mac":"EC:FA:BC:86:CF:1E","Webserver":2,"WifiConfig":4}}
08:33:01 MQT: stat/sonoffk/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.137","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_86CF1E","MqttUser":"sonoffk","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:33:01 MQT: stat/sonoffk/STATUS7 = {"StatusTIM":{"UTC":"Sun May 12 05:33:01 2019","Local":"Sun May 12 08:33:01 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+03:00","Sunrise":"07:12","Sunset":"22:20"}}
08:33:01 MQT: stat/sonoffk/STATUS10 = {"StatusSNS":{"Time":"2019-05-12T08:33:01","HX711":{"Weight":5}}}
08:33:01 MQT: stat/sonoffk/STATUS11 = {"StatusSTS":{"Time":"2019-05-12T08:33:01","Uptime":"0T01:14:51","Vcc":3.468,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"POWER":"OFF","Wifi":{"AP":2,"SSId":"Kristalli","BSSId":"00:14:D1:B1:06:00","Channel":10,"RSSI":72,"LinkCount":2,"Downtime":"0T00:00:09"}}}
```
- [*] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
07:39:48 MQT: tele/sonoffk/SENSOR = {"Time":"2019-05-12T07:39:48","HX711":{"Weight":5,"Count":56}}
07:40:00 CMD: sensor34 6
07:40:00 RSL: Received Topic /sensor34, Data Size 1, Data 6
07:40:00 RSL: Group 0, Index 34, Command SENSOR, Data 6
07:40:00 MQT: stat/sonoffk/RESULT = {"Sensor34":{"WeightRef":5000,"WeightCal":21,"WeightMax":100000,"WeightItem":90.0}}
07:40:42 CMD: sensor34 6 0
07:40:42 RSL: Received Topic /sensor34, Data Size 3, Data 6 0
07:40:42 RSL: Group 0, Index 34, Command SENSOR, Data 6 0
07:40:42 MQT: stat/sonoffk/RESULT = {"Sensor34":{"WeightRef":5000,"WeightCal":21,"WeightMax":100000,"WeightItem":0.0}}
07:40:43 CFG: Saved to flash at F4, Count 90, Bytes 3584
07:45:08 MQT: tele/sonoffk/SENSOR = {"Time":"2019-05-12T07:45:08","HX711":{"Weight":5}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Attach HX711, calibrate, put a weight on the scale. send sensor34 6 50

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The scale should display the weight sent by command sensor34 6 xxxx

arendst, 2019-05-13:
The goal of option 6 is to count tne amount of equal items on the scale.

To do this put one item on the scale, enter it's weight as option 6, then add unknown amount of items on scale and it will report the amount of items.

thucar, 2019-05-13:
Ah! That makes sense now that you mention it. So there is no way to tell the scale (you currently have xxKg weight on you)? I'm trying to monitor a weight of an object over a long period of time and device restarts are posing a real issue.

ascillato2, 2019-05-19:
Closing as this feature has been added.

Thanks everyone for working on this :+1:

closed 2019-05-19 by ascillato2