arendst/Tasmota #5787: MiDesk Lamp support doesnt work in latest dev (#define ROTARY_V1)

Jason2866, 2019-05-12:
### BUG DESCRIPTION
Compiler stops with error:
```
/workspace/Sonoff-Tasmota/sonoff/support_rotary.ino: In function 'void RotaryHandler()':
/workspace/Sonoff-Tasmota/sonoff/support_rotary.ino:133:34: error: 'LightSetDimmer' was not declared in this scope
LightSetDimmer((uint8_t)d);
^

```

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : _____
- [x] Tasmota binary firmware version number used : latest dev 6.5.0.9 self compiling
- [x] Development IDE - Compiler / Upload tools used : Gitpod / none
- [] Provide the output of command ``status 0`` :
```
None since compile does fail

@s-hadinger i think it is related to the change in light driver

```

### TO REPRODUCE
uncomment in my_user_config.h
`#define ROTARY_V1 `    

### EXPECTED BEHAVIOUR
Successful compile

[image]

Staars, 2019-05-13:
With this commit: https://github.com/arendst/Sonoff-Tasmota/commit/31e80c8c3e6c19ba538e17a7f3f8c013c8155e3b#diff-2b00018f67277999e00c6cff03ec519a

LightSetDimmer() was removed and must be refactored now.

Maybe https://github.com/s-hadinger can have a look?

s-hadinger, 2019-05-13:
Oops, I thought LightSetDimmer() was not used anymore.

Sent PR #5792 to solve this.

PR #5792 (2019-05-13): Put back LightSetDimmer, see #5787
## Description:
Put back LightSetDimmer(), was still used by support_rotary.ino.

**Related issue:** fixes #5787 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched (Also remember to update _changelog.ino_ file)
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works.
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md#c [... cut]
closed 2019-05-13 by arendst