arendst/Tasmota #5789: Sonoff Pow R2 Energy Today jumping 100Wh in 10 sec

marcelroeder, 2019-05-12:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
Hello,
i use an Sonoff Pow R2 flashed with tasmota, to measure the energy production of my little photovoltaic system. My problem is that the value "Energy Today" is jumping 100Wh within 10 seconds. 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff Pow R2
- [x] Tasmota binary firmware version number used : 6.5.0.7 / (self-compiled)
- [x] Development IDE - Compiler / Upload tools used : arduino 1.8.9
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

13:14:30 CMD: status 0
13:14:30 MQT: stat/sonoffsolar/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoff"],"Topic":"sonoffsolar","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:14:30 MQT: stat/sonoffsolar/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:31:15","StartupUTC":"2019-05-12T10:43:15","Sleep":0,"CfgHolder":4617,"BootCount":18,"SaveCount":91,"SaveAddress":"F9000"}}
13:14:30 MQT: stat/sonoffsolar/STATUS2 = {"StatusFWR":{"Version":"6.5.0.7(sonoff)","BuildDateTime":"2019.04.15 07:59:19","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
13:14:30 MQT: stat/sonoffsolar/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ASUS",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
13:14:30 MQT: stat/sonoffsolar/STATUS4 = {"StatusMEM":{"ProgramSize":572,"Free":428,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","0FDAE394","000783A0","23B617CE","00003BC0"]}}
13:14:30 MQT: stat/sonoffsolar/STATUS5 = {"StatusNET":{"Hostname":"sonoffsolar-2206","IPAddress":"192.168.0.202","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"BC:DD:C2:8E:68:9E","Webserver":2,"WifiConfig":4}}
13:14:30 MQT: stat/sonoffsolar/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.124","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8E689E","MqttUser":"test","MqttCount":3,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:14:30 MQT: stat/sonoffsolar/STATUS7 = {"StatusTIM":{"UTC":"Sun May 12 11:14:30 2019","Local":"Sun May 12 13:14:30 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:12","Sunset":"21:20"}}
13:14:30 MQT: stat/sonoffsolar/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:14:30 MQT: stat/sonoffsolar/STATUS10 = {"StatusSNS":{"Time":"2019-05-12T13:14:30","ENERGY":{"TotalStartTime":"2019-04-18T00:00:00","Total":66.339,"Yesterday":0.448,"Today":2.041,"Power":642,"ApparentPower":696,"ReactivePower":267,"Factor":0.92,"Voltage":232,"Current":3.004}}}
13:14:30 MQT: stat/sonoffsolar/STATUS11 = {"StatusSTS":{"Time":"2019-05-12T13:14:30","Uptime":"0T00:31:15","Vcc":3.463,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Wifi":{"AP":1,"SSId":"ASUS","BSSId":"08:62:66:8E:70:E0","Channel":5,"RSSI":30,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
A small increase of the value "Energy Today" for about 1-5Wh, but not 100Wh.

### SCREENSHOTS
Here is a screenshot out of grafana, which shows the problem. 
At 12:55:10 the "Energy Today" value was 1.764 kWh.
At 12:55:20 the "Energy Today" value was 1.866 kWh.
That is an increase of 102 Wh. But the increase should be about 2 Wh.
[image]

### ADDITIONAL CONTEXT
This problem occurs irregulary. On some days it occurs 3-4 times, on other days not once. 

**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2019-05-12:
Can confirm, same here with one of 5 Pow 2. The issue is based on wrong data coming from the energy chip. The checksum of the data is correct but the data contains fantasy values (can be Seen with loglevel 4). I still couldn't decide if this is a communication issue on the serial line between the ESP and the cse7766 or it's an power supply issue of this single Sonoff Pow R2 or the cse7766 is bad. 
Will also try to get some more details

ascillato2, 2019-05-13:
Sorry, seems to be just a hardware defect. Please, if you can, try with another POW R2 or with a PZEM.

marcelroeder, 2019-05-13:
Ok, i need to order a new Pow R2 and will test it again. Thanks for the informations

rt400, 2019-05-13:
i have the same issue...
i try to make calibrtion also but sometime he got on current high measure .
 i use 3x PZEM that connect to 3x WEMOS D1 MINI.
[image]
for example.
i did many time calibrtion but he always got for couple second high current measure. the problem on 2 of them. the third one work great.

meingraham, 2019-05-13:
@rt400 As the PZEM is a dedicated energy monitor, device calibration **_in TASMOTA_** is currently not supported.

https://github.com/arendst/Sonoff-Tasmota/wiki/PZEM004T-Energy-Monitoring-Custom-Devices

curzon01, 2019-05-13:
@rt400 this issue here is specific to Sonoff Pow R2 having CSE7766 chip and where energy today value step only exceeded range, pls don't mix things.

arendst, 2019-05-13:
@curzon01 if possible could you provide some serial output of bad response from CSE7766.

It might have something to do with low power detection I introduced some time ago. Default valid received code goes down to about 5W. Trickery makes power measurements down to 1W possible but there might be something wrong in processing on my side.

curzon01, 2019-05-13:
Theo, I'd debug today where the problem occurs and unfortunately it seems the reason is not your assumtion. One of my Pow R2 realy delivers sometimes garbadge from CSE so it might be a CSE7766 quality issue. For example:

```
OK  2019-05-13 12:47:52 55 5A 02 F9 B8 00 03 3F 00 3F DE 00 21 12 51 38 F8 00 2D C3 71 5F 84 0A
OK  2019-05-13 12:47:52 55 5A 02 F9 B8 00 03 3C 00 3F DE 00 21 12 51 38 F8 00 2D C3 71 5F 8D 10
OK  2019-05-13 12:47:52 55 5A 02 F9 B8 00 03 3C 00 3F DE 00 21 12 51 38 F8 00 2D C8 71 5F 96 1E
Bad 2019-05-13 12:47:52 55 5A 02 F9 B8 00 03 3C 00 3F DE 00 21 12 51 38 F8 00 FF 7D 9F 27 55 5A
OK  2019-05-13 12:47:53 55 5A 02 F9 B8 00 03 3C 00 3F DE 00 21 12 51 38 F8 00 2D B3 71 5F B1 24
OK  2019-05-13 12:47:53 55 5A 02 F9 B8 00 03 3C 00 3F DE 00 21 12 51 38 F8 00 2D B3 71 5F BA 2D
OK  2019-05-13 12:47:53 55 5A 02 F9 B8 00 03 3C 00 3F DE 00 21 12 51 38 F8 00 2D BF 71 5F C3 42
```
decoding the data you will see for the `Bad` line the checksum is ok but in this example the cf_pulse value is stepping down from previous 24470 to 10069 and then back to next correct 24497.
Unfortunately there are several other combinations of data I'd found where for example I/V/PCal values are jumping randomly.

I'd made small change in CseEverySecond() preventing overload values which can only be from a load with more than 5kW which is out of operation range of the Pow. 
Let me check it 'till tomorrow and you can have a look into my upcoming PR

closed 2019-05-14 by arendst

meingraham, 2019-05-20:
Does the Sonoff S31 have overtemp circuitry like the POW? If so, does [e472d32](https://github.com/arendst/Sonoff-Tasmota/commit/e472d3232a99568a2949995e75147bf16d4535f5) also add overtemp detection to the S31?

If not, nevermind :wink:

Mike

arendst, 2019-05-20:
Both POW and S31 have NO OverTemp circuitry.

Only Shelly 1PM and Shelly 2.5 have overtemp circuitry

meingraham, 2019-05-20:
Thx. The mingling of the "CSE7766 Sensor (Sonoff S31/Pow R2)" and the reference to "Add device OverTemp (>73 Celsius) detection to selected Energy Monitoring devices" in this thread is what caused my confusion.

goosst, 2019-06-25:
I can report a probably related issue, the energy calculation is incorrect on the device I'm using:
- Program Version | 6.5.0(release-sonoff)
- calibrated with a 40W lightbulb (works accurately)
- status0:

> 19:53:28 MQT: stat/sonoffPowR2/STATUS = {"Status":{"Module":43,"FriendlyName":["SonoffPow"],"Topic":"sonoffPowR2","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

I use the device to track energy usage of a robotic mower, the reported power and energy is plotted below (coming through the tele messages). As you can see the energy is a few orders of magnitude too large for the used power and makes very high jumps. 

[image]

[image]

or directly taken from browsing to the device:
[image]

PR #5793 (2019-05-14): CSE7766 Sensor (Sonoff S31/Pow R2) prevent invalid energy load steps
## Description:

prevent invalid load delta steps even checksum of datastream from CSE is valid (issue #5789)

**Related issue (if applicable):** fixes #5789

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched (Also remember to update _changelog.ino_ file)
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works.
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA [... cut]