arendst/Tasmota #5791: Rule go in blinking the relais

riddik14, 2019-05-12:
I have update from 6.4.1 core 2.3 to 6.5 release core 2.3 on SonOff. When i have Actived the rule 

rule1 on power1#state do backlog POWER %value%; ruletimer1 60 endon on rules#timer=1 do POWER OFF endon

And i turn on the light with button, the light go in blinking. Why? Is a bug or my rule have a error?

Tanks u

meingraham, 2019-05-13:
Every time the power state changes, this rule will fire. So, you push the button, you set the power to the new state. 60 seconds later, the power state changes. The rule triggers. Repeat.

Have you tried [PulseTime 60](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#pulsetime)

ascillato2, 2019-05-13:
Closing as the question has been answered. 

Please, if you need further assistant, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-13 by ascillato2