arendst/Tasmota #5797: ifan2 flashing

lukaszwitkowski, 2019-05-15:
hello 
I use Tasmota in my home.
I bay ifan2 and I have problem with flashing that esp
when I uploading firmware I have that error

warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08

i have gpio0 connected to group and I use 220v to power ifan2 when flashing

Jason2866, 2019-05-15:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-05-15:
Closing and locking this issue as the user is not following the wiki.

closed 2019-05-15 by ascillato2