arendst/Tasmota #5799: Core 2.5.1 and Tasmota

helmar74, 2019-05-15:
Hi,

I "just" have a few questions according Core 2.5.1:
Which issues are fixed if using Core 2.5.1?
Is there a documentation, which features are used with Core 2.5.1?
Is it a good idea to use Core 2.5.1 or should an (older) core be used?

Thanks a lot 

Helmar

Jason2866, 2019-05-15:
If everything is working without an issue, stay with the core you are using now.
If you have issues give it a try. Changes see here https://github.com/esp8266/Arduino/releases

helmar74, 2019-05-15:
I have currently the issue with mqtt reconnects. I really tried a lot, but I couldn't find an solution. Will try Core 2.5.1

arendst, 2019-05-15:
What Tasmota version are you using while having the mqtt reconnects? Pls provide issue information as requested during opening this issue.

Jason2866, 2019-05-15:
Have you tried core 2.3.0 for solving mqtt reconnects? Helped many users.

helmar74, 2019-05-15:
Yes I tried 2.3.0, but I had several other issues with it (e.g. not working OTA).
@arendst : I'm using 6.5.0 production release and also 6.5.0.3 on a IR Bridge.
Here's the Bridges status 0:
19:28:31 MQT: stat/sonoffIRBridge_1/STATUS = {"Status":{"Module":62,"FriendlyName":["Sonoff IR Bridge 1"],"Topic":"sonoffIRBridge_1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:28:31 MQT: stat/sonoffIRBridge_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffStandardGruppe","OtaUrl":"http://10.10.10.50:8080/static/firmware/sonoffStandard.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:18","StartupUTC":"2019-05-15T12:18:13","Sleep":50,"CfgHolder":4617,"BootCount":15,"SaveCount":37,"SaveAddress":"FA000"}}
19:28:31 MQT: stat/sonoffIRBridge_1/STATUS2 = {"StatusFWR":{"Version":"6.5.0.3(sonoff)","BuildDateTime":"2019.05.15 14:13:57","Boot":31,"Core":"2_5_1","SDK":"2.2.1(cfd48f3)"}}
19:28:31 MQT: stat/sonoffIRBridge_1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
19:28:31 MQT: stat/sonoffIRBridge_1/STATUS4 = {"StatusMEM":{"ProgramSize":582,"Free":420,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000407","0FAAE19C","00078FA0","23BA1FCE","002033C0"]}}
19:28:31 MQT: stat/sonoffIRBridge_1/STATUS5 = {"StatusNET":{"Hostname":"sonoffIRBridge-1","IPAddress":"10.10.10.160","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.10.2","Mac":"EC:FA:BC:20:81:36","Webserver":2,"WifiConfig":4}}
19:28:31 MQT: stat/sonoffIRBridge_1/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.10.50","MqttPort":1883,"MqttClientMask":"sonoffIRBridge-1","MqttClient":"sonoffIRBridge-1","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:28:31 MQT: stat/sonoffIRBridge_1/STATUS7 = {"StatusTIM":{"UTC":"Wed May 15 17:28:31 2019","Local":"Wed May 15 19:28:31 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"05:28","Sunset":"20:42"}}
19:28:31 MQT: stat/sonoffIRBridge_1/STATUS10 = {"StatusSNS":{"Time":"2019-05-15T19:28:31"}}
19:28:31 MQT: stat/sonoffIRBridge_1/STATUS11 = {"StatusSTS":{"Time":"2019-05-15T19:28:31","Uptime":"0T00:10:18","Vcc":3.399,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID","BSSId":"CC:CE:1E:AB:0A:3F","Channel":6,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:04"}}}

helmar74, 2019-05-15:
I've updated several devices and will provide feedback

arendst, 2019-05-15:
Try latest development version as mqtt retry has changed since 6.5.0

helmar74, 2019-05-15:
OK, give it a try ;-)

helmar74, 2019-05-15:
Updated several devices to 6.5.0.10 and Core 2.5.1.
Will provide feedback about reconnects in a few days

helmar74, 2019-05-15:
Here's a first result:
The Devices has still mqtt reconnects.
A Sonoff SC has 4 wifi reconnects (and of course then 4 mqtt reconnects)
A Wemos Mini also had 4 wifi reconnects (but is connected to an other WIFI than the SC)
A IR Bridge (which is connected to the same WIFI as Sonoff SC) had no wifi and mqtt reconnects until now...

Pho7ography, 2019-05-15:
Hi, same here, core 2.3.0 loose wifi and mqtt, core 2.4.2 loose connection to mqtt.

Jason2866, 2019-05-15:
@Pho7ography  without any infos (minimal `Status 0` ) no one can help.

ascillato2, 2019-05-19:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-19 by ascillato2

helmar74, 2019-05-19:
I flashed 3 Devices with Core 2.5.1 and Tasmota 6.5.0.10 Development Release.
- IR Bridge doesn't have any reconnects more any longer
- Sonoff SC still have Wifi Reconnects (6 since last restart) and mqtt reconnects (12 since restart). Sonoff SC and IR Bridge are located in the same place and are connected to the same WIFI
- Wemos Mini (is located in cellar) has 50 Wifi (and of course 50 mqtt reconnects). Is connected to an other Wifi

Plarkass, 2020-04-16:
Hi @helmar74 ,

Could you explain me how to update core ?
I didn't find yet any doc for that.

ascillato, 2020-04-16:
Hi @Plarkass 

The core is selected when you are compiling your own version. See compilation section on the docs and the file platformio.ini.

Anyway, the Actual version of the core is 2.6.3. We do not recommend 2.5.1 anymore because it has several bugs and stability issues.

If you flash the precompiled bins, provided in http://thehackbox.org/tasmota , they have already the latest version of the core with latest patches.

helmar74, 2020-04-17:
> Hi @helmar74 ,
> 
> Could you explain me how to update core ?
> I didn't find yet any doc for that.

Hi,

either you use a pre compiled binary or you compile for you own.
I'm using Arduino IDE for compilation, it's described here:
https://tasmota.github.io/docs/Arduino-IDE/
(Also the usage of the Core).
To use a other core you can change the URL (named in documentation):
`https://github.com/esp8266/Arduino/releases/download/2.6.1/package_esp8266com_index.json`

Regards

Helmar

Plarkass, 2020-04-20:
Hi,

Thanks for all of your explaination.
I actually use binaries from http://thehackbox.org/tasmota/ and i am still on Core/SDK STAGE/2.2.2-dev(38a443e), i think i miss something.

Below my plug infos:
Program Version | 8.2.0.4(fc74cb2-tasmota)
-- | --
2020-04-19T23:00:13
STAGE/2.2.2-dev(38a443e)

ascillato, 2020-04-20:
The name STAGE means that your core is the one that is being developed now. The beta. Yours is higher than 2.6.3.
The 2.2.2 you see there is the version of the SDK (that is the recommended one).
So, you are using latest versions

Plarkass, 2020-04-22:
Thnaks a lot for these explanation