arendst/Tasmota #5801: Rule does not work anymore on 6.5.0

mikeww1, 2019-05-15:
After installing 6.5.0 with rules enabled on my Sonoff 2 ch glas switch I found out that my usual timer does not work anymore. I use the rule from the example:

Rule1 on button1#state do backlog power1 %value%; RuleTimer1 300 endon on #Timer=1 do power1 off endon
Rule1 1

That worked fine under 6.4.1. Now the rule seems to register correctly but it does not switch the lamp off. There is also no message from this rule in the console.

DavinKD, 2019-05-15:
Shouldn't it be Rule1 on button1#state do backlog power1 %value%; RuleTimer1 300 endon on Rules#Timer=1 do power1 off endon

Jason2866, 2019-05-15:
Please close this issue since your question is answered.

mikeww1, 2019-05-15:
Sorry does not work. Here is the log:
`19:57:53 CMD: Rule1 on button1#state do backlog power1 %value%; RuleTimer1 300 endon on Rules#Timer=1 do power1 off endon
19:57:53 MQT: sonoff-heizraum/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":410,"Rules":"on button1#state do backlog power1 %value%; RuleTimer1 300 endon on Rules#Timer=1 do power1 off endon"}
19:58:12 CMD: Rule1 1
19:58:12 MQT: sonoff-heizraum/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":410,"Rules":"on button1#state do backlog power1 %value%; RuleTimer1 300 endon on Rules#Timer=1 do power1 off endon"}
19:58:23 MQT: sonoff-heizraum/tele/STATE = {"Time":"2019-05-15T19:58:23","Uptime":"6T03:42:31","Vcc":3.184,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"POS Systemhaus","BSSId":"F4:06:8D:05:1A:5C","Channel":1,"RSSI":62,"LinkCount":1,"Downtime":"0T00:00:04"}}
19:58:23 MQT: sonoff-heizraum/stat/RESULT = {"POWER1":"ON"}
19:58:23 MQT: sonoff-heizraum/stat/POWER1 = ON
20:00:33 MQT: sonoff-heizraum/tele/HASS_STATE = {"Version":"6.5.0(sonoff)","BuildDateTime":"2019.05.06 17:24:26","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Sonoff T1 2CH","RestartReason":"Software/System restart","Uptime":"6T03:44:41","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:04","MqttCount":34,"BootCount":2,"SaveCount":45,"IPAddress":"10.0.0.69","RSSI":"64","LoadAvg":25}
20:00:45 MQT: sonoff-heizraum/tele/STATE = {"Time":"2019-05-15T20:00:45","Uptime":"6T03:44:53","Vcc":3.186,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"POS Systemhaus","BSSId":"F4:06:8D:05:1A:5C","Channel":1,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:02:00 MQT: sonoff-heizraum/tele/UPTIME = {"Time":"2019-05-15T20:02:00","Uptime":"6T03:46:08"}
20:05:42 MQT: sonoff-heizraum/tele/HASS_STATE = {"Version":"6.5.0(sonoff)","BuildDateTime":"2019.05.06 17:24:26","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Sonoff T1 2CH","RestartReason":"Software/System restart","Uptime":"6T03:49:50","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:04","MqttCount":34,"BootCount":2,"SaveCount":45,"IPAddress":"10.0.0.69","RSSI":"64","LoadAvg":30}
20:06:06 MQT: sonoff-heizraum/tele/STATE = {"Time":"2019-05-15T20:06:06","Uptime":"6T03:50:14","Vcc":3.184,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":141,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"POS Systemhaus","BSSId":"F4:06:8D:05:1A:5C","Channel":1,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:04"}}`

mikeww1, 2019-05-15:
I try to make more readable:

19:57:53 CMD: Rule1 on button1#state do backlog power1 %value%; RuleTimer1 300 endon on Rules#Timer=1 do power1 off endon
19:57:53 MQT: sonoff-heizraum/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":410,"Rules":"on button1#state do backlog power1 %value%; RuleTimer1 300 endon on Rules#Timer=1 do power1 off endon"}
19:58:12 CMD: Rule1 1

After that there are no more log entries in connection with the rule and the light is still on.

Jason2866, 2019-05-15:
Please try the prebuild version from http://thehackbox.org/tasmota/release/sonoff.bin
This address can be used for OTA update too. The rule is correct and should work so something else is faulty.

mikeww1, 2019-05-17:
I flashed the prebuild version but that does not make any difference. The rule does not work.
I will now get back to an older version because this function is the most important one for me.
I leave this here open. Maybe it is a bug in the 6.5.0 version?

mikeww1, 2019-05-18:
OK I could solve the problem. The rule works but only when you physically switch on the light.
I tested it with the web interface. When you switch on the light via web interface the rule is not triggered.
I dont know if this behaviour is intended.

Jason2866, 2019-05-18:
As designed since button(s) or switch(es) are connected hardware.

mikeww1, 2019-11-13:
I have to come back to this issue. Yesterday I configured Mqtt to connect this switch to Openhab.
The Sonoff switch is now triggered by Openhab via Mqtt. It switches on but never switches off.
That means that the rule is not triggered by mqtt. I cannot believe that this is an intended behavior.
I reread the Wiki but could not find a definition whether this should work or not. On the other hand why should a rule not be triggered by a mqtt command?

mikeww1, 2019-11-13:
That is why I reopen the issue.

Jason2866, 2019-11-13:
It does what it should do:
`Rule1 on button1#state do backlog power1 %value%; RuleTimer1 300 endon on Rules#Timer=1 do power1 off endon`
You do not press a button when you send a command (to power on) via mqtt.

ascillato2, 2019-11-13:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-13 by ascillato2

meingraham, 2019-11-13:
To work, use a rule that triggers and event when the button is pressed. The openHAB message can trigger the same event. Then, the actions you want to perform are triggered based on the event rather than the button press.

As suggested above - for further assistance please address it via the Discord chat.