arendst/Tasmota #5804: LC Tech Wifi Relay *always* starts up with relay ON

richard-parker, 2019-05-15:
Hi there,

Despite trying several configurations, I'm unable to stop this device triggering its relay state to ON after a reboot. As I'm trying to use this as a momentary switch to trigger a garage door (using `pulsetime 10` for a 1-second push), having it trigger the relay to ON on startup has the effect of opening our garage door.

I've tried `setoption0 0` to have tasmota remember its last known state. 

My `PowerOnState` is `0`.

I've also followed the setup guide on this Wiki and have set the appropriate `rule`:

`Rule1 on System#Boot do Baudrate 9600 endon on Power1#State=1 do SerialSend5 A00101A2 endon on Power1#State=0 do SerialSend5 A00100A1 endon`

I've ensured *retain messages* is OFF on MQTT.

Below is my log - any help appreciated.

`
00:00:00 SRC: Restart
00:00:00 Project sonoff SonoffGarage Version 6.5.0(release-sonoff)-2_3_0
00:00:00 RUL: POWER1#BOOT performs "Baudrate 9600"
00:00:00 SRC: Rule
00:00:00 RSL: Received Topic /Baudrate, Data Size 4, Data 9600
00:00:00 RSL: Group 0, Index 1, Command BAUDRATE, Data 9600
00:00:00 RSL: stat/sonoffGarage/RESULT = {"Baudrate":9600}
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 xxx in mode 11N as sonoffGarage-4669...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sonoffGarage-4669 with IP address 10.10.xxx.xxx
00:00:06 APP: (UTC) Wed May 15 19:43:55 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
20:43:56 HTP: Main Menu
20:43:57 MQT: Attempting connection...
20:43:57 MQT: Connected
20:43:57 MQT: tele/sonoffGarage/LWT = Online (retained)
20:43:57 MQT: cmnd/sonoffGarage/POWER = 
20:43:57 MQT: Subscribe to cmnd/sonoffGarage/#
20:43:57 MQT: Subscribe to cmnd/sonoffs/#
20:43:57 MQT: Subscribe to cmnd/sonoffGrg_fb/#
20:43:57 SRC: MQTT
20:43:57 RSL: Received Topic cmnd/sonoffGarage/power, Data Size 2, Data ON
20:43:57 RSL: Group 0, Index 1, Command POWER, Data ON
20:43:57 MQT: stat/sonoffGarage/RESULT = {"POWER":"ON"}
20:43:57 MQT: stat/sonoffGarage/POWER = ON
20:43:57 MQT: tele/sonoffGarage/INFO1 = {"Module":"1 Channel","Version":"6.5.0(release-sonoff)","FallbackTopic":"cmnd/sonoffGrg_fb/","GroupTopic":"sonoffs"}
20:43:57 MQT: tele/sonoffGarage/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffGarage-4669","IPAddress":"10.10.xxx.xxx"}
20:43:57 MQT: tele/sonoffGarage/INFO3 = {"RestartReason":"Software/System restart"}
20:43:57 MQT: stat/sonoffGarage/RESULT = {"POWER":"ON"}
20:43:57 MQT: stat/sonoffGarage/POWER = ON
20:43:57 RUL: POWER1#STATE=1 performs "SerialSend5 A00101A2"
20:43:57 SRC: Rule
20:43:57 RSL: Received Topic /SerialSend5, Data Size 8, Data A00101A2
20:43:57 RSL: Group 0, Index 5, Command SERIALSEND, Data A00101A2
20:43:57 MQT: stat/sonoffGarage/RESULT = {"SerialSend":"Done"}
20:43:58 CFG: Saved to flash at FA, Count 145, Bytes 3584
20:43:59 APP: Boot Count 49
20:44:00 CFG: Saved to flash at F9, Count 146, Bytes 3584
20:44:00 HTP: Console
`
Thanks.

meingraham, 2019-05-15:
It appears that there is a retained message in your broker:
```
20:43:57 SRC: MQTT
20:43:57 RSL: Received Topic cmnd/sonoffGarage/power, Data Size 2, Data ON
20:43:57 RSL: Group 0, Index 1, Command POWER, Data ON
```

Please, close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota software feature requests and bug reporting. Troubleshooting is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-15 by ascillato2

clabnet, 2019-10-13:
I have the same identical issue with LC Tech Wifi one relay single.
My settings: 
PowerOnState = 0; PowerRetain= OFF; Cleared retained messages on mosquitto broker.
The relay go to power ON at boot. It toggle on/off regularly after the command.
@richard-parker I do not know other way to solve this issue. And you ?

richard-parker, 2019-12-14:
> I have the same identical issue with LC Tech Wifi one relay single.
> My settings:
> PowerOnState = 0; PowerRetain= OFF; Cleared retained messages on mosquitto broker.
> The relay go to power ON at boot. [...]

Sorry for the late reply! Check again for a retained message? I still have this issue; the relay comes up at power up to Power On, then immediately powers Off, which suggests may be a hardware issue. Due to this behaviour I’m not able to rely on it as a garage door trigger, since a relay switched to on even for a brief period opens the garage door.