arendst/Tasmota #5805: setOption56 and setOption57 with hidden SSIDs

dlaszlo, 2019-05-15:
I have three AP with the same SSID name. I would like to use setOption56 and setOption57 but my SSID is hidden (and I would like to use hidden SSID). At the moment Tasmota does not list the hidden networks when it tries to select the strongest signal. It is reasonable for me because Tasmota compares the listed SSID names (which is empty when the SSID is hidden) and the stored (which is stored in the configuration) SSID names. 

Is it possible to solve this problem?

This is the related forum thread: https://groups.google.com/forum/#!topic/sonoffusers/LxJYc9yAwDQ

Thank you,
Best regards,
Dávid László

meingraham, 2019-05-15:
@arendst & @ascillato 

I worked with Dávid on Google Groups (the link he provided) and after arriving at the reason for his issue, I suggested that he open an issue. The library call allows to scan hidden networks, however, the Tasmota logic expects to use the SSID in it's scans. For hidden networks, the SSID is not available.

I suggested that he post this issue to see if you had any considerations for adapting SetOption56 & SetOption57 for hidden SSIDs. It's more a philosophical question perhaps. We just wanted to get your thoughts.

Regards.

Mike

digiblur, 2019-05-16:
Curious why the need for masked SSIDs, can't say I can call them hidden anymore.

joba-1, 2019-05-16:
There is a usecase for hidden ssids (not security, but convenience of users of other ssids by not cluttering scan lists with networks that are not intended to be found by scans anyways), so...

If you want to use signal quality for AP selection of hidden ssids, you would have to store the bssid of all access points making up the hidden ssid and compare these. Could be a long list - what do you want to sacrifice for the required space? ( me: nothing ;) )

Jason2866, 2019-05-16:
My 2 cents. Most ***not*** needed feature for Tasmota. Hide SSID is such a useless option. One more or less SSID in a network scan list....

dlaszlo, 2019-05-16:
The hidden SSID is not enough to secure a network (this is true), but this is a part of a network configuration what I would like to use:

I have a configured network:
- Virtual APs, and separated subnet for Sonoff devices, switches, sensors and other similar devices, which cannot reach the internet or other things in the local network.
- I cannot reach this network from the Internet or normal local network, only through an RPI server.
- The hidden SSID is only a part of this list.

My use case: If XY (who is an average person) scan networks, he will not see my network (for home smart devices). Or he must think this is not a normal network for him (to reach the Internet for example). 

I have started this topic because of the following reasons:

- The trivial solution is to store BSSId, but I agree with maybe this is not worth it because of memory reasons. 
Maybe there is another way to check the hidden network if it is a particular network (by SSID name, it is stored in the configuration), 
The "normal connect to the hidden network" function works, maybe it is possible. But I don't know the ESP WIFI API at this level.

- I read the WIKI, how SetOption56 works, and I turned this option on, and it did not work. I did not read about the hidden networks. 
Maybe It would be enough to add this information to WIKI (if the WIKI does not contain it). 

- When we tried to figure out what is the problem, I got the following message: "Connect failed with AP timeout" but not this happened, it did not find the network after the network scan.

Thank you very much.

dlaszlo, 2019-05-16:
@Jason2866: "Hide SSID is such a useless option. One more or less SSID in a network scan list...."
I can agree with this. :) But this is a popular wifi setting and at this moment setOption56 can cause surprise for users.

Jason2866, 2019-05-16:
Info added in commands section setoptions

dlaszlo, 2019-05-16:
@Jason2866: Thank you. :) Can I close this issue or this is not my competence?

meingraham, 2019-05-16:
Dávid - since you opened the issue, you are able to close it. You should see an extra button in the comment box. You don't need to enter a comment. You can just click Close.

closed 2019-05-16 by dlaszlo