arendst/Tasmota #5806: IR code format conversion

jjmuriel, 2019-05-16:
Hi,
Please, can someone tell me how to convert the format of the IR code for obtain the raw code, inside this commands:
"Data":1..(2^32)-1 = data frame as 32 bit decimal.
    e.g., IRsend {"Protocol":"NEC","Bits":32,"Data":2170978686}
or
"Data":0x1..0xFFFFFFFF = data frame as 32 bit hexadecimal.
    e.g., IRsend {"Protocol":"NEC","Bits":32,"Data":0x8166817E}

That is, change the "2170978686" or ":0x8166817E" code to raw format and vice versa. Thanks.

meingraham, 2019-05-16:
Have you reviewed the information on raw encoding in the wiki?
https://github.com/arendst/Sonoff-Tasmota/wiki/IRSend-RAW-Encoding

That article includes a link to https://github.com/ToniA/Raw-IR-decoder-for-Arduino

Please close this issue and address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota software feature requests and bug reporting. Troubleshooting is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

jjmuriel, 2019-05-16:
Traducir del español
I'll check the links. Thank you for your quick response.  Sorry, in the future I'll use the forum.

closed 2019-05-16 by jjmuriel