arendst/Tasmota #5807: Codes from wall-remote only discovered in raw-mode

10mabo, 2019-05-16:
Hello,

I have flashed a sonoff rf bridge V2 with tasmota and the portisch firmware. I have a portable remote control and a wall-remote. The portable remote works just fine, but with the wall-remote, the bridge receives codes only in raw mode (RfRaw 177). In normal operation mode (after powering on the bridge), nothing is received.

Picture of the wall-remote: https://i.ebayimg.com/images/g/o0YAAOSwIWVY9xfz/s-l500.jpg

I woud like to use the bridge mostly as an rf receiver that reacts on button presses of the rf remotes to control non-rf devices for example wifi plugs (with node-red).

As the bridge seems to discover the wall-remotes button presses only in raw mode - is it possible to configure the bridge to permanently stay in raw mode? Or what else could I do... is there a mode that discovers raw an "normal" codes and sends it over mqtt?

Best regards,
Marc

meingraham, 2019-05-16:
Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and general setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

10mabo, 2019-05-16:
@meingraham ok, thanks!

closed 2019-05-16 by 10mabo