arendst/Tasmota #5812: Tasmota 6.5.0 cannot connect to WLAN after LED strip is toggled

probonopd, 2019-05-17:
### BUG DESCRIPTION

Tasmota 6.5.0 cannot connect to WLAN after LED strip is toggled

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : NodeMCU 1.0
- [x] Tasmota binary firmware version number used : sonoff.bin 6.5.0 from GitHub Releases
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :

### TO REPRODUCE

```
me@host:~$ sudo chmod a+rwx /dev/ttyUSB0 ;  sudo '/home/me/Downloads/esptool-2.6/esptool.py' --chip esp8266 --port /dev/ttyUSB0 erase_flash

me@host:~$ sudo chmod a+rwx /dev/ttyUSB0 ;  sudo '/home/me/Downloads/esptool-2.6/esptool.py' --chip esp8266 --port /dev/ttyUSB0 write_flash 0x00000 '/home/me/Downloads/sonoff.bin'

sonoff-XXXX access point is visible, I can connect to it, do a network scan, enter my credentials, click Save

ESP reboots, connects to the WLAN

After resetting, Serial port says

00:00:00 Project sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 h48 in mode 11N as sonoff-XXXX...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-4090 with IP address 192.168.XXX.XXX

In the web interface, I select Configuration -> Configure Module -> Generic(18), click Save

After resetting, Serial port says

00:00:00 Project sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 h48 in mode 11N as sonoff-XXXX...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-4090 with IP address 192.168.XXX.XXX
In the web interface, I select Configuration -> Configure Module -> D2 GPIO4 -> WS2812 (7), click Save

After resetting, Serial port says

00:00:00 Project sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 h48 in mode 11N as sonoff-4090...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-4090 with IP address 192.168.0.25
In the web interface, I click "Toggle"

11:01:06 WIF: Connecting to AP1 h48 in mode 11N as sonoff-4090...
11:01:13 WIF: Connect failed as AP cannot be reached
11:01:13 WIF: Connecting to AP1 h48 in mode 11N as sonoff-4090...
11:01:20 WIF: Connect failed as AP cannot be reached
11:01:21 WIF: Connect failed as AP cannot be reached
11:01:21 WIF: Connecting to AP1 h48 in mode 11N as sonoff-4090...

From thereon, it does not connect to WLAN anymore. Also not after resetting

At this point, I can only revive functionality by

me@host:~$ sudo chmod a+rwx /dev/ttyUSB0 ;  sudo '/home/me/Downloads/esptool-2.6/esptool.py' --chip esp8266 --port /dev/ttyUSB0 erase_flash

and reflashing new firmware.
```

joba-1, 2019-05-17:
could be power issue. What if you disconnect the strip and reboot? Still no wlan?

probonopd, 2019-05-17:
Yes, the same happens when the strip is not attached at all.

Jason2866, 2019-05-18:
Please post `status 0`

probonopd, 2019-05-18:
22:05:35 RSL: stat/sonoff/STATUS = {"Status":{"Module":52,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
22:05:35 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:00","StartupUTC":"2019-05-18T21:01:35","Sleep":50,"CfgHolder":4617,"BootCount":4,"SaveCount":19,"SaveAddress":"F9000"}}
22:05:35 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019-03-24T13:45:27","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:05:35 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["h48",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","280500000100000000000000000000000000","00000000"]}}
22:05:35 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":447,"Free":556,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0F402390","00048000","00000096","000000C0"]}}
22:05:35 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2246","IPAddress":"192.168.XXX.XXX","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.XXX.XXX","Mac":"60:01:94:XX:XX:XX","Webserver":2,"WifiConfig":2}}
22:05:35 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_XXXXXX","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:05:35 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat May 18 21:05:35 2019","Local":"Sat May 18 22:05:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:04","Sunset":"20:28"}}
22:05:35 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:05:35 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-05-18T22:05:35","ENERGY":{"TotalStartTime":"2019-05-18T22:00:28","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
22:05:35 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-05-18T22:05:35","Uptime":"0T00:04:00","Vcc":3.208,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":11,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:05"}}}

probonopd, 2019-05-18:
```
me@host:~$ sudo chmod a+rwx /dev/ttyUSB0 ;  sudo '/home/me/Downloads/esptool-2.6/esptool.py' --chip esp8266 --port /dev/ttyUSB0 erase_flash

me@host:~$ sudo chmod a+rwx /dev/ttyUSB0 ;  sudo '/home/me/Downloads/esptool-2.6/esptool.py' --chip esp8266 --port /dev/ttyUSB0 write_flash 0x00000 '/home/me/Downloads/sonoff.bin'

sonoff-XXXX access point is visible, I can connect to it, do a network scan, enter my credentials, click Save

ESP reboots, connects to the WLAN

00:00:05 HTP: Web server active on sonoff-XXXX with IP address 192.168.XXX.XXX
23:20:16 CMD: status 0
23:20:16 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:20:16 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:21","StartupUTC":"2019-05-18T22:19:55","Sleep":50,"CfgHolder":4617,"BootCount":2,"SaveCount":7,"SaveAddress":"F5000"}}
23:20:16 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:20:16 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["h48",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
23:20:16 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":15,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
23:20:16 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-XXXX","IPAddress":"192.168.XXX.XXX","Gateway":"192.168.YYY.YYY","Subnetmask":"255.255.255.0","DNSServer":"192.168.YYY.YYY","Mac":"5C:CF:7F:XX:XX:XX","Webserver":2,"WifiConfig":4}}
23:20:16 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B55C74","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:20:16 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat May 18 22:20:16 2019","Local":"Sat May 18 23:20:16 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:04","Sunset":"20:28"}}
23:20:16 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-05-18T23:20:16"}}
23:20:16 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-05-18T23:20:16","Uptime":"0T00:00:21","Vcc":2.799,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"h48","BSSId":"XX:XX:XX:XX:XX:XX","Channel":11,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:05"}}}

In the web interface, I select Configuration -> Configure Module -> Generic(18), click Save

After resetting, Serial port says

00:00:00 Project sonoff Sonoff Version 6.5.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 h48 in mode 11N as sonoff-XXXX...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-XXXX with IP address 192.168.XXX.XXX
23:21:50 CMD: status 0
23:21:50 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:21:50 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:12","StartupUTC":"2019-05-18T22:21:38","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":9,"SaveAddress":"FB000"}}
23:21:50 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:21:50 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["h48",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
23:21:50 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":15,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
23:21:50 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-XXXX","IPAddress":"192.168.XXX.XXX","Gateway":"192.168.YYY.YYY","Subnetmask":"255.255.255.0","DNSServer":"192.168.YYY.YYY","Mac":"5C:CF:7F:XX:XX:XX","Webserver":2,"WifiConfig":4}}
23:21:50 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B55C74","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:21:50 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat May 18 22:21:50 2019","Local":"Sat May 18 23:21:50 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:04","Sunset":"20:28"}}
23:21:50 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-05-18T23:21:50"}}
[... cut ...]
23:24:26 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-05-18T23:24:26"}}
23:24:26 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-05-18T23:24:26","Uptime":"0T00:01:30","Vcc":2.771,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":129,"POWER":"ON","Dimmer":10,"Color":"191919","HSBColor":"0,0,10","Channel":[9,9,9],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"h48","BSSId":"XX:XX:XX:XX:XX:XX","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

Jason2866, 2019-05-19:
Please erase flash with esptool.
Flash Tasmota again with the release version from here or OTA source www.thehackbox.org 
If the issues still persists you have probably a bad NodeMCU with a counterfeit Vreg or esp chip.
We had already users with the this issue.

probonopd, 2019-05-19:
> Please erase flash with esptool.
Flash Tasmota again with the release version from here 

This is exactly what I did:

```
me@host:~$ sudo chmod a+rwx /dev/ttyUSB0 ;  sudo '/home/me/Downloads/esptool-2.6/esptool.py' --chip esp8266 --port /dev/ttyUSB0 erase_flash

me@host:~$ sudo chmod a+rwx /dev/ttyUSB0 ;  sudo '/home/me/Downloads/esptool-2.6/esptool.py' --chip esp8266 --port /dev/ttyUSB0 write_flash 0x00000 '/home/me/Downloads/sonoff.bin'
```

I have tried with two different devices, with nothing attached to them.
Is there a reliable test to find out whether I have a "counterfeit Vreg or esp chip"?

Jason2866, 2019-05-19:
Sorry no. Tasmota is a good indicator because it stresses the CPU. Bad power supply and/or heat problems (cpu) are resulting in wifi issues. You can try if the NodeMCU runs @160 Mhz. If it fails you have for 100% a bad one. Entry for platformio.ini
```
; set CPU frequency to 80MHz (default) or 160MHz
;board_build.f_cpu         = 80000000L
board_build.f_cpu         = 160000000L
```

probonopd, 2019-05-19:
Can anyone confirm that the above configuration works on any WEMOS NodeMCU 1.0 (without anything attached to it)?

Jason2866, 2019-05-19:
Does work.

```
15:51:26 CMD: gpio
15:51:26 MQT: stat/sonoff-1585B3/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"7 (WS2812)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO5":"0 (None)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}

```
```
15:50:47 CMD: Status 0
15:50:47 MQT: stat/sonoff-1585B3/STATUS = {"Status":{"Module":18,"FriendlyName":["149"],"Topic":"sonoff-1585B3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:50:47 MQT: stat/sonoff-1585B3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:43","StartupUTC":"2019-05-19T13:50:04","Sleep":50,"CfgHolder":4617,"BootCount":14,"SaveCount":51,"SaveAddress":"F7000"}}
15:50:47 MQT: stat/sonoff-1585B3/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:50:47 MQT: stat/sonoff-1585B3/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
15:50:47 MQT: stat/sonoff-1585B3/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":12,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1625C2","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
15:50:47 MQT: stat/sonoff-1585B3/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1585B3-1459","IPAddress":"192.168.2.149","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"B4:E6:2D:15:85:B3","Webserver":2,"WifiConfig":5}}
15:50:47 MQT: stat/sonoff-1585B3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-%06X","MqttClient":"sonoff-1585B3","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:50:47 MQT: stat/sonoff-1585B3/STATUS7 = {"StatusTIM":{"UTC":"Sun May 19 13:50:47 2019","Local":"Sun May 19 15:50:47 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"05:26","Sunset":"20:52"}}
15:50:47 MQT: stat/sonoff-1585B3/STATUS10 = {"StatusSNS":{"Time":"2019-05-19T15:50:47"}}
15:50:47 MQT: stat/sonoff-1585B3/STATUS11 = {"StatusSTS":{"Time":"2019-05-19T15:50:47","Uptime":"0T00:00:43","Vcc":2.787,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":73,"Color":"BABABA","HSBColor":"0,0,73","Channel":[72,72,72],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":9,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

probonopd, 2019-05-19:
Thanks @Jason2866. Now I am puzzled. Will try with yet another hardware.

Jason2866, 2019-05-19:
Try a other GPIO. Take GPIO 12,13,14

probonopd, 2019-05-19:
Same with GPIO 14. I really have trouble understanding what is going on. How could an ESP8266 chip be "counterfeit", and how could a bad voltage regulator lead to the ESP not connect anymore (even after rebooting) once one has pressed "toggle"? Some clarification would help. Where can I read about this whole "counterfeit" thing?

Jason2866, 2019-05-19:
Please close since it is not a Tasmota issue.
Thx!

closed 2019-05-19 by ascillato2

probonopd, 2019-06-16:
Note to self:
Wemos D1 Mini with Tasmota 6.4.1 and WS2812B on D4 is known to work.

https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-WS2812B---WS2813B-Ledstrip

probonopd, 2019-06-17:
The exact same configuration is not working on Wemos Nodemcu 1.0 boards. Tried with several ones, including factory new.

These are the ones that are not working:

[image]

Jason2866, 2019-06-20:
Faked or bad ESP. I have this same looking board. Every sketch fails if you try to run at 160 Mhz.

jh83, 2019-09-04:
I'm having the same issue. Wifi works as expected until I toggle the Ws2812. Then it says that it tries to connect and fails. 

I have tested this on two different sonoff basic modules. Same issue on both. 

The bin used file is the precompiled one downloaded from the web. I have also tried to compile my self thru arduino and disable domotiz etc. 

Hardware is sonoff basic and gpio14 is configured as the pin for Ws2812. The same issue exists even if the led strip is not connected. 

I have also noticed that when the device reboots into ap mode, then the wlab of the device is not visible eventhou the device claims that it is on. So it feels like an issue with the board. 

WiFi and board is stable until I toggle the power2 on/off.

Did you ever manage to find a solution?

probonopd, 2019-09-04:
Me? I did not find a solution for the Wemos Nodemcu 1.0 boards but it does work for me with the Wemos D1 Mini (clone).

Jason2866, 2019-09-05:
It is "just" a power supply issue most of the time. Wifi fails on ESP device is power isnt good or clean.