arendst/Tasmota #5813: Max pwm channel with hue emulation for alexa

supermario1111, 2019-05-17:
Hi! I have some question with hue emulation.
I am using a simple esp07s module with Tasmota 6.5.0. I am using only the debug message for controlling my rollershutter system. 
Now it is working, but it is not the best solution. I have 4 logical output what I can controll trough alexa in simple on/off mode, and I can controll one channel (the last) with dimming. Now I can turn my 4 rollershutter independently in on/off state (full up and full down), and altogether with dimming (in that case everything goes in tha same position). 
So the question, can I have 4 or more channel with independent dimming settings? Not with color shading, but with 4 real different device. So if I ask alexa for example: set kitchen to 50, then I want to get an uart debug message only with the corresponding channel and dimming value. Now, if I ask it, it works, but I get all the time the last channel with the dimming value, and I can not get the right channel data. 
The other question, what is the maximum number of channel for this hue emulation? If I set more channel in the "Module parameters" page, then I can find only 4 with alexa (and only the last channel has dimming value). If I set more channel for PWM output, it works only with the last channel, so there is no two channel with dimming, but one with dimming, and with any other color shading.
I have attached a picture from my settings (older picture from phone). 
Thank you in advance!
Mario
[image]

s-hadinger, 2019-05-17:
Tasmota emulates a Hue Bridge but only supports 1 light. The light can have up to 5 PWM, for RGB color and Warm/Cold white. Your configuration is currently not supported.

It would be feasible to extend the Hue emulation code to emulate multiple lights, but the code increase would be significant.

Jason2866, 2019-05-17:
Maybe this fork is a solution https://github.com/stefanbode/Sonoff-Tasmota

closed 2019-05-19 by ascillato2