arendst/Tasmota #5814: Strange Switch Behavior 4CH Pro with 6.5.10.0

alfista2600, 2019-05-17:
Hi - I just got my first 4CH Pro and flashed it with 6.5.10.0.

Whenever I toggle a relay from the UI to ON, it toggles the relay before it (e.g. 1 if 2) to OFF a moment later. I only trigger ON commands and get these results in the console.

`21:04:18 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
21:04:18 RSL: stat/sonoff/POWER1 = ON
21:04:18 RSL: stat/sonoff/RESULT = {"POWER2":"OFF"}
21:04:18 RSL: stat/sonoff/POWER2 = OFF
21:04:22 RSL: stat/sonoff/RESULT = {"POWER2":"ON"}
21:04:22 RSL: stat/sonoff/POWER2 = ON
21:04:22 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
21:04:22 RSL: stat/sonoff/POWER1 = OFF
21:04:25 RSL: stat/sonoff/RESULT = {"POWER3":"ON"}
21:04:25 RSL: stat/sonoff/POWER3 = ON
21:04:25 RSL: stat/sonoff/RESULT = {"POWER2":"OFF"}
21:04:25 RSL: stat/sonoff/POWER2 = OFF
21:04:28 RSL: stat/sonoff/RESULT = {"POWER4":"ON"}
21:04:28 RSL: stat/sonoff/POWER4 = ON
21:04:28 RSL: stat/sonoff/RESULT = {"POWER3":"OFF"}
21:04:28 RSL: stat/sonoff/POWER3 = OFF`

Any ideas?

meingraham, 2019-05-17:
It's in hardware interlock mode. You need to change the dip switch settings.

Please close this issue.

alfista2600, 2019-05-17:
thanks, and closed

closed 2019-05-17 by alfista2600