arendst/Tasmota #5819: Setting HSBColor1 decreases Dimmer value if it is not set to 100

Medel, 2019-05-18:
Setting HSBColor1 decreases Dimmer value if it is not 100

`21:12:37 CMD: dimmer 100
21:12:37 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FE610000","HSBColor":"23,100,100","Channel":[99,38,0,0]}
21:12:53 CMD: HSBColor1 23
21:12:53 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF610000","HSBColor":"23,100,100","Channel":[100,38,0,0]}
21:12:53 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF610000","HSBColor":"23,100,100","Channel":[100,38,0,0]}
21:13:03 CMD: HSBColor1 22
21:13:03 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF5D0000","HSBColor":"22,100,100","Channel":[100,36,0,0]}
21:13:03 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF5D0000","HSBColor":"22,100,100","Channel":[100,36,0,0]}
21:13:27 CMD: dimmer 90
21:13:27 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":90,"Color":"E5530000","HSBColor":"22,100,90","Channel":[89,32,0,0]}
21:13:32 CMD: HSBColor1 23
21:13:32 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":89,"Color":"E2550000","HSBColor":"23,100,89","Channel":[88,33,0,0]}
21:13:32 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":89,"Color":"E2550000","HSBColor":"23,100,89","Channel":[88,33,0,0]}
21:13:37 CMD: HSBColor1 23
21:13:37 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":88,"Color":"E0550000","HSBColor":"23,100,88","Channel":[87,33,0,0]}
21:13:37 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":88,"Color":"E0550000","HSBColor":"23,100,88","Channel":[87,33,0,0]}
21:13:43 CMD: HSBColor1 20
21:13:43 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":87,"Color":"DC490000","HSBColor":"20,100,86","Channel":[86,28,0,0]}
21:13:43 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":87,"Color":"DC490000","HSBColor":"20,100,86","Channel":[86,28,0,0]}`

s-hadinger, 2019-05-18:
This is due to rounding errors. Brightness in internally stored as 0..255, whereas Dimmer is 0..100. There are conversion from one to the other that seem to round too aggressively. I will check if I can minimize these rounding errors.

ascillato2, 2019-05-19:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-19 by ascillato2

s-hadinger, 2019-05-19:
Please switch to 6.5.0.9 or greater, it has already been fixed.