arendst/Tasmota #5822: Bootloop with Sensair S8 (SAir)

toto79, 2019-05-19:
### BUG DESCRIPTION
...goes into boot loop if SAir Tx (64) and SAir Rx(65) is selected. It does not matter if the sensor is connected.Without mqtt it's ok, with mqtt device goes into bootloop

 
07:59:57 MQT: tele/SonNodeMCU/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x40244c34 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used Wemos D1 pro, NodeMCU V2, Sonoff S20 : _____
- [x] Tasmota binary firmware version number used : 6.5.11 / (pre-compiled)
- [x] Upload tools used : EspFlasher.exe
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
08:09:19 CMD: status 0
08:09:19 SRC: WebConsole from 192.168.178.114
08:09:19 RSL: Received Topic /status, Data Size 1, Data 0
08:09:19 RSL: Group 0, Index 1, Command STATUS, Data 0
08:09:19 RSL: stat/SonNodeMCU/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"SonNodeMCU","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:09:19 RSL: stat/SonNodeMCU/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff-sensors-4m.bin","RestartReason":"Exception","Uptime":"0T00:00:05","StartupUTC":"2019-05-19T06:09:14","Sleep":50,"CfgHolder":4617,"BootCount":100,"SaveCount":232,"SaveAddress":"3F4000"}}
08:09:19 RSL: stat/SonNodeMCU/STATUS2 = {"StatusFWR":{"Version":"6.5.0.11(292efcc-sensors)","BuildDateTime":"2019-05-18T20:12:33","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
08:09:19 RSL: stat/SonNodeMCU/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mesh2",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["10008009","280500000100000000000000000000000000","00000001"]}}
08:09:19 RSL: stat/SonNodeMCU/STATUS4 = {"StatusMEM":{"ProgramSize":569,"Free":3504,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDEE394","0007A3A4","B7FFBFCC","01DABBC0"]}}
08:09:19 RSL: stat/SonNodeMCU/STATUS5 = {"StatusNET":{"Hostname":"Son-NodeMCU","IPAddress":"192.168.178.63","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"80:7D:3A:69:2B:FC","Webserver":2,"WifiConfig":4}}
08:09:19 RSL: stat/SonNodeMCU/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.39","MqttPort":1883,"MqttClientMask":"SonNodeMCU","MqttClient":"SonNodeMCU","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:09:19 RSL: stat/SonNodeMCU/STATUS7 = {"StatusTIM":{"UTC":"Sun May 19 06:09:19 2019","Local":"Sun May 19 08:09:19 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:03","Sunset":"21:29"}}
08:09:19 RSL: stat/SonNodeMCU/STATUS10 = {"StatusSNS":{"Time":"2019-05-19T08:09:19"}}
08:09:19 RSL: stat/SonNodeMCU/STATUS11 = {"StatusSTS":{"Time":"2019-05-19T08:09:19","Uptime":"0T00:00:05","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Mesh2","BSSId":"34:81:C4:xx:xx:xx","Channel":6,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
08:09:19 RSL: stat/SonNodeMCU/STATUS11 = {"StatusSTS":{"Time":"2019-05-19T08:09:19","Uptime":"0T00:00:05","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Mesh2","BSSId":"34:81:C4:xx:xx:xx","Channel":6,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}}
08:09:19 MQT: Attempting connection...
08:09:20 MQT: Connected
08:09:20 MQT: tele/SonNodeMCU/LWT = Online (retained)
08:09:20 MQT: cmnd/SonNodeMCU/POWER = 
08:09:20 MQT: Subscribe to cmnd/SonNodeMCU/#
08:09:20 MQT: Subscribe to cmnd/sonoffs/#
08:09:20 MQT: Subscribe to cmnd/SonNodeMCU_fb/#
08:09:20 MQT: tele/SonNodeMCU/INFO1 = {"Module":"ESP-Motorshiel","Version":"6.5.0.11(292efcc-sensors)","FallbackTopic":"cmnd/SonNodeMCU_fb/","GroupTopic":"sonoffs"}
08:09:20 MQT: tele/SonNodeMCU/INFO2 = {"WebServerMode":"Admin","Hostname":"Son-NodeMCU","IPAddress":"192.168.178.63"}
08:09:20 MQT: tele/SonNodeMCU/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x40244c34 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
flash a new Device, activate mqtt connection to broker, select Gpio 1 to SAir TX (64) and Gpio3 to SAir Rx (65), click save

### EXPECTED BEHAVIOUR

no bootloops and show measuring

### ADDITIONAL CONTEXT
It doesn't matter if i select other gpio than 1&3 , i also tried gpio 12&14.
If mqtt is not activated measuring in WebIf is ok, but i try to set weblog 4, the device is rebooting. I think there is something wrong with the SAir plugin. I use a basic/cheap S8 with only co2 measuring.

ok:
20190320-161207-8782053-sonoff-sensors-4m.bin works

not ok:
20190323-191239-dcc69bd-sonoff-sensors-4m.bin - measuring is ok, but `after` some seconds device will reboot

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-05-19:
Verified. Solving.

toto79, 2019-05-19:
works for me. 
thanks.

closed 2019-05-19 by toto79

itarika2, 2020-04-19:
I have the same bug — bootloop on NodeMCU v2 with Senseair S8, but Fatal exception is 29:

`MQT: tele/AirMonitor1/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000e1b2","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40285b10","401013ac","4026c2c7","40277378","4026c2bb","40259be3","4025d9d3","40278566","4027a07a","4025f99b","4025f56c","4025f0ec","40276f50","40251181","4025f5a7","40261518","402614dc","4025f55c","4025f530","4025f4fb","40252368","4025fa69","4026fd79","4026fdbe","40246c34","40000f49","40000f49","40000f49","40000e19","401058a9","401058af"]}}`

Status 0:
`15:55:23 MQT: stat/AirMonitor/STATUS = {"Status":{"Module":18,"FriendlyName":["AirMonitor"],"Topic":"AirMonitor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:55:23 MQT: stat/AirMonitor1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:04:57","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":58,"BCResetTime":"2020-04-10T21:20:59","SaveCount":269,"SaveAddress":"FA000"}}
15:55:23 MQT: stat/AirMonitor/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(0396892-sensors)","BuildDateTime":"2020-04-10T15:01:54","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"374/699"}}
15:55:23 MQT: stat/AirMonitor/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SMRTHM12",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:55:23 MQT: stat/AirMonitor/STATUS4 = {"StatusMEM":{"ProgramSize":603,"Free":400,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16400E","FlashMode":3,"Features":["00000809","8FDAA387","0415A3A4","B7FFBFCD","01DA9BC4","24367CC7","00000002"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
15:55:23 MQT: stat/AirMonitor/STATUS5 = {"StatusNET":{"Hostname":"AirMonitor","IPAddress”:"192.168.0.183","Gateway”:"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer”:"192.168.0.1","Mac”:”12:12:12:12:12:12","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:55:23 MQT: stat/AirMonitor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.83","MqttPort":1883,"MqttClientMask":"AirMonitor","MqttClient":"AirMonitor","MqttUser”:"user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:55:23 MQT: stat/AirMonitor/STATUS7 = {"StatusTIM":{"UTC":"2020-01-10T14:55:23","Local":"2020-01-10T15:55:23","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+01:00","Sunrise":"08:41","Sunset":"17:13"}}
15:55:23 MQT: stat/AirMonitor/STATUS10 = {"StatusSNS":{"Time":"2020-01-10T15:55:23","S8":{"CarbonDioxide":768}}}
15:55:23 MQT: stat/AirMonitor/STATUS11 = {"StatusSTS":{"Time":"2020-01-10T15:55:23","Uptime":"0T00:04:57","UptimeSec":297,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId”:"user","BSSId":”11:11:11:11:11:11","Channel":1,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:06"}}}
15:55:23 MQT: stat/AirMonitor/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000e1b2","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40285b10","401013ac","4026c2c7","40277378","4026c2bb","40259be3","4025d9d3","40278566","4027a07a","4025f99b","4025f56c","4025f0ec","40276f50","40251181","4025f5a7","40261518","402614dc","4025f55c","4025f530","4025f4fb","40252368","4025fa69","4026fd79","4026fdbe","40246c34","40000f49","40000f49","40000f49","40000e19","401058a9","401058af"]}}`

This module has correct date, and works fine with other sensors.

itarika2, 2020-04-20:
After few self reboots need force hardware reboot.
Here you are more exceptions:

`MQT: stat/AirMonitor1/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4025d897","00000000","00000000"],"EXCVADDR":"00000019","DEPC":"00000000","CallChain":["40259be3","4025d9d3","40278566","4025f99b","4025f56c","4025f0ec","40276f50","401010ee","402511c0","4025f5a7","402521a5","40261518","40252258","402614dc","4025f55c","4025f530","4025f4fb","40252394","4025fa69","4026fd79","4026fdbe","40246c34","40000f49","40000f49","40000f49","40000e19","401058a9","401058af","4010000d","4026880c","402687bd"]}}`

`MQT: tele/AirMonitor1/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4025d897","00000000","00000000"],"EXCVADDR":"00000019","DEPC":"00000000","CallChain":["40259be3","4025d9d3","40278566","4025f99b","4025f56c","4025f0ec","40276f50","40251181","4025f5a7","40261518","402614dc","4025f55c","4025f530","4025f4fb","4025fa69","4026fd79","4026fdbe","40246c34","40000f49","40000f49","40000f49","40000e19","401058a9","401058af","4010000d","4026880c","402687bd","4024d22e","4027f5f1","4027f5f1","4027f5f2"]}}`

arendst, 2020-04-20:
Cannot reproduce