arendst/Tasmota #5825: Problem with Core 2.3.0 and Device with 2MB Flash

ihavenonick, 2019-05-19:
with Core 2.3.0 not OK
[image]

with Core 2.5.1 OK
[image]

Jason2866, 2019-05-19:
Issue template is missing. Anyway 2MB flash is NOT supported from Tasmota.
Tasmota uses always 1M flash no SPIFFS for all devices. This is needed and done for OTA upgrade possibility for all devices with Tasmota.

ihavenonick, 2019-05-19:
It worked the whole time and works with the master version as well. Thus, this statement is not correct

Jason2866, 2019-05-19:
You are wrong take a look https://github.com/esp8266/Arduino/tree/09826c6d87e54c15422da90f060377c9ef188947/tools/sdk/ld

ihavenonick, 2019-05-19:
Look at the pictures, as you can see that it works who the 2.5.1 Core at the dev. Version takes, or the master version 6.5.0

ihavenonick, 2019-05-19:
[image]

Jason2866, 2019-05-19:
Yes because 2.5.1 is a DIFFERENT NEWER core and it does support.
Core 2.3.0 does not.

ihavenonick, 2019-05-19:
and why does the Screnshot say something else?

Jason2866, 2019-05-19:
Because you have used a manually builded linker script. I did not say it isnt possible.
It isnt official supported.

Jason2866, 2019-05-19:
Please close this. It isnt a issue. It is as wanted and designed.

ihavenonick, 2019-05-19:
OK, I think it's wanted to use the 2.5.1 core. Too bad that the current dev. Versions that do not contain anymore.

closed 2019-05-19 by ihavenonick

Jason2866, 2019-05-19:
You can still use the NEVER official 2M0 linker script that you have used for building the version
you have made before. It is a setup of your build environment. It is in NO way a Tasmota sofware thing.
The 2M0 linker script was never included in Tasmota.

ihavenonick, 2019-05-19:
Yes, it just does not work anymore.
I will just use the 2.5.1 core, so it works as it had previously worked with the 2.3.0.