arendst/Tasmota #5826: POLL: Remove Belkin emulation in favour of Hue emulation only

arendst, 2019-05-19:
As suggested here (#5672) the need for Belkin emulation seems to be obsolete as Hue emulation can do the same and more.

Everyone in favour of removing Belkin support gives a thumbs up. Who wants to keep belkin support (which doesn't work at the moment) gives a thumbs down.

Pls feel free to respond.

ascillato, 2019-05-19:
I agree on removing Belkin Emulation also because this Emulation is not complete. It doesn't inform its changes, it only receives the commands as informed in #1700

altelch, 2019-05-19:
It's time to give up some old code ;-)

s-hadinger, 2019-05-19:
Quick test shows it's 4.4kb less code without Belkin emulation.

arendst, 2019-05-20:
OK. 

I'll start unwinding the hue and the wemo code and make both wemo and hue optional by defines. That makes removing wemo easy while keeping compatibility as long as possible.

blakadder, 2019-05-20:
i don't even know what belkin is :D

arendst, 2019-05-20:
With the latest dev release the define USE_EMULATION has been replaced by USE_EMULATION_WEMO and USE_EMULATION_HUE. Both can be enabled and/or disabled.

This allows for removing the Belkin Wemo code for those who do not need it while legacy users may still use it.

In the future Belkin Wemo might be dropped unless pressing reasons inhibit it ;-)

Jason2866, 2019-05-20:
Thx, 4k code saved for other nice stuff :-)
[image]
[image]

znanev, 2019-05-20:
Excuse me for the off-topic, but is there a way to present the device in Hue Bridge emulation mode as switch to Alexa (the way that Wemo emulation did)?

I just tried the latest firmware with emulation set to Hue Bridge and Alexa "sees" all devices as lights. And I can't find a way (from the Alexa app at least) to change the device type to switch.

Anyway, I gave my voice in favour of removing Belkin Wemo support. The wrong type of the device in Alexa is just an insignificant nuisance, I can live with that :)

s-hadinger, 2019-05-20:
Not possible as far as I know. The Hue protocol is currently exclusively for lights. Unless Philips extends it.

wongnam, 2019-05-21:
Removing the wemo, I don't regret that Hue bridge was able to replace it much better. however, how wonderful it is if you have an Apple homekit, I'm joking, LOL.

arendst, 2019-05-22:
Thx for your input. 

As you may have noticed the latest version allows for disabling Wemos (or Hue) individually. For now both stay enabled as my test Node-Red simulation still works with Wemos.

It might be getting disabled in the (near) future considering the amount of responses though.

closed 2019-05-22 by arendst

Magic73, 2020-08-31:
Sorry, but why HUE should be better than Wemo?
It's ok for you to say "Alexa, turn off all the lights" and it turn off event the non-light devices?
or thing the opposite: "Alexa, turn on all the ligthts" and it turn on your Sprinkles Irrigation ??

I would like to extend WEMO emu to a multi device like Sonoff CH 4...

dev-inciter, 2020-11-04:
I really want wemo emulation to remain available. Hue only supports lights to this day but not everything ist a light bulb. When I say "Alexa, turn of the lights" or "Alexa, turn of all lights" I don't want Alexa to turn off my computer or any other none light device. This is not possbile with only Hue emulation.

s-hadinger, 2020-11-04:
This poll is long expired and Wemo was finally kept.