arendst/Tasmota #5829: bmp280 and bh1750 + Node MCU

echo4019, 2019-05-20:
Hi, I'm trying to build a weather sensor using bmp280, bh1750 + Node MCU. 

I'm using Tasmota 6.5 and setting the module type to 18 generic.

I have the BMP280 connected to 3v3, GND, D1 & D2. The BH1750 is connected to VIN, GND, D3 & D4. 

The problem i am having is that I am only receiving values from one sensor. I relise these are both I2C so is there something i need to do differently?

[image]

joba-1, 2019-05-20:
I2C is a bus. All I2C devices connect to it. Physically attach SDA and SCL
of both sensors to the same 2 pins, e.g. D1 and D2.
-- 
iphone so i typo

echo4019, 2019-05-20:
Ah that makes sense. 

Thanks for the help!

Jason2866, 2019-05-20:
Please close. Thx

closed 2019-05-20 by ascillato2