arendst/Tasmota #5833: ESP Chip id in MQTT message

SargonofAssyria, 2019-05-21:
Could 'ESP Chip id" also be present is MQTT STATUS4?

curzon01, 2019-05-21:
except STATUS4 shows memory info and IMHO might be the wrong place for it:
use [last three bytes of MAC](https://github.com/esp8266/Arduino/issues/921l)

--
Please address this to [Support's Community](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md#supports-community).
The community channels are a better and more dynamic for helping you.
Github issues are mainly used for Tasmota software feature requests and bug reporting. 

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Jason2866, 2019-05-21:
For what is this info needed? Please describe the use case needed for.
There is only a chance to be considered for implementation with a good reason.

closed 2019-05-21 by ascillato2