arendst/Tasmota #5834: 6.5.0.11 wifi issue

sylvaing, 2019-05-21:
Hi, I use tasmota for many devices, sonoff basic and RF and shelly 1.
Currently I'm on Tasmosta 6.5.0.1 with core 2.3 due to many test with my wifi, Now it's ok!!

But I received **Shelly 1 V3**, I flashed this with 6.5.0.11 from thehackbox with core 2.3.
At th beginning all i's OK, but after my short configuration Shelly loop on disconnect from wifi every few second:

** Exemple:
64 bytes from 192.168.0.126: icmp_seq=3462 ttl=255 time=9.760 ms
Request timeout for icmp_seq 3463
Request timeout for icmp_seq 3464
Request timeout for icmp_seq 3465
Request timeout for icmp_seq 3466
Request timeout for icmp_seq 3467
Request timeout for icmp_seq 3468
Request timeout for icmp_seq 3469
64 bytes from 192.168.0.126: icmp_seq=3470 ttl=255 time=174.068 ms
Request timeout for icmp_seq 3471
Request timeout for icmp_seq 3472
Request timeout for icmp_seq 3473
Request timeout for icmp_seq 3474
Request timeout for icmp_seq 3475
Request timeout for icmp_seq 3476
Request timeout for icmp_seq 3477
Request timeout for icmp_seq 3478
64 bytes from 192.168.0.126: icmp_seq=3479 ttl=255 time=8.005 ms

and from my syslog server:

2019-05-21 | 08:12:52 | Notice | 192.168.0.126 | user | chevet-sylvain | ESP-APP: (UTC) Tue May 21 06:10:46 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
2019-05-21 | 08:12:51 | Notice | 192.168.0.126 | user | chevet-sylvain | ESP-HTP: Web server active on chevet-sylvain with IP address 192.168.0.126
2019-05-21 | 08:12:51 | Notice | 192.168.0.126 | user | chevet-sylvain | ESP-WIF: Connected
2019-05-21 | 08:12:44 | Notice | 192.168.0.126 | user | chevet-sylvain | ESP-APP: (UTC) Tue May 21 06:10:39 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
2019-05-21 | 08:12:44 | Notice | 192.168.0.126 | user | chevet-sylvain | ESP-HTP: Web server active on chevet-sylvain with IP address 192.168.0.126
2019-05-21 | 08:12:44 | Notice | 192.168.0.126 | user | chevet-sylvain | ESP-WIF: Connected
2019-05-21 | 08:12:36 | Notice | 192.168.0.126 | user | chevet-sylvain | ESP-HTP: Web server active on chevet-sylvain with IP address 192.168.0.126
2019-05-21 | 08:12:36 | Notice | 192.168.0.126 | user | chevet-sylvain | ESP-WIF: Connected

When I shutdown my WIFI AP, tasmota connect to this just for 30s, and restart in "loop restart".

SO I Try, 6.5.0.11 with core 2.4 => it's the same thing
I try 6.5.0.11 with core 2.5.1 => it's the same thing
I try to downgrade to 6.5.0.1 with core 2.3 => all it's ok!!!

All flash procedure have been completed by OTA, and not with serial flash.

I think that I d'int have special config I just configure:
Wifi ( of course) with my hostname (chevet-sylvain)
configuration module -> shelly 1 (46)
configuration MQTT ->
   -HOST -> my ip of my mqtt server
   - my user
   - my password
   -change topic to: chevet-sylvain
   -change full topic from %prefix%/%topic%/ to %topic%/%prefix%/
syslog host and syslog level to my syslog server
others -> Friendly name 1 to : chevet-sylvain

It's seems, but I'm not sure that issue came when I change Friendly name

So for the moment I stay on 6.5.0.1 with 2.3. But I prefer to talk to you in order to improve this.

Thank you for all.

Jason2866, 2019-05-21:
Yes, this is known behaviour and not under the control of Tasmota. It is -you wrote correct- core related Cores are not maintainend from Tasmota. Tasmota just uses cores. 
See https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#what-is-this-arduino-core
You have the choice of using the core which works best for you

Please close this issue because nothing can be done in Tasmota (code) to change this.

Jason2866, 2019-05-21:
Tried your config. works without any issues
[image]
Ping wird ausgeführt für 192.168.2.149 mit 32 Bytes Daten:
```
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=4ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128
Antwort von 192.168.2.149: Bytes=32 Zeit=3ms TTL=128

Ping-Statistik für 192.168.2.149:
    Pakete: Gesendet = 23, Empfangen = 23, Verloren = 0
    (0% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 3ms, Maximum = 4ms, Mittelwert = 3ms
```

ascillato2, 2019-05-21:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-21 by ascillato2

sylvaing, 2019-05-21:
@Jason2866 thanks for your test.
In your fist reply, my problem isn't core version, because all is ok with **6.5.0.1** core 2.3.0. but isn't on **6.5.0.11** core 2.3.0.
For mor information I tested this on two Shelly 1 v3, and I have same issue.

@ascillato2 ok for closing. If I retested this I'll try to have more information. IfI found some times.. :-D I will retest this.

Jason2866, 2019-05-21:
@sylvaing  i have tested latest 6.5.0.11 with core 2.3.0 too and have no problems.
My Shelly1 v3 is running since a few days without any issue 
[image]