arendst/Tasmota #5837: Tasmota resets configuration when GPIO12 is set to COUNTER1

llagendijk, 2019-05-21:
### BUG DESCRIPTION
On my NodeMCU everything works graet with PWM and I2C.
When I add GPIO12 as COUNTER1 and save is pressed, the module reboots but all configuration is cleared: no counter1 but also other configuration is gone

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : nodeMCU
- [X] Tasmota binary firmware version number used :  6.5.0.11/ self-compiled from todays checkout)
- [X] Development IDE - Compiler / Upload tools used : platformIO 3.6.7 / OTA
- [X] Provide the output of command ``status 0`` :
```17:48:42 MQT: stat/sonoff_afzuiger/STATUS = {"Status":{"Module":18,"FriendlyName":["afzuiger"],"Topic":"sonoff_afzuiger","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:48:42 MQT: stat/sonoff_afzuiger/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domoticz.home.fazant.net/tasmota/sonoff-ownconfig.bin","RestartReason":"Software/System restart","Uptime":"0T01:21:38","StartupUTC":"2019-05-21T15:27:04","Sleep":50,"CfgHolder":4617,"BootCount":110,"SaveCount":2417,"SaveAddress":"F7000"}}
17:48:42 MQT: stat/sonoff_afzuiger/STATUS2 = {"StatusFWR":{"Version":"6.5.0.11(sonoff)","BuildDateTime":"2019-05-21T17:18:56","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
17:48:42 MQT: stat/sonoff_afzuiger/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"loghost.home.fazant.net","LogPort":514,"SSId":["Pheasant3",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
17:48:42 MQT: stat/sonoff_afzuiger/STATUS4 = {"StatusMEM":{"ProgramSize":449,"Free":552,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","00986384","041503A1","20001410","012080C0"]}}
17:48:42 MQT: stat/sonoff_afzuiger/STATUS5 = {"StatusNET":{"Hostname":"sonoff_afzuiger-4882","IPAddress":"192.168.158.46","Gateway":"192.168.158.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.158.2","Mac":"DC:4F:22:88:F3:12","Webserver":2,"WifiConfig":4}}
17:48:42 MQT: stat/sonoff_afzuiger/STATUS6 = {"StatusMQT":{"MqttHost":"domoticz.home.fazant.net","MqttPort":1883,"MqttClientMask":"Afzuiger","MqttClient":"Afzuiger","MqttUser":"afzuiger","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:48:42 MQT: stat/sonoff_afzuiger/STATUS7 = {"StatusTIM":{"UTC":"Tue May 21 16:48:42 2019","Local":"Tue May 21 17:48:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
17:48:42 MQT: stat/sonoff_afzuiger/STATUS10 = {"StatusSNS":{"Time":"2019-05-21T17:48:42","BME280":{"Temperature":31.5,"Humidity":22.8,"Pressure":1017.8},"PressureUnit":"hPa","TempUnit":"C"}}
17:48:42 MQT: stat/sonoff_afzuiger/STATUS11 = {"StatusSTS":{"Time":"2019-05-21T17:48:42","Uptime":"0T01:21:38","SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":85,"Fade":"ON","Speed":10,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Pheasant3","BSSId":"0E:EC:DA:8A:7A:15","Channel":1,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```May 21 17:27:38 sonoff_afzuiger-4882 ESP-MQT: stat/sonoff_afzuiger/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"loghost.home.fazant.net","LogPort":514,"SSId":["Pheasant3",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
May 21 17:27:38 sonoff_afzuiger-4882 ESP-MQT: stat/sonoff_afzuiger/STATUS4 = {"StatusMEM":{"ProgramSize":449,"Free":552,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","00986384","041503A1","20001410","012080C0"]}}
May 21 17:27:38 sonoff_afzuiger-4882 ESP-MQT: stat/sonoff_afzuiger/STATUS5 = {"StatusNET":{"Hostname":"sonoff_afzuiger-4882","IPAddress":"192.168.158.46","Gateway":"192.168.158.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.158.2","Mac":"DC:4F:22:88:F3:12","Webserver":2,"WifiConfig":4}}
May 21 17:27:38 sonoff_afzuiger-4882 ESP-MQT: stat/sonoff_afzuiger/STATUS6 = {"StatusMQT":{"MqttHost":"domoticz.home.fazant.net","MqttPort":1883,"MqttClientMask":"Afzuiger","MqttClient":"Afzuiger","MqttUser":"afzuiger","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
May 21 17:27:38 sonoff_afzuiger-4882 ESP-MQT: stat/sonoff_afzuiger/STATUS7 = {"StatusTIM":{"UTC":"Tue May 21 15:27:38 2019","Local":"Tue May 21 16:27:38 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
May 21 17:27:38 sonoff_afzuiger-4882 ESP-MQT: stat/sonoff_afzuiger/STATUS10 = {"StatusSNS":{"Time":"2019-05-21T16:27:38","BME280":{"Temperature":30.9,"Humidity":24.8,"Pressure":1017.4},"PressureUnit":"hPa","TempUnit":"C"}}
May 21 17:27:38 sonoff_afzuiger-4882 ESP-MQT: stat/sonoff_afzuiger/STATUS11 = {"StatusSTS":{"Time":"2019-05-21T16:27:38","Uptime":"0T00:00:34","SleepMode":"Dynamic","Sleep":0,"LoadAvg":100,"POWER":"ON","Dimmer":85,"Fade":"OFF","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Pheasant3","BSSId":"0E:EC:DA:8A:7A:15","Channel":1,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:05"}}}
May 21 17:27:39 sonoff_afzuiger-4882 ESP-MQT: tele/sonoff_afzuiger/STATE = {"Time":"2019-05-21T16:27:39","Uptime":"0T00:00:35","SleepMode":"Dynamic","Sleep":0,"LoadAvg":128,"POWER":"ON","Dimmer":85,"Fade":"OFF","Speed":5``

```

### TO REPRODUCE
- Configure module as usual.
- Add GPIO12 as COUNTER1
- press save
- Module reboots as usual but returns with empty configuration (all GPIO settings are gone)

### EXPECTED BEHAVIOUR
Configurations set shall survive a restart

llagendijk, 2019-05-21:
I forgot to mention that the same configuration worked until I compiled a tasmota version with the ADC changes. This may be coincidence though

arendst, 2019-05-22:
Casued by core 2.5.2 which is strict on Interrupt Service Routine (ISR) location. 

In this case it is run from flash where core 2.5.2 wants it to be in (scarce) IRAM. 

Short term solution is do not use core 2.5.2.
Longer term solution is that I have to reconsider my ISR's and add defines to switch them to IRAM starting with core 2.5.x

Thx for testing.

Jason2866, 2019-05-22:
@arendst  have you seen this https://github.com/esp8266/Arduino/issues/6087#issuecomment-492015020 ?

arendst, 2019-05-22:
@Jason2866 No I hadn't but as later cores provide more IRAM anyway I now enabled the use of IRAM in cases needed for all cores above 2.3.0

llagendijk, 2019-05-22:
Seems to work ok, but get the following after a restart:

10:27:08 MQT: tele/sonoff_afzuiger/LWT = Online (retained)
10:27:08 MQT: cmnd/sonoff_afzuiger/POWER = 
10:27:08 MQT: tele/sonoff_afzuiger/INFO1 = {"Module":"Generic","Version":"6.5.0.12(sonoff)","FallbackTopic":"cmnd/Afzuiger_fb/","GroupTopic":"sonoffs"}
10:27:08 MQT: tele/sonoff_afzuiger/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff_afzuiger-4882","IPAddress":"192.168.158.46"}
10:27:08 MQT: tele/sonoff_afzuiger/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40207b7c epc2:0x00000000 epc3:0x400043d0 excvaddr:0x00000000 depc:0x00000000"}
10:27:08 MQT: stat/sonoff_afzuiger/RESULT = {"POWER":"ON"}

Do I need to worry about the fatal exception?

arendst, 2019-05-22:
How often do you get the exception(s)?

llagendijk, 2019-05-22:
After a restart 1 the module mostly failed: the fan (on PWM1) returned to its previous setting, but the WIFI never came up. I now erased the module and reflashed it.  It now works, sort of. It keeps rebooting:
00:00:00 CFG: Loaded from flash at F5, Count 15
00:00:00 Project sonoff Sonoff Version 6.5.0.12(sonoff)-2_5_2
00:00:00 WIF: Connecting to AP1 Pheasant3 in mode 11N as sonoff_afzuiger-4882...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff_afzuiger-4882 with IP address 192.168.158.46
11:57:16 MQT: Attempting connection...
11:57:17 MQT: Connected
11:57:17 MQT: tele/sonoff_afzuiger/LWT = Online (retained)
11:57:17 MQT: cmnd/sonoff_afzuiger/POWER = 
11:57:17 MQT: tele/sonoff_afzuiger/INFO1 = {"Module":"Generic","Version":"6.5.0.12(sonoff)","FallbackTopic":"cmnd/DVES_88F312_fb/","GroupTopic":"sonoffs"}
11:57:17 MQT: tele/sonoff_afzuiger/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff_afzuiger-4882","IPAddress":"192.168.158.46"}
11:57:17 MQT: tele/sonoff_afzuiger/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40207b7c epc2:0x00000000 epc3:0x400043ee excvaddr:0x00000000 depc:0x00000000"}
11:57:17 MQT: stat/sonoff_afzuiger/RESULT = {"POWER":"ON"}
11:57:17 MQT: stat/sonoff_afzuiger/POWER = ON

arendst, 2019-05-22:
Just to backtrace, this only happens when counter is configured?

llagendijk, 2019-05-22:
Right it crashes on the counter. When I remove the counter from the configuration, the module does not restart anymore. I need to disconnect the counter to be able to reconfigure the module as otherwise it crashes before configuration changes can be saved

I was looking for a memory map in the platformio results so I could see where the code was crashing, but there does not seem to be one. Please let me know if you need more information from me.

B.t.w. I am really impressed  by your quick reactions. Much appreciated!

arendst, 2019-05-22:
If you compile yourself you might want to try this:

in file xsns_01_counter.ino change code to look like this:

```
#define XSNS_01             1

unsigned long last_counter_timer[MAX_COUNTERS]; // Last counter time in micro seconds

#ifndef ARDUINO_ESP8266_RELEASE_2_3_0       // Fix core 2.5.x ISR not in IRAM Exception
void CounterUpdate(uint8_t index) ICACHE_RAM_ATTR;
void CounterUpdate1(void) ICACHE_RAM_ATTR;
void CounterUpdate2(void) ICACHE_RAM_ATTR;
void CounterUpdate3(void) ICACHE_RAM_ATTR;
void CounterUpdate4(void) ICACHE_RAM_ATTR;
#endif  // ARDUINO_ESP8266_RELEASE_2_3_0

void CounterUpdate(uint8_t index)
{
  unsigned long counter_debounce_time = micros() - last_counter_timer[index -1];
  if (counter_debounce_time > Settings.pulse_counter_debounce * 1000) {
    last_counter_timer[index -1] = micros();
    if (bitRead(Settings.pulse_counter_type, index -1)) {
      RtcSettings.pulse_counter[index -1] = counter_debounce_time;
    } else {
      RtcSettings.pulse_counter[index -1]++;
    }

//    AddLog_P2(LOG_LEVEL_DEBUG, PSTR("CNTR: Interrupt %d"), index);
  }
}

void CounterUpdate1(void)
{
  CounterUpdate(1);
}
```
The goal here is to add function CounterUpdate to IRAM too.

llagendijk, 2019-05-22:
and that change solved my issue.  I will close the issue and wait for the push.

Thanks again for the very quick replies, much appreciated.

closed 2019-05-22 by llagendijk