arendst/Tasmota #5838: mDNS (#USE_DISCOVERY) consumes 23.5kb on core 2.5.2

s-hadinger, 2019-05-21:
### BUG DESCRIPTION
Not exactly a bug but an anomaly. I just discovered that compiling with mDNS option (#USE_DISCOVERY) on core_2_5_2 adds 23.5kB of code size instead of the 8kB on 2_4_2 or 2_3_0 - that's a lot!

I didn't have time yet to find out why. Opening to the hive mind.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : any
- [x] Tasmota binary firmware version number used : 6.5.0.10 self-compiled
- [x] Development IDE - Compiler / Upload tools used : Platform.io (Mac) 
- [ ] Provide the output of command ``status 0`` : not relevant
- [ ] Provide the output of console when you experience your issue if apply : not relevant

Jason2866, 2019-05-21:
Arduino team changed (added) mDNS code a lot since 2.4.2.
I think it is correct. Anyway mDNS isnt a good idea to use, does generate more problem than it solves.
Disable mDNS by default in my_user_config.h is imho the next logical step after switching it off by default.

s-hadinger, 2019-05-21:
Agreed. We should consider disabling by default with such a high impact on code size.
Closing the issue.

closed 2019-05-21 by s-hadinger