arendst/Tasmota #5839: Compile error with Arduino IDE 1.8.9 and core 2.5.2 or 2.5.1

peiperakos, 2019-05-21:
Hello Friends! Sorry to post here!

but i have a compile error in windows arduino ide 1.8.9 with core 2.5.1 and 2.5.2
i compile every time with arduino ide for months with out problem with tasmota for example with core 2.5.0 WORKS PERFECT!

the compile error i have is :

`In file included from C:\Program Files (x86)\Arduino\libraries\NeoPixelBus-2.2.9\src/NeoPixelBus.h:67:0,

                 from C:\Users\Statos r00t\Desktop\nodeMCU 1.0\IOT-Tasmota\Sonoff-Tasmota\Tasmota 6.5\Sonoff-Tasmota-development-6.5.0.12(ALARM)\sonoff\xplg_ws2812.ino:25:

C:\Program Files (x86)\Arduino\libraries\NeoPixelBus-2.2.9\src/internal/NeoEsp8266DmaMethod.h:51:127: error: declaration of C function 'void rom_i2c_writeReg_Mask(uint32_t, uint32_t, uint32_t, uint32_t, uint32_t, uint32_t)' conflicts with

     void rom_i2c_writeReg_Mask(uint32_t block, uint32_t host_id, uint32_t reg_add, uint32_t Msb, uint32_t Lsb, uint32_t indata);

                                                                                                                               ^

In file included from C:\Users\Statos r00t\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/esp8266_peri.h:25:0,

                 from C:\Users\Statos r00t\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Arduino.h:38,

                 from sketch\sonoff.ino.cpp:1:

C:\Users\Statos r00t\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/esp8266_undocumented.h:9:13: error: previous declaration 'void rom_i2c_writeReg_Mask(int, int, int, int, int, int)' here

 extern void rom_i2c_writeReg_Mask(int, int, int, int, int, int);

             ^`

can anyone help me to get compile with new cores ?

Jason2866, 2019-05-21:
I answered you in Discord. Please close.
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-21 by ascillato2

mortenmoulder, 2019-05-26:
@Jason2866 If you're going to write a comment like that, please at least give the response here as well. We can't use a private Discord conversation for anything, so please post the answer here as well.

Jason2866, 2019-05-26:
Issue was a individual setup problem.