arendst/Tasmota #5840: White command not found

websylv, 2019-05-21:
### BUG DESCRIPTION
The command white described in https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#light doesn't work 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Device used (i.e. Sonoff Basic) : H801
- [ X] Tasmota binary firmware version number used : 6.5 / (pre-compiled )
- [ X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

20:05:16 CMD: STATUS 0 
20:05:16 MQT: stat/ledSalon/STATUS = {"Status":{"Module":20,"FriendlyName":["ledSalon"],"Topic":"ledSalon","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:05:16 MQT: stat/ledSalon/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T01:11:42","StartupUTC":"2019-05-21T17:53:34","Sleep":50,"CfgHolder":4617,"BootCount":23,"SaveCount":104,"SaveAddress":"F4000"}}
20:05:16 MQT: stat/ledSalon/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:05:16 MQT: stat/ledSalon/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"nuc01.webhu.net","LogPort":514,"SSId":["b2f",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00028009","280500000100000000000000000000000000","00000200"]}}
20:05:16 MQT: stat/ledSalon/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
20:05:16 MQT: stat/ledSalon/STATUS5 = {"StatusNET":{"Hostname":"ledSalon","IPAddress":"192.168.99.212","Gateway":"192.168.99.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.99.10","Mac":"80:7D:3A:33:BB:51","Webserver":2,"WifiConfig":0}}
20:05:16 MQT: stat/ledSalon/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.webhu.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_33BB51","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:05:16 MQT: stat/ledSalon/STATUS7 = {"StatusTIM":{"UTC":"Tue May 21 19:05:16 2019","Local":"Tue May 21 20:05:16 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:01","Sunset":"20:32"}}
20:05:16 MQT: stat/ledSalon/STATUS10 = {"StatusSNS":{"Time":"2019-05-21T20:05:16"}}
20:05:16 MQT: stat/ledSalon/STATUS11 = {"StatusSTS":{"Time":"2019-05-21T20:05:16","Uptime":"0T01:11:42","Vcc":3.208,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":20,"Color":"25,51,0,0,0","HSBColor":"91,100,20","Channel":[9,20,0,0,0],"CT":499,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"b2f","BSSId":"C8:0E:14:12:CB:50","Channel":9,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [X ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

20:05:57 CMD: weblog 4
20:05:58 MQT: stat/ledSalon/RESULT = {"WebLog":4}
20:05:58 MQT: tele/ledSalon/STATE = {"Time":"2019-05-21T20:05:58","Uptime":"0T01:12:24","Vcc":3.210,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":20,"Color":"25,51,0,0,0","HSBColor":"91,100,20","Channel":[9,20,0,0,0],"CT":499,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"b2f","BSSId":"C8:0E:14:12:CB:50","Channel":9,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:07"}}
20:05:58 CFG: Saved to flash at FB, Count 105, Bytes 3584
20:06:08 WIF: Checking connection...
20:06:08 WIF: Connected
20:06:09 CMD: white 100
20:06:09 SRC: WebConsole from 192.168.99.117
20:06:09 RSL: Received Topic /white, Data Size 3, Data 100
20:06:09 RSL: Group 0, Index 1, Command WHITE, Data 100
20:06:09 MQT: stat/ledSalon/RESULT = {"Command":"Unknown"}
20:06:18 CMD: White 100
20:06:18 SRC: WebConsole from 192.168.99.117
20:06:18 RSL: Received Topic /White, Data Size 3, Data 100
20:06:18 RSL: Group 0, Index 1, Command WHITE, Data 100
20:06:18 MQT: stat/ledSalon/RESULT = {"Command":"Unknown"}

### TO REPRODUCE
just run White command

### EXPECTED BEHAVIOUR
turn on or off white led

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-05-21:
White command only works for RGBW devices (4 x PMW leds). Yours is 5 x PMW (RGBWC).

Use CT and Dimmer instead.

Jason2866, 2019-05-21:
There is no command `white 100`
Use `color 12`

websylv, 2019-05-21:
> White command only works for RGBW devices (4 x PMW leds). Yours is 5 x PMW (RGBWC).
> 
> Use CT and Dimmer instead.

problem is ct an dimmer don't turn on white led
y need to set first a color XXXXXXFF to turn on only the white led
is there any way to turn off the 5th pwm (i don't use)  to be able to turn on the white led only without have to play with color

websylv, 2019-05-21:
> There is no command `white 100`
> Use `color 12`

color 12 only work for rgb led not the w1 channel
and if a check the wiki normally there is a withe command : 

> 

White | 1..100 = set white channel brightness in single white channel lights : 
`
White | 1..100 = set white channel brightness in single white channel lights
`

s-hadinger, 2019-05-21:
For 5 x PMW devices, CT does turn on white LEDs, and turn off RGB leds.

'CT 153' for cold white.
'CT 500' for warm white.
Then use Dimmer 1..100 to adjust brightness.

If this doesn't work, let me know.

curzon01, 2019-05-22:
> and if a check the wiki normally there is a withe command :
> 
> `White | 1..100 = set white channel brightness in single white channel lights`

as described in wiki `White` command works for single white channel lights only like AiLight... (RGBW devices). In general, commands not supported by a device will be displayed as `{"Command":"Unknown"}`

closed 2019-05-23 by ascillato2