arendst/Tasmota #5841: SGP30 Abs Humidity not working

TONGXD, 2019-05-22:
SGP30 Abs Humidity always shows 0.0000g/m³,no change!
Sonoff-Tasmota 6.5.0.12

Jason2866, 2019-05-22:
Issue template?

gemu2015, 2019-05-22:
you need a valid temperature AND relative humidity sensor to get absolute humidity from the sgp30.

blakadder, 2019-05-22:
Maybe a short write up on the SGP30 use in tasmota on the wiki would help in the future?

gemu2015, 2019-05-22:
yes this would help. 
the problem however is that there is only one variable named global_update not showing if both needed variables are present and updated. better would be an indication of each global variable separately  then the sgp30 would only show a value when both needed globals were set.

arendst, 2019-05-22:
@gemu2015 you're right but for the moment just test for a non-zero number as global humidity as zero humidity is highly unlikely.

ascillato2, 2019-05-22:
Closing this issue as it is solved. Thanks

closed 2019-05-22 by ascillato2