arendst/Tasmota #5847: HTTP OPTIONS requests require authorization

tiberiuana, 2019-05-22:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION

The web server does not allow unauthenticated OPTIONS requests.

When making a CORS request to a URL with login required, browsers first perform an OPTIONS preflight *without credentials*. This is expected behaviour, as per the spec: (https://fetch.spec.whatwg.org/#preflight-request): 

    For a CORS-preflight request, request’s credentials mode is always "omit", 

### REQUESTED INFORMATION

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [NA] Device used (i.e. Sonoff Basic) : _____
- [NA] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [NA] Development IDE - Compiler / Upload tools used : ____ / ____
- [NA] Provide the output of command ``status 0`` :

### TO REPRODUCE
_Steps to reproduce the behavior:_

Make an `OPTIONS` request to any URL that requires login:

`curl -v http://mysonoff/cs -X OPTIONS`

This fails with `401 Unauthorized`.

Side issue to consider: when authorized, the `OPTIONS` request seems to return the full page, just like a `GET` request. Without authorization, this should of course not happen. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

200 OK response with correct CORS headers -- and empty body.

**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2019-05-23:
For the describe request Tasmota has to implement cross-origin resource sharing (CORS). So it might be an enhancment request due to (CORS) is one possible (and yes, a common implemented feature in all common browser engines) solution for cross origin requests - alternative could be e. g. JSONP (but that's limited).
I can't estimate the workload and resource requirement but keep in mind it's an ESP not a Xeon with 32GB - for those who needs it would be a alternative to use a proxy in front?

tiberiuana, 2019-05-23:
My use case for needing this: I want to be able to talk to Tasmota devices directly (without needing a separate MQTT server or running a separate app) from a webpage (PWA app). 

I know I can get the status and send commands using `/cm` but, as far as I can tell, it's not possible to also get the weblog through a command. I would prefer having access to the log so I could limit polls to ~60s instead of hammering the webserver to catch all state changes.

Looking at the code, Tasmota already includes code for CORS preflight -- but only for `/u2` (firmware upload) -- it would be useful to also have it for `/cs`.

I'll send a PR, it seems that covering the console would only take one line of code.

PR #5852 (2019-05-23): Add CORS preflight support to console URL
Add CORS preflight support to `/cs` URL to allow access to the log from browser apps.

## Description:

**Related issue (if applicable):** fixes #5847

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched (Also remember to update _changelog.ino_ file)
  - [x] Only one feature/fix was added per PR.
  - [ ] The code change is tested and works.
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonof [... cut]
closed 2019-05-23 by arendst