arendst/Tasmota #5849: Hue Emulation missbehavior

chillje, 2019-05-23:
Hello,
I run 3x "Sonoff S20 plugs" and 6x Shelly1 with tasmota.
The " Sonoff S20 Plugs " are running with "Hue Emu", the Shelly1 with "Belkin WeMo".
All are learned and used with Amazons "Alexa". 

Since yesterday the Sonoff devices don't react correctly anymore when using Alexa commands. Only one Sonoff device reacts to the voice commands for all three Sonoff devices. 

Short scenario:
Alexa, switch on Sonoff1 -> Sonoff 1 switches on
Alexa, switch off Sonoff2 -> Sonoff 1 switches off
Alexa, switch on Sonoff3 -> Sonoff 1 switches on

I have already removed all Sonoff devices from Alexa and added them again.
Nothing has changed in the behaviour, only Sonoff2 now reacts to all commands for the different Sonoff devices.

I updated Sonoff1 via tasmota OTA via the website to a new tasmota version, but that didn't help either.

Changing Sonoff1 from "Hue Emu" to "Belkin WeMo" was not possible. Alexa doesn't find the device with the Belkin emulation.

I guess there is a problem with the "Hue Emulation". 
Maybe an update in the protocol that Alexa is already aware of?

ihavenonick, 2019-05-23:
Same Problem here, I think Amazon have something changes on the HUE detection.
This problem ist exist with all Tasmota versions wen use the HUE emulator.

ascillato2, 2019-05-23:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

wernerk63, 2019-05-23:
Over 20 Sonoff Devices,
works perfect with the Workarounds (Hue brightness) before, 
and then with Master 6.5.0.0 since publish for over 45 days.

now - same Problem with HueBridgeEmu after Alexa search Devices, since ca. 24h in Germany.

2x 4chpro and 2x T1EU with HueBridgeEmu and Tasmota Master 6.5.0.0 reacts false on Alexa.
Last night i do some Troubleshooting, it seams that Alexa only adresses/recognize one Hue Bridge.
With one HueBridgeEmu all is working, after adding the second one,  a false Device switch..

The weblog's (Level 4) shows:
1. Alexa Search Devices, found and enumarate Hue Light's :
    /LastcharMac/light1 to light4  eg. light2 for each/all  Devices  
    (all correct, with different 'LastCharMac'/light x)

2. After adding the Second Device with HueBridgeEmu  and switch a 'Light' a wrong Device switch...
    Weblog of the wrong switched Device show's the 'LastcharMac'/light x  of the wrong Device,
    and not the 'HueAdress' (LastCharMac/light x) of the per VoiceCommand adressed Device

Maybe, it should works further, if i had not do a 'Search Devices'..... ( because of NetworkChanges)
I think, there is something changes on the Alexa HUE Support.

Max0412, 2019-05-23:
Same problem here, i´ve added 2x H801 and enabled the Hue Emulation. 
Both are found in the Alexa App but they both only switch the same H801.

s-hadinger, 2019-05-23:
This is strange, when you manually change the configuration of the light via the Tasmota console, the change is well reflected in Alexa. However when sending the commands from Alexa, it looks like it's sending UDP packets to try to discover which Hue bridge it should send to.

Will need to add more logs to troubleshoot.

rherriage, 2019-05-24:
I'm running into the same issue as well.  I noticed it this morning.  I'm running 6 x Sonoff ifan02s with Tasmota 6.5.  The status will change if I manually change the lights/fans on/off, but the app nor verbal commands with an echo will turn them off/on.

ascillato, 2019-05-24:
Please, complete the troubleshooting template in order to have more debug information. Thanks

Max0412, 2019-05-24:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Phillips Hue Emulation activated for 2 H801 with LED Strips
Phillips Hue Emulation activated for 1 Shelly1 as I/O Device

The H801 devices are both found in Alexa (Fernsehtisch and COUCH), but both(in-app and with voice command) just switch COUCH in the alexa app. Alexa does seem to recognize the current state/color of both H801 individually. - COUCH was set up first. -
The Shelly 1 does switch with voice command but doesn´t in-app. It does recognize the current state of the Shelly.

STATUS 0 of COUCH

`20:56:56 MQT: stat/Wohnzimmer/STATUS = {"Status":{"Module":20,"FriendlyName":["COUCHLED"],"Topic":"Wohnzimmer","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:56:56 MQT: stat/Wohnzimmer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"1T01:15:15","StartupUTC":"2019-05-23T18:41:41","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":109,"SaveAddress":"F7000"}}
20:56:56 MQT: stat/Wohnzimmer/STATUS2 = {"StatusFWR":{"Version":"6.5.0.9(sonoff)","BuildDateTime":"2019-05-18T12:56:57","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
20:56:56 MQT: stat/Wohnzimmer/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TP-Link_B33E",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
20:56:56 MQT: stat/Wohnzimmer/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","01003BC0"]}}
20:56:56 MQT: stat/Wohnzimmer/STATUS5 = {"StatusNET":{"Hostname":"Wohnzimmer-4859","IPAddress":"192.168.0.107","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"193.31.25.87","Mac":"3C:71:BF:31:92:FB","Webserver":2,"WifiConfig":4}}
20:56:56 MQT: stat/Wohnzimmer/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.100","MqttPort":1883,"MqttClientMask":"COUCH_%06X","MqttClient":"COUCH_3192FB","MqttUser":"maxl","MqttCount":37,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:56:56 MQT: stat/Wohnzimmer/STATUS7 = {"StatusTIM":{"UTC":"Fri May 24 19:56:56 2019","Local":"Fri May 24 20:56:56 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:58","Sunset":"20:36"}}
20:56:56 MQT: stat/Wohnzimmer/STATUS10 = {"StatusSNS":{"Time":"2019-05-24T20:56:56"}}
20:56:56 MQT: stat/Wohnzimmer/STATUS11 = {"StatusSTS":{"Time":"2019-05-24T20:56:56","Uptime":"1T01:15:15","Vcc":3.405,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"POWER":"OFF","Dimmer":61,"Color":"9B91000000","HSBColor":"56,100,61","Channel":[61,57,0,0,0],"CT":153,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"TP-Link_B33E","BSSId":"0C:80:63:E5:B3:3E","Channel":6,"RSSI":90,"LinkCount":18,"Downtime":"0T00:01:12"}}}`

STATUS 0 of the Shelly1

`20:58:32 CMD: status 0
20:58:32 SRC: WebConsole from 192.168.0.101
20:58:32 RSL: Received Topic /status, Data Size 1, Data 0
20:58:32 RSL: Group 0, Index 1, Command STATUS, Data 0
20:58:32 MQT: stat/Wohnzimmer/STATUS = {"Status":{"Module":46,"FriendlyName":["Deckenlicht Wohnzimmer"],"Topic":"Wohnzimmer","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:58:32 MQT: stat/Wohnzimmer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"2T01:55:39","StartupUTC":"2019-05-22T18:02:53","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":147,"SaveAddress":"F9000"}}
20:58:32 MQT: stat/Wohnzimmer/STATUS2 = {"StatusFWR":{"Version":"6.5.0.9(sonoff)","BuildDateTime":"2019-05-11T04:03:33","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
20:58:32 MQT: stat/Wohnzimmer/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TP-Link_B33E",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
20:58:32 MQT: stat/Wohnzimmer/STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":18,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","01003BC0"]}}
20:58:32 MQT: stat/Wohnzimmer/STATUS5 = {"StatusNET":{"Hostname":"Wohnzimmer-4064","IPAddress":"192.168.0.104","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"193.31.25.87","Mac":"B4:E6:2D:55:EF:E0","Webserver":2,"WifiConfig":4}}
20:58:32 MQT: stat/Wohnzimmer/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.100","MqttPort":1883,"MqttClientMask":"WZ_Deckenleuchte_%06X","MqttClient":"WZ_Deckenleuchte_55EFE0","MqttUser":"maxl","MqttCount":44,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:58:32 MQT: stat/Wohnzimmer/STATUS7 = {"StatusTIM":{"UTC":"Fri May 24 19:58:32 2019","Local":"Fri May 24 20:58:32 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:58","Sunset":"20:36"}}
[... cut ...] HLED","modelid":"LCT007","uniqueid":"3c:71:bf:31:92:fb:00:11-1","swversion":"5.50.1.19085"})
21:02:57 HTP: Hue API (/OD4gGlAzzyzQGfFrCZ59qPYYAFsDlNEPvFiNjQZZ/lights/1)
21:02:57 HTP: Hue Result ({"state":{"on":true,"bri":155,"colormode":"hs","xy":[0.44658,0.52597],"hue":10223,"sat":254,"ct":153,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"COUCHLED","modelid":"LCT007","uniqueid":"3c:71:bf:31:92:fb:00:11-1","swversion":"5.50.1.19085"})`

hope that did help.

jay-wilkinson, 2019-05-25:
On an Adafruit HUZZAH ESP8266 breakout: 6.5.0 release 2.4.2 Core standard sonoff.bin:
Connected 3 LEDs. Set them to PWM1 PWM2 PWM3. On GPIO 2, 13, and 14 respectively. I get 1 toggle button and 1 dimmer bar. (It does toggle and dim all 3 together) And:
{"1":{"state":{"on":true,"bri":19,"hue":0,"sat":0,"xy":[0.5, 0.5],"ct":500,"alert":"none","effect":"none","colormode":"hs","reachable":true},"type":"Extended color light","name":"Sonoff10","modelid":"LCT007","uniqueid":"ec:fa:bc:05:a9:50:00:11-1","swversion":"5.50.1.19085"}}
from page http://192.168.1.235/api/pi/lights

*****
Then I set them as Relay1, Relay2, and Relay3 respectively.
*****
3 Toggle buttons and:
{"1":{"state":{"on":false,"bri":4,"hue":0,"sat":0,"xy":[0.5, 0.5],"ct":500,"alert":"none","effect":"none","colormode":"hs","reachable":true},"type":"Extended color light","name":"Sonoff10","modelid":"LCT007","uniqueid":"ec:fa:bc:05:a9:50:00:11-1","swversion":"5.50.1.19085"},"2":{"state":{"on":false,"bri":4,"hue":0,"sat":0,"xy":[0.5, 0.5],"ct":500,"alert":"none","effect":"none","colormode":"hs","reachable":true},"type":"Extended color light","name":"Sonoff2","modelid":"LCT007","uniqueid":"ec:fa:bc:05:a9:50:00:11-2","swversion":"5.50.1.19085"},"3":{"state":{"on":false,"bri":4,"hue":0,"sat":0,"xy":[0.5, 0.5],"ct":500,"alert":"none","effect":"none","colormode":"hs","reachable":true},"type":"Extended color light","name":"Sonoff3","modelid":"LCT007","uniqueid":"ec:fa:bc:05:a9:50:00:11-3","swversion":"5.50.1.19085"}}

Alexa is able to turn them on and off independently. Downside is she seems to think the relays or PWMs are color ("colormode":"hs") and displays a color slider in the app on my Amazon Fire. 

Also I checked in discord and someone had said that the color thing is intended. And that there wasn't enough space in the code to fix it. So I doubt that'll ever get fixed unless it gets forked. 

Summary:
I think there's a problem using PWM with Emulated Hue due to setting Emulation to be color which would use multiple pins and PWMs. But this should give you a place to look.

s-hadinger, 2019-05-25:
@jay-wilkinson This is a different question from the main topic of this issue. Could you please open a separate Issue so we don't get people confused?

s-hadinger, 2019-05-25:
I added some logs and I can confirm that Alexa sends a UDP UPNP discover just before sending the command to Tasmota. I guess this is a check to make sure the device hasn't change IP address since last discover. This is where something must be wrong, and Alexa gets confused by the Tasmota answer.

```
09:22:30 UDP Packet received m-search*http/1.1host:239.255.255.250:1900man:"ssdp:discover"mx:3st:ssdp:all
09:22:32 Packet 1 HTTP/1.1 200 OK
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=100
EXT:
LOCATION: http://192.168.1.203:80/description.xml
SERVER: Linux/3.14.0 UPnP/1.0 IpBridge/1.17.0
hue-bridgeid: 840D8EFFFE59313E
ST: upnp:rootdevice
USN: uuid:f6543a06-da50-11ba-8d8f-840d8e59313e::upnp:rootdevice

09:22:32 Packet 2 HTTP/1.1 200 OK
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=100
EXT:
LOCATION: http://192.168.1.203:80/description.xml
SERVER: Linux/3.14.0 UPnP/1.0 IpBridge/1.17.0
hue-bridgeid: 840D8EFFFE59313E
ST: uuid:f6543a06-da50-11ba-8d8f-840d8e59313e
USN: uuid:f6543a06-da50-11ba-8d8f-840d8e59313e

09:22:32 Packet 3 HTTP/1.1 200 OK
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=100
EXT:
LOCATION: http://192.168.1.203:80/description.xml
SERVER: Linux/3.14.0 UPnP/1.0 IpBridge/1.17.0
hue-bridgeid: 840D8EFFFE59313E
ST: urn:schemas-upnp-org:device:basic:1
USN: uuid:f6543a06-da50-11ba-8d8f-840d8e59313e

09:22:32 UPP: Hue 3 response packets sent to 192.168.1.43:50000
09:22:32 HTP: Hue setup
09:22:32 Send XML description <?xml version="1.0"?><root xmlns="urn:schemas-upnp-org:device-1-0"><specVersion><major>1</major><minor>0</minor></specVersion><URLBase>http://192.168.1.203:80/</URLBase><device><deviceType>urn:schemas-upnp-org:device:Basic:1</deviceType><friendlyName>Amazon-Echo-HA-Bridge (192.168.1.203)</friendlyName><manufacturer>Royal Philips Electronics</manufacturer><modelDescription>Philips hue Personal Wireless Lighting</modelDescription><modelName>Philips hue bridge 2012</modelName><modelNumber>9290002
09:22:46 HTP: Main Menu
```

wernerk63, 2019-05-25:
**Some Tests / Reproduction Issue:**
Result:   Alexa App and AVS switches the light Nr. of the Last learned/recognized HueBridge

One Bridge -> all ok  (HueBridge1/light1+ HueBridge1/light2)

Learn/find Second Bridge (and delete not used Bridges in this Scenario e.g. HueBridge3) 
Alexa switches the light Nr. of the Second Bridge.  Non 'double'  light's  switched correct
(HueBridge2/light 1, HueBridge2/light2, HueBridge2/light3, HueBridge2/light4)

In my opinion it isn't an Tasmota issue, we need to monitor (and decode) Network traffic between Alexa and the Bridges.....unfortunately so far no way found

rherriage, 2019-05-25:
an update.... i tried 6.5.0.12 and now the echo dot voice is working, but the app still isn't working.   It will return status, but it won't change status.  here is the template:

### ISSUE DESCRIPTION - TROUBLESHOOTING
Alexa stopped working with hue emulation a couple of days ago.  This happened all at the same time on 6 x Sonoff IFAN02s running 6.5.0.  The alexa app nor echo dots could change the status of the switch; however, the app couldsee what the current status is (can see if the switch is on or off if changed manually).  I updated one to 6.5.0.12 and now voice command works from echo, but the IOS app does not

  
- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (i.e. Sonoff Basic) : _sonoff IFAN02____
- [x ] Tasmota binary firmware version number used : __6.5 core 2.5 and 6.5.0.12 core 2.5.1 pre-compiled
- [x ] Development IDE - Compiler / Upload tools used : _arduino ide___ / _OTA upgrade___
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
14:48:59 CMD: status 0
14:48:59 RSL: STATUS = {"Status":{"Module":44,"FriendlyName":["Play Room lights"],"Topic":"Play Room","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:48:59 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020501/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:00","StartupUTC":"2019-05-25T13:38:59","Sleep":50,"CfgHolder":4617,"BootCount":274,"SaveCount":283,"SaveAddress":"F8000"}}
14:48:59 RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0.12(4d070bf-sonoff)","BuildDateTime":"2019-05-24T16:02:07","Boot":7,"Core":"2_5_1","SDK":"2.2.1(cfd48f3)"}}
14:48:59 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["rustytoo",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008001","280500000100000000000000000000000000","00000000"]}}
14:48:59 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":428,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","001783A0","23B617CC","01003BC0"]}}
14:48:59 RSL: STATUS5 = {"StatusNET":{"Hostname":"Play Room-7970","IPAddress":"192.168.5.57","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer":"208.67.222.222","Mac":"DC:4F:22:DB:9F:22","Webserver":2,"WifiConfig":4}}
14:48:59 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat May 25 13:48:59 2019","Local":"Sat May 25 14:48:59 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:57","Sunset":"20:37"}}
14:48:59 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-05-25T14:48:59"}}
14:48:59 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-05-25T14:48:59","Uptime":"0T00:10:00","SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"POWER1":"OFF","FanSpeed":0,"Wifi":{"AP":1,"SSId":"rustytoo","BSSId":"58:AC:78:F8:F0:B1","Channel":6,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:04"}}}
14:49:03 APP: Serial logging disabled

```
- [ x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
this is from the app test.  You can see that it returns the state, but it will not make a change.

14:51:44 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Play Room lights","modelid":"LCT007","uniqueid":"dc:4f:22:db:9f:22:00:11-1","swversion":"5.50.1.19085"})
14:51:48 WIF: Checking connection...
14:51:48 WIF: Connected
14:52:08 WIF: Checking connection...
14:52:08 WIF: Connected
14:52:10 HTP: Information
14:52:13 HTP: Hue API (/o61z9UvSfD2xIFxfM2iVhuo4xHz4XlpUseCkrfY0/lights/1)
14:52:13 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Play Room lights","modelid":"LCT007","uniqueid":"dc:4f:22:db:9f:22:00:11-1","swversion":"5.50.1.19085"})
14:52:28 WIF: Checking connection...
14:52:28 WIF: Connected
14:52:44 UDP: Packet (199)
14:52:44 UDP: Packet (199)
14:52:44 UDP: Packet (199)
14:52:44 UDP: Packet (199)
14:52:44 UDP: Packet (199)
14:52:44 UDP: Packet (199)
14:52:44 UDP: Packet (199)
14:52:44 UDP: Packet (199)
14:52:48 WIF: Checking connection...
14:52:48 WIF: Connected
14:52:57 UDP: Packet (199)
14:52:57 UDP: Packet (199)
14:52:57 UDP: Packet (199)
14:52:58 UDP: Packet (199)
14:52:59 UDP: Packet (199)
14:53:00 UDP: Packet (199)
14:53:00 UDP: Packet (199)
14:53:00 UDP: Packet (199)
14:53:00 UDP: Packet (199)
14:53:01 UDP: Packet (199)
14:53:02 UDP: Packet (199)
14:53:03 UDP: Packet (199)
14:53:03 UDP: Packet (199)
14:53:03 UDP: Packet (199)
14:53:03 UDP: Packet (199)
14:53:04 UDP: Packet (199)
14:53:05 UDP: Packet (199)
14:53:08 WIF: Checking connection...
14:53:08 WIF: Connected
14:53:14 HTP: Hue API (/o61z9UvSfD2xIFxfM2iVhuo4xHz4XlpUseCkrfY0/lights/1)
14:53:14 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Play Room lights","modelid":"LCT007","uniqueid":"dc:4f:22:db:9f:22:00:11-1","swversion":"5.50.1.19085"})
```
**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-05-25:
I've been spending some time on it. I confirm that voice control works but app or on-screen control (Echo Spot) does not work. It does not seem linked to Tasmota. But still digging.

s-hadinger, 2019-05-25:
Interestingly, when you use the app, it looks like the commands are always dispatched to the Tasmota device with the smallest IP address. Though it works ok when using voice commands.
I have tweaked a lot of parameters in Tasmota (BridgeIDs...) with no change. I will contact Alexa support.

wongnam, 2019-05-26:
@s-hadinger Please note that Echo Show, Echo Spot, Sonos One do not have the capability to discovery devices locally.

Supermicha81, 2019-05-26:
I have the same problem. But my issues only appear, if I try to switch Sonoffs in Groups.
All devices were listed correctly in the app and react fine, if I anounce them directly.
If I try to switch a Alexa group, the result is unpredictable.
Sometimes only on device or only on channel of an 4ch device react, sometimes a different multichannel device react. After deleting the devices from Alexa and search again, the behavior is newly different and other devices react on the same command.

Looks for me, that Amazon make some strage changes...

maybee off-topic:
Last night Alexa was not able to find a Sonoff Basic (WeMo-Emu). Only if I tried to search after televisons in the Alexa-App, the missing Basic was found... really strange...

I use only Echo-Dots (2.Gen) and all my Tasmota-Devices runs on 6.4.1.9-mod-1.43.9, the multichannel devices with Hue-Emu, the singlechannel with WeMo-Emu...

I'm sorry, today I'm out of time to create some logs...
If nessaccary, I will do later...

Update: My wife approved 5 Minutes ago, that at least one Alexa-Group with a multichannel-device works fine... I will observe.

Cemal32, 2019-05-26:
I have the same problem since two days.

I think the issue is caused by Phillips end of support of the Hue Bridge V1.

I’ve installed Yonomi to test, if Yonomi can detect my Sonoff 3CH Switches.

Yonomi detects then as Hue Bridges V1 and controls them perfectly.

So my assumption is, that Anazon also ended their support for Hue Bridges V1 and Supports only the V2. As Tasmota seems to emulate Hue Bridge V1, Alexa can’t control the Switches correctly.

arendst, 2019-05-26:
@Cemal32 you're right considering the Philips Hue Bridge V1 support see https://www2.meethue.com/en-gb/hue-bridge-upgrade

It ends in April 2020 but starting from April 2019

> Voice control unavailable
> 
> Voice assistants that function via cloud connection, such as Amazon Alexa and the Google Assistant, will not work with the Hue Bridge v1.
> 

So either Amazon has to be convinced that many users with Hue V1 hardware still need support or we have to find a way to emulate Hue V2.

Jason2866, 2019-05-26:
Happy that i never bought Hue hardware. Not the first time Philips left customers standing in the rain.
I can remember the stopped support services for TVs in the past which where included...

altelch, 2019-05-26:
If it's true that this is caused by a switch to v2 API then we have a problem: https://daenney.github.io/2019/04/09/emulating-philips-hue-bridge
After initial Handshake SSL is required to talk to the bridge. Tasmota would have to implement SSL as HTTP endpoint but with which ressources...hopefully it's only a minor incompatibility after an API change. Is the same behaviour detected by other emulation projects like Homeassistant or EspAlexa (https://github.com/Aircoookie/Espalexa)?

arendst, 2019-05-26:
That's an easy one: https://github.com/Aircoookie/Espalexa/issues/71

Same problem. Nice future.

s-hadinger, 2019-05-26:
I did change the advertised device name in Tasmota to "Philips hue bridge 2015" instead of "Philips hue bridge 2012" and it does not change anything. Alexa doesn't care about the model and uses the well documented Philips Hue API: https://developers.meethue.com/develop/hue-api/

I'm confident it has no link to the depreciation of Philips hardware nor linked to HTTP/HTTPS protocol. This would not explain why it still works on Voice Control.

It really looks like Alexa is confusing multiple Hue bridges on the same network. I'll contact Alexa on Monday to see if it's a bug on Alexa side.

rherriage, 2019-05-26:
what's weird is it is now voice control is intermittently working.  I can find no rhyme or reason.  My devices are not changing IPs (host DHCP lease), so it's not a discovery thing.   One device that was working yesterday is not working today.  I've got two devices on the 5/24 build, and one works, the other doesn't.  I've tried to force updates on all my echo dots (none needed them).  I've even tried to just turn them off, and try them one by one.  I've got echo, echo dot 2nd (like 4 of them) and echo dot 3rd gen..

Again, it'll detect status, but won't change on/off. 

I'm not sure why this would be related to the lack of support on V1, since support ended 4/20, and this just stopped working a few days ago.    It's definitely possible, but seems weird that even if it's not supported by phillips they would just nix that code in echo.

clabnet, 2019-05-26:
Me too. Only for tracKing thread.

wernerk63, 2019-05-26:
With one HueBridge Alexa App+Voice works correct for me. ( -> delete all other devices !)
After add a Second Bridge App+ Voice works also,  but Alexa seams to be confused about HueID and switches the false Bridge Device. (-> also delete all other not used Devices in this scenario)
(m.E. it switches the last configured -> s-hadinger say this with the lowest IP) .
With more Bridges/Multichannel Devices in Alexa it gets very confused, the Status in App hangs or errors occur, voice switches or not.
Be patiently, wait for Alexa feedback.

znanev, 2019-05-26:
I was using WeMo emulation for all my Tasmota devices until a week ago. Alexa had no troubles finding or controlling the Tasmota devices individually or in groups.
Then I decided that Philips Hue emulation is the way forward so I updated the firmwares of all Tasmota devices to latest dev. version available and changed the emulation to Philips Hue. Since then controlling the devices in groups from Alexa has unpredictable results. Controlling the devices individually works with no troubles, but switching groups of devices on/off is impossible when more that one device is in Hue emulation mode.
So if it turns out that Amazon did something on their end related to the old Philips V1 devices, this will be a deal breaker for me. I'll have to either live with WeMo emulation (although partially broken and deprecated), or if this doesn't work - changing the voice assistant to some DIY one...

altelch, 2019-05-27:
Ok, just compiled a 6.5.0.12(sonoff)-2_5_2 and flashed it on a D1 mini, activated HUE and did a device search. Device was detected by Alexa and I can still switch my old Hue emulated devices or the new one without problems. I only have Echo and Echo-Dot Gen.1 devices on firmware 637566720 (Dot) and 635556820 (Echo). Maybe this is related to Gen.2 or later devices.

peteakalad, 2019-05-27:
Same issue for me, basic with hue emulation on no longer working. Readded few times, seemed to work once but then stopped working. 

I have two tickets echo dots and a v2 hue bridge if that matters..

Reverted to using wemo and that’s working ok with Alexa.

znanev, 2019-05-27:
@altelch Did you try to control Hue emulated devices in groups?

I also have no problems controlling Hue emulated devices from Alexa (mine is 3rd gen) individually, but once I add two or more in a group - the behaviour is erratic and unpredictable.

altelch, 2019-05-27:
@znanev Ok, can reproduce the issue. If creating a group over two emulated bridges, the bridge with the highest IP (or latest added bridge) gets switched. After removing the newest device from the group and only leaving devices on one emulated bridge still the latest added bridge gets the requests also none of it's devices are member of the group.

altelch, 2019-05-27:
Test 2: Created a group with only devices from latest added bridge. Switching works. Add a device from an older bridge, devices on older bridge don't get switched. Remove devices from latest added bridge, Echo responds with "No answer from device xxx". Didn't see any SSDP requests in front of switching commands. Something in Group-handling of Alexa got messed up I'd guess.

wernerk63, 2019-05-27:
No groups.......,  
only the existence of a second HueBridge in Alexa causes the issue.
Test One Hue Bridge 2 Devices (T1EU) every device in a Group works correct.
Another Group (sonoffs basic, s20) with wemo devices works all the Time.

znanev, 2019-05-27:
@wernerk63 Indeed existence of multiple Hue emulated bridges seems to cause troubles with Alexa. A single Hue emulated bridge (switch with RGB leds, emulated as two devices) works fine with Alexa. Adding another Hue emulated bridge somehow "steals" the packets from the earlier added ones... 

Can anyone with at least two "real" Hue bridge v1 devices verify if Alexa exhibits the same problems?

jemarr, 2019-05-27:
Same problem since four days ago... Amazon has changed something.

I tried Fauxmo and Espalexa Hue emulation libs as well with same results.

I have three ESP8266 boards with 1, 2 and 4 devices each. The last discovered devices board (casually is the lowest fixed IP adress) is which Alexa takes as 1st, 2nd.... device on Routines and Groups, doesn't matter the device name.

When you call by voice all devices runs well, the problem is just with Groups and Routines.

I've 2 Echo Dot 3rt Gen.

Supermicha81, 2019-05-28:
I have another maybee intersting fact.
If I rename devices in the Alexa App. They are no longer controllable...
In detail: On my Sonoff 4ch with Hue-Emu the first channel ist named "TV" and Alexa switch this without any Problem. I tried to rename it to "Fernseher" and it does not work anymore.
Changing back to "TV" works fine.

If I set the friendly name in Tasmota to "Fernseher" and search for new devices, Alexa is able to switch the device "Fernseher" without problems.

Maybee this helps to delimit the problem..

P.S. Alexa was not able to discover the deleted and renamed device als lamp or socket
I have Alexa tried to look after television-device to find the new Sonoff.

Supermicha81, 2019-05-29:
Is it possible to "duplicate" the Wemo-Emulation? So that multichannel-devices emulate 2 or 4 single Wemos?
This would also solve the problem that switches will have a dimming slider in the Alexa-App.

rherriage, 2019-05-29:
so... one thing that I've noticed.  For the IFAN02 to work with fan speeds, multiple need to be set properly.  Like for high, switch 1 and 3 needs to be turned on, with 2 off.  In order to accommodate this, I've set up routines (set the fan to high), etc.  

If I call the friendly names individually (turn fan 1 on, turn fan 3 on) they work.  However the "routine" do do that does not.  Also, I've noticed that if I call the friendly name for the light "office light" it will also turn on.  But with the light attached to the office alexa, "turn the lights on" used to work, but it doesn't.   

I've tested every SONOFF IFAN02 in my house with the same result, except 1.  I've noticed that one random SONOFF will work properly.  And it seems to change which one works properly from day to day without me running a rediscover from the webpage.

s-hadinger, 2019-05-30:
All, although this is slightly off-topic, I wanted to dive into why Belkin Wemo emulation failed.
But to my surprise it actually worked with 6.5.0.13 (core 2_5_2).
It also appears that Alexa does not mix up devices with Wemo emulation.

Can you confirm? This would be a temporary workaround for some of you.

znanev, 2019-05-30:
@s-hadinger I can confirm that Wemo emulation works for me too. It also used to work before I switched over to Hue emulation.

I'm not sure however how all this corresponds with the recently closed poll #5826 ?

arendst, 2019-05-30:
The poll was initiated after users complaint of unusable wemo functionality.

As it turns out now it still works fine wemo will not be disabled any time soon.

Sorry for the miscommunication.

wernerk63, 2019-05-30:
looks similar, probably the same trigger
https://github.com/bwssytems/ha-bridge/issues/1093#issuecomment-495918797

znanev, 2019-05-30:
> @s-hadinger I can confirm that Wemo emulation works for me too. It also used to work before I switched over to Hue emulation.
> 

I was too quick to report... Now, having removed all Hue emulated devices and groups from Alexa, and discovered only WeMo emulated devices, groups are still unusable.

Individual devices respond fine, but when in groups - nothing turns on/off.

s-hadinger, 2019-05-30:
I got additional information from Alexa. There is a work-around to solve this issue, assigning a different lightId to each light. Normally lights have ids "1", "2"... on each bridge. Enforcing unique lightIds ensures that Alexa does not get confused.

The new lightId is computed as follows: (last byte of IP address * 16) + light number.

Ex: for 192.168.1.100 - light 1, it would be (100*16+1 = 1601)

This works fine if Tasmota devices don't change IP address, you may have issues if the devices change IP addresses.

PR coming in a few minutes.

s-hadinger, 2019-05-30:
I forgot to mention, you need to reflash all devices, remove all of them from Alexa and trigger a new discovery from Alexa.

Jason2866, 2019-05-30:
@s-hadinger  Great news!  
Big THX for your investigations you spend effort and time in.

closed 2019-05-30 by arendst

Jason2866, 2019-05-30:
@s-hadinger  is the IP based solution for Hue ID from Alexa Team or did you create it?
If it was yours, maybe considering to use MAC address (@meingraham  had this idea)
So there would be no change if getting a different IP adress

meingraham, 2019-05-30:
@Jason2866 and I were discussing this on Discord. The requirement is unique IDs, not the method to ensure uniqueness. If the IP address changes, the light will "announce" a different ID. Will Alexa have to "re-discover" it because the ID changed? With a MAC based uniqueness algorithm, uniqueness is still ensured, but the ID for the light will not change.

s-hadinger, 2019-05-30:
IP address was my ID, let me explain below.

I agree that MAC would be better, but keep in mind that lightId is an integer, and not sure whether we can go up to 32 bits. This would mean hashing the MAC to let's say 26 bits (to keep 4 bits for multiple lights on the same device), with a potential risk of collision.

We need to do more testing obviously and IP may not be the best way, but I wanted to share the good news as soon as possible.

Boris16, 2019-05-30:
Great solution. Thanks also from my side. 
I think the last byte of the IP-address is more unique. The MAC-address consists of 6 bytes. You have to find a CRC-sum or other algorithm to reduce the length. And that may not be 100% unique.

I have another question: As mentioned above, the V1 protocol will be discontinued in April 2020. Do we have problems in one year again?

Boris16, 2019-05-30:
I have just come up with another idea:
Why can not I assign a unique number (DeviceID=0..255) in the configuration to each Tasmota device. Is indeed a parameter more that you have to specify. But one is independent of IP and MAC address.

rherriage, 2019-05-30:
thanks @s-hadinger!!!  Like... a lot!  This solution will work perfectly for me, as I create a host-based DHCP lease for my tasmota devices.  Assuming this works, I'm ecstatic.  Tired of my wife asking me why it doesn't work anymore ><   

Thanks again, truly.

znanev, 2019-05-30:
Just tested the latest build (up to commit 2641e8a), and I can confirm that most of the Hue emulation functionality works correctly with Alexa.

However one of my devices, consisting of a relay and 3 LEDs (RGB), has another issue with Alexa. I'm not sure if I should open a new issue, because the workaround applied may be related to the following problem:
Hue emulated Tasmota device is correctly discovered as two lights (let's say DeviceSwitch - controlling a relay, and DeviceLight - controlling the RGB LEDs). Selecting either device in Alexa App presents the brightness slider and "Set Colour" option. These work fine for both devices, however pressing the light on/off button works correctly only for the "first" device - DeviceSwitch. When pressing the light on/off button on the "second" device - DeviceLight, the relay is switched off if it was previously switched on, but a "Device malfunction" error message appears. Pressing this same on/off button once the relay has been switched off from the "second" device has no effect and just "Device malfunction" error appears. This was not the case before - pressing the light on/off button of DeviceLight turned on/off the LEDs and the light on/off button of DeviceSwitch turned on/off the relay (i.e. they were not mixed up as they are now).

meingraham, 2019-05-31:
Unless a user assigns a static IP address, the lease could expire and a new IP address assigned giving the device a different Hue ID.

Each octet of the MAC address is a number in the range of 0 to 255 (like each octet of the IP address). The three most significant are assigned to a vendor (Espressif for these devices) and the lowest three enumerated within the ranges assigned to the vendor. Some hashing of the least significant octet (or two or three) are as random as the IP address... but constant over the life of the device.

An alternative to the MAC address would be the ESP chip ID. You could mask off a certain number of bits from the chip ID.

s-hadinger, 2019-05-31:
Good feedback, thanks.

It's important to ensure that there is unicity within the local network so using the last 8 bits (or 10-12 if you have a large home) looks safe. The main issue is when you reboot your router and all devices, if you haven't assigned static IPs they will change addresses. I don't know how Alexa behaves when it happens. I.e. is Alexa confused if lightIds change.

My main concern is when you change IP addresses, a new Tasmota device may re-advertize a lightId already assigned to another Tasmota.

I propose to combine it with the last byte of Mac address to reduce probability of conflict if IP addresses are reused.

What about : (8 last bits of MAC) followed by (8-10 last bits of IP address) followed by (2-4 bits of local light)

Jason2866, 2019-05-31:
As long the IP address is part of the lightID there will be issues when device gets a new IP.
So a wiki entry is neccessary anyway. I am fine with this. DHCP server has to be configured to give
devices with known MAC addresses always the same IP.
If there is a optimizing possibility to avoid the same lightID in a big LAN it is a good idea to do.

arendst, 2019-05-31:
@s-hadinger did you find out what the max number of bits for a lightid may be?

On the philips hue developers pages (https://developers.meethue.com/develop/hue-api/datatypes-and-time-patterns/) I cannot find the datatype for lightid. From the list I guess it's just 16 bits (uint16) as that is the biggest one documented there.

In that case I suggest to use the last 12 bits from the MAC address and 4 bits for local light. You just do not have more bits available.

Below some dumps from my MAC addresses on my routers:

[image]
[image]

All addresses are unique at the last 12 bits. It's tricky but not all MAC addresses are being used by Alexa either. Using MAC only solves any IP address change issue.

s-hadinger, 2019-05-31:
@arendst I missed your response and proposed another PR, but indeed it needs more testing.

It looks like we can go at least up to 24 bits. So we can try 16 bits from MAC and 4 for local light.

I will be off a few days, feel free to adjust the code accordingly :)

arendst, 2019-05-31:
I will. Happy holidays

s-hadinger, 2019-05-31:
We need to choose between an id guaranteed distinct (IP) but that can change and an id not guaranteed distinct (hash of Mac) but that doesn't change. 

That's why I'm in favor of combining both. Anyways I don't know how Alexa rediscovers a Hue bridge when its IP changed. I didn't have time to try.

[30 more replies]

PR #5890 (2019-05-30): Work-around for Philips Hue emulation issue (#5849)
## Description:

This is a work-around for #5849. Normally lights have ids "1", "2"... on each bridge. Enforcing unique lightIds ensures that Alexa does not get confused.

The new lightId is computed as follows: (last byte of IP address * 16) + light number.

Ex: for 192.168.1.100 - light 1, it would be (100*16+1 = 1601)

This works fine if Tasmota devices don't change IP address, you may have issues if the devices change IP addresses.

**Related issue (if applicable):** fixes #5849

## Checklist:
  - [x] The pull request is done against the  [... cut]