arendst/Tasmota #5850: Compile Error V6.5.12 with Rules enabled

helmar74, 2019-05-23:
Hi,

just for information:
If rules are enabled in my_user_config.h :
#define USE_RULES 

I get the following error during compile (using Core 2.5.2):
`Arduino: 1.8.9 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), nodemcu, 26 MHz, 40MHz, DOUT (compatible), 1M (no SPIFFS), 2, nonos-sdk 2.2.1 (legacy), v2 Higher Bandwidth (no features), Disabled, None, Only Sketch, 115200"

c:\temp\tasmotadev\06_sonoff_Development_Release\sonoff_dev_6512\sonoff_dev_6512.ino: In function 'void MqttPublishTeleState()':

sonoff_dev_6512:1953:19: error: 'RulesTeleperiod' was not declared in this scope

   RulesTeleperiod();  // Allow rule based HA messages

                   ^

c:\temp\tasmotadev\06_sonoff_Development_Release\sonoff_dev_6512\sonoff_dev_6512.ino: In function 'void PerformEverySecond()':

sonoff_dev_6512:2040:25: error: 'RulesTeleperiod' was not declared in this scope

         RulesTeleperiod();  // Allow rule based HA messages

                         ^

Mehrere Bibliotheken wurden für "TasmotaSerial.h" gefunden
 Benutzt: D:\Users\User\Desktop\arduino-1.8.9_Tasmota_252_Development\portable\sketchbook\libraries\TasmotaSerial-2.3.1
 Nicht benutzt: D:\Users\User\Desktop\arduino-1.8.9_Tasmota_252_Development\portable\sketchbook\libraries\TasmotaSerial-2.3.0
 Nicht benutzt: D:\Users\User\Desktop\arduino-1.8.9_Tasmota_252_Development\portable\sketchbook\libraries\TasmotaSerial-2.2.0
Mehrere Bibliotheken wurden für "Tsl2561Util.h" gefunden
 Benutzt: D:\Users\User\Desktop\arduino-1.8.9_Tasmota_252_Development\portable\sketchbook\libraries\Joba_Tsl2561-2.0.7
 Nicht benutzt: D:\Users\User\Desktop\arduino-1.8.9_Tasmota_252_Development\portable\sketchbook\libraries\Joba_Tsl2561-2.0.10
Mehrere Bibliotheken wurden für "Adafruit_SGP30.h" gefunden
 Benutzt: D:\Users\User\Desktop\arduino-1.8.9_Tasmota_252_Development\portable\sketchbook\libraries\Adafruit_SGP30-1.0.3
 Nicht benutzt: D:\Users\User\Desktop\arduino-1.8.9_Tasmota_252_Development\portable\sketchbook\libraries\Adafruit_SGP30-1.0.0.13
Mehrere Bibliotheken wurden für "ArduinoJson.h" gefunden
 Benutzt: D:\Users\User\Desktop\arduino-1.8.9_Tasmota_252_Development\portable\sketchbook\libraries\ArduinoJson-5.13.4
 Nicht benutzt: D:\Users\User\Desktop\arduino-1.8.9_Tasmota_252_Development\portable\sketchbook\libraries\ArduinoJson-5.11.2
exit status 1
'RulesTeleperiod' was not declared in this scope

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`

helmar74, 2019-05-23:
I think it's core related!

Compiling Version 6.5.0.3 with Core 2.5.2 is exactly 400KB.
Compiling the same version with Core 2.5.1 has a size of 385KB.

Regards

Helmar

arendst, 2019-05-23:
Did you ever read the error messages you receive?

It clearly states that at least different versions of several libraries are in your compile path.

I bet you just do not have the correct libraries loaded.

helmar74, 2019-05-23:
I deleted all libraries and copid newly. 
I can compile without an error with "default" my_user_config.h

If I enable in my_user_config.h :

> // Select none or only one of the below defines
> #define USE_RULES                                // Add support for rules (+8k code)
> #define USE_SCRIPT                               // Add support for script (+15k code)
> 
>   #define USE_EXPRESSION                         // Add support for expression evaluation in rules (+3k2 code, +64 bytes mem)
>   #define SUPPORT_MQTT_EVENT                     // Support trigger event with MQTT subscriptions (+3k5 code)

I'll get the following compile error:

> Arduino: 1.8.9 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), nodemcu, 26 MHz, 40MHz, DOUT (compatible), 1M (no SPIFFS), 2, nonos-sdk 2.2.1 (legacy), v2 Higher Bandwidth (no features), Disabled, None, Only Sketch, 115200"
> 
> C:\temp\06_sonoff_Development_Release\sonoff_dev_6512\sonoff_dev_6512.ino: In function 'void MqttPublishTeleState()':
> 
> sonoff_dev_6512:1953:19: error: 'RulesTeleperiod' was not declared in this scope
> 
>    RulesTeleperiod();  // Allow rule based HA messages
> 
>                    ^
> 
> C:\temp\06_sonoff_Development_Release\sonoff_dev_6512\sonoff_dev_6512.ino: In function 'void PerformEverySecond()':
> 
> sonoff_dev_6512:2040:25: error: 'RulesTeleperiod' was not declared in this scope
> 
>          RulesTeleperiod();  // Allow rule based HA messages
> 
>                          ^
> 
> exit status 1
> 'RulesTeleperiod' was not declared in this scope
>

helmar74, 2019-05-23:
When I disable:
"//#define USE_SCRIPT                               // Add support for script (+15k code)"

compilation works (it's disabled per default)

curzon01, 2019-05-23:
`// Select `**none or only one**` of the below defines`
```
#define USE_RULES // Add support for rules (+8k code)
#define USE_SCRIPT // Add support for script (+15k code)
```

find the mistake...

--
Please address this to [Support's Community](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md#supports-community).
The community channels are a better and more dynamic for helping you.
Github issues are mainly used for Tasmota software feature requests and bug reporting. 

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

helmar74, 2019-05-23:
OK, now it's clear ;-)
Thanks a lot

closed 2019-05-23 by helmar74

ilya89vet, 2019-07-08:
> #define SUPPORT_MQTT_EVENT // Support trigger event with MQTT subscriptions (+3k5 code)

What this line is responsible for? how I do she understand adds a feature to ipolzovat mqtt an event as the trigger of scenarios or I am wrong?