arendst/Tasmota #5851: Minimalversion too big 6.5.12 and Core 2.5.2

helmar74, 2019-05-23:
Hi,

I played a little bit around with the new core 2.5.2 and the current development release 6.5.12.
My "normal" firmware has about 600KB (compiled size)
When I compile the minimal version it is 404 KB and the OTA Upgrade of "normal" firmware failed, as there's too less memory.
I tried a lot of things to reduce the size of the minimal version, but it nothing helped.
(e.g. disabling stuff in "my_user_config.h)
Reducing the "normal" firmware's size isn't an option, as I'm using a lot of sensors, displays and so on.

Has anybody an idea how to shrink the size of the minimal firmware?
Is the size related to the new core (former tasmotaversions with core version 2.5.1) had about 385KB?

Thanks a lot in advance

helmar

helmar74, 2019-05-23:
I think it's core related!

Compiling Version 6.5.0.3 with Core 2.5.2 is exactly 400KB.
Compiling the same version with Core 2.5.1 has a size of 385KB.

Regards

Helmar

arendst, 2019-05-23:
I suggest you build a minimal version using core 2.3.0 which will be smallest possible.

Your production code can than still use core 2.5.2

helmar74, 2019-05-23:
OK, will do so.

closed 2019-05-23 by helmar74

Jason2866, 2019-05-23:
Disable mDNS if you havent done already. In core 2.5.2. there is big code size increase for mDNS.

helmar74, 2019-05-23:
@Jason2866 : 
Thanks. Code Size reduced from 600KB down to 591KB with mDNS disabled