arendst/Tasmota #5856: Support for ex-store wifi dimmer v4

martijnr17, 2019-05-24:
I've bought a 2 Kanal RS232 WiFi WLan Dimmer Modul V4 with ESP8266.
It is not on the supported list, I thought because it supports mqtt it works out of the box with domoticz but not.
I have no idea of programming but think it should be pretty easy for the pro's.
The module is well documented how to control over serial.
https://blog.ex-store.de/content/wifidimmer-v4

Jason2866, 2019-05-24:
Sorry we dont have the device and we cant buy everything...
No it isnt that easy. The described functions are for the included firmware.
How and what have to be done (if possible) to get it working needs completely (re)engeenired

martijnr17, 2019-05-24:
Sorry my mistake. More work than I thought. To bad.

closed 2019-05-24 by ascillato2