arendst/Tasmota #5858: duplicate mqtt message

ventzitt, 2019-05-24:
the problem is multiple mqtt messages 2 or 3
11:20:22 MQT: tele/Boiler/SENSOR = {"Time":"2019-05-24T11:20:22","ENERGY":{...
11:20:22 MQT: tele/Boiler/SENSOR = {"Time":"2019-05-24T11:20:22","ENERGY":....
11:20:23 MQT: tele/Boiler/SENSOR = {"Time":"2019-05-24T11:20:23","ENERGY":...
 I have read the following thread 
https://github.com/arendst/Sonoff-Tasmota/issues/2891
but it seems is something different
  
status 0:
09:22:59 MQT: stat/Boiler/STATUS = {"Status":{"Module":18,"FriendlyName":["Boiler"],"Topic":"Boiler","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:22:59 MQT: stat/Boiler/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"9T22:49:45","StartupUTC":"2019-05-14T09:33:14","Sleep":50,"CfgHolder":4617,"BootCount":144,"SaveCount":216,"SaveAddress":"F4000"}}
09:22:59 MQT: stat/Boiler/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:22:59 MQT: stat/Boiler/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ventsislav",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
09:22:59 MQT: stat/Boiler/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":11,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16400E","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
09:22:59 MQT: stat/Boiler/STATUS5 = {"StatusNET":{"Hostname":"Boiler-3831","IPAddress":"192.168.1.213","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"3C:71:BF:32:4E:F7","Webserver":2,"WifiConfig":4}}
09:22:59 MQT: stat/Boiler/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_324EF7","MqttUser":"ventzitt","MqttCount":4,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:22:59 MQT: stat/Boiler/STATUS7 = {"StatusTIM":{"UTC":"Fri May 24 08:22:59 2019","Local":"Fri May 24 09:22:59 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:58","Sunset":"20:36"}}
09:22:59 MQT: stat/Boiler/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:22:59 MQT: stat/Boiler/STATUS10 = {"StatusSNS":{"Time":"2019-05-24T09:22:59","ENERGY":{"TotalStartTime":"2019-04-05T11:40:58","Total":64.192,"Yesterday":2.129,"Today":0.472,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":1.00,"Frequency":50,"Voltage":246,"Current":0.000}}}
09:22:59 MQT: stat/Boiler/STATUS11 = {"StatusSTS":{"Time":"2019-05-24T09:22:59","Uptime":"9T22:49:45","Vcc":2.736,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"ventsislav","BSSId":"D4:5F:25:EA:D1:BC","Channel":6,"RSSI":50,"LinkCount":2,"Downtime":"0T00:00:25"}}}
0

weblog 4 is:
09:30:04 MQT: stat/Boiler/RESULT = {"WebLog":4}
09:30:04 DMP: 01 04 14 09 89 00 00 00 00 00 0A 00 00 0A 27 00 01 01 F4 00 64 00 00 D9 9A
09:30:04 DMP: 01 04 14 09 89 00 00 00 00 00 0A 00 00 0A 27 00 01 01 F4 00 64 00 00 D9 9A
09:30:04 CFG: Saved to flash at FB, Count 217, Bytes 3584
09:30:05 DMP: 01 04 14 09 87 00 00 00 00 00 09 00 00 0A 27 00 01 01 F4 00 64 00 00 B6 B0
09:30:06 DMP: 01 04 14 09 83 00 00 00 00 00 0B 00 00 0A 27 00 01 01 F4 00 64 00 00 40 C2
09:30:07 DMP: 01 04 14 09 83 00 00 00 00 00 0B 00 00 0A 27 00 01 01 F4 00 64 00 00 40 C2
09:30:08 DMP: 01 04 14 09 80 00 00 00 00 00 0B 00 00 0A 27 00 01 01 F4 00 64 00 00 04 86
09:30:09 DMP: 01 04 14 09 80 00 00 00 00 00 0A 00 00 0A 27 00 01 01 F4 00 64 00 00 06 07
09:30:10 DMP: 01 04 14 09 83 00 00 00 00 00 09 00 00 0A 27 00 01 01 F4 00 64 00 00 47 80
09:30:11 DMP: 01 04 14 09 84 00 00 00 00 00 09 00 00 0A 27 00 01 01 F4 00 64 00 00 F2 F4
09:30:12 DMP: 01 04 14 09 84 00 00 00 00 00 09 00 00 0A 27 00 01 01 F4 00 64 00 00 F2 F4
09:30:13 DMP: 01 04 14 09 7C 00 00 00 00 00 0B 00 00 0A 27 00 01 01 F4 00 64 00 00 64 C2
09:30:14 DMP: 01 04 14 09 7B 00 00 00 00 00 0B 00 00 0A 27 00 01 01 F4 00 64 00 00 D1 B6
09:30:15 DMP: 01 04 14 09 7E 00 00 00 00 00 0C 00 00 0A 27 00 01 01 F4 00 64 00 00 16 3D
09:30:16 DMP: 01 04 14 09 7F 00 00 00 00 00 0D 00 00 0A 27 00 01 01 F3 00 64 00 00 9C 80
09:30:17 DMP: 01 04 14 09 7F 00 00 00 00 00 0D 00 00 0A 27 00 01 01 F4 00 64 00 00 29 40
09:30:18 DMP: 01 04 14 09 83 00 00 00 00 00 0C 00 00 0A 27 00 01 01 F4 00 64 00 00 4B 85
09:30:19 WIF: Checking connection...
09:30:19 WIF: Connected
09:30:19 DMP: 01 04 14 09 7B 00 00 00 00 00 0C 00 00 0A 27 00 01 01 F4 00 64 00 00 DA F1
09:30:19 MQT: tele/Boiler/STATE = {"Time":"2019-05-24T09:30:19","Uptime":"9T22:57:05","Vcc":2.738,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"ventsislav","BSSId":"D4:5F:25:EA:D1:BC","Channel":6,"RSSI":52,"LinkCount":2,"Downtime":"0T00:00:25"}}
09:30:19 MQT: tele/Boiler/SENSOR = {"Time":"2019-05-24T09:30:19","ENERGY":{"TotalStartTime":"2019-04-05T11:40:58","Total":64.192,"Yesterday":2.129,"Today":0.472,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":1.00,"Frequency":50,"Voltage":243,"Current":0.000}}
09:30:19 MQT: tele/Boiler/SENSOR = {"Time":"2019-05-24T09:30:19","ENERGY":{"TotalStartTime":"2019-04-05T11:40:58","Total":64.192,"Yesterday":2.129,"Today":0.472,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":1.00,"Frequency":50,"Voltage":243,"Current":0.000}}
09:30:20 DMP: 01 04 14 09 7E 00 00 00 00 00 0C 00 00 0A 27 00 01 01 F4 00 64 00 00 16 3D
09:30:20 MQT: tele/Boiler/SENSOR = {"Time":"2019-05-24T09:30:20","ENERGY":{"TotalStartTime":"2019-04-05T11:40:58","Total":64.192,"Yesterday":2.129,"Today":0.472,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":1.00,"Frequency":50,"Voltage":243,"Current":0.000}}
09:30:21 DMP: 01 04 14 09 7E 00 00 00 00 00 0C 00 00 0A 27 00 01 01 F3 00 64 00 00 A3 FD
09:30:22 DMP: 01 04 14 09 82 00 00 00 00 00 0D 00 00 0A 27 00 01 01 F4 00 64 00 00 74 F8
09:30:23 DMP: 01 04 14 09 81 00 00 00 00 00 0C 00 00 0A 27 00 01 01 F3 00 64 00 00 87 FD
09:30:24 DMP: 01 04 14 09 7F 00 00 00 00 00 0C 00 00 0A 27 00 01 01 F4 00 64 00 00 2B C1
09:30:25 DMP: 01 04 14 09 7F 00 00 00 00 00 0C 00 00 0A 27 00 01 01 F3 00 64 00 00 9E 01
09:30:26 DMP: 01 04 14 09 80 00 00 00 00 00 0C 00 00 0A 27 00 01 01 F4 00 64 00 00 0F C1
09:30:27 DMP: 01 04 14 09 81 00 00 00 00 00 0C 00 00 0A 27 00 01 01 F3 00 64 00 00 87 FD
09:30:28 DMP: 01 04 14 09 82 00 00 00 00 00 0B 00 00 0A 27 00 01 01 F4 00 64 00 00 7D 3E
09:30:29 DMP: 01 04 14 09 82 00 00 00 00 00 0B 00 00 0A 27 00 01 01 F4 00 64 00 00 7D 3E
09:30:30 DMP: 01 04 14 09 82 00 00 00 00 00 0B 00 00 0A 27 00 01 01 F3 00 64 00 00 C8 FE
09:30:31 DMP: 01 04 14 09 83 00 00 00 00 00 0A 00 00 0A 27 00 01 01 F4 00 64 00 00 42 43
09:30:32 DMP: 01 04 14 09 7F 00 00 00 00 00 0A 00 00 0A 27 00 01 01 F4 00 64 00 00 22 07
09:30:33 DMP: 01 04 14 09 80 00 00 00 00 00 09 00 00 0A 27 00 01 01 F3 00 64 00 00 B6 04
09:30:34 DMP: 01 04 14 09 80 00 00 00 00 00 09 00 00 0A 27 00 01 01 F3 00 64 00 00 B6 04
09:30:35 DMP: 01 04 14 09 85 00 00 00 00 00 0A 00 00 0A 27 00 01 01 F4 00 64 00 00 CA CB
09:30:36 DMP: 01 04 14 09 81 00 00 00 00 00 0A 00 00 0A 27 00 01 01 F4 00 64 00 00 3B FB
09:30:37 DMP: 01 04 14 09 83 00 00 00 00 00 09 00 00 0A 27 00 01 01 F3 00 64 00 00 F2 40
09:30:38 DMP: 01 04 14 09 84 00 00 00 00 00 09 00 00 0A 27 00 01 01 F4 00 64 00 00 F2 F4
09:30:39 WIF: Checking connection...
09:30:39 WIF: Connected

arendst, 2019-05-24:
See #5028, #4813, #4130, #4145, #3795, #3778, #3660, #3648

In short, set PowerDelta to 0

ventzitt, 2019-05-24:
I have searched for hours it is a shame I couldn't find it and is been posted in soo many threads.
Sorry for that, 
Thank you, Theo. PowerDelta 0, and it is done. Thank you once again

closed 2019-05-24 by ventzitt