arendst/Tasmota #5862: 4d070bfb Fails to build from source

phatina, 2019-05-26:
### BUG DESCRIPTION
Using **Platformio**, Tasmota fails to build from source at 4d070bfb.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : shelly 2.5
- [x] Tasmota binary firmware version number used : self-compiled / (pre-compiled or self-compiled ?)
- [x] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :

CONSOLE OUTPUT HERE:
```
/home/phatina/Work/sonoff-tasmota-latest/sonoff/xdrv_01_webserver.ino: In function 'bool WebCommand()':
/home/phatina/Work/sonoff-tasmota-latest/sonoff/xdrv_01_webserver.ino:2464:1: error: a function-definition is not allowed here before '{' token
{
^
In file included from lib/NeoPixelBus-2.2.9/src/NeoPixelBus.h:60:0,
from /home/phatina/Work/sonoff-tasmota-latest/sonoff/xplg_ws2812.ino:25:
lib/NeoPixelBus-2.2.9/src/internal/NeoBitmapFile.h:30:22: error: expected '}' before end of line
#pragma pack(push, 2)
^
lib/NeoPixelBus-2.2.9/src/internal/NeoBitmapFile.h: At global scope:
lib/NeoPixelBus-2.2.9/src/internal/NeoBitmapFile.h:30:22: error: expected declaration before end of line
*** [.pioenvs/sonoff/src/sonoff.ino.cpp.o] Error 1
```

### TO REPRODUCE
`$ pio run`

### EXPECTED BEHAVIOUR
Compile and generate firmware.bin.

arendst, 2019-05-26:
No problem:

[image]

Jason2866, 2019-05-26:
I think you have old entries in your lib folder. Download the latest build and copy in a NEW empty folder.

closed 2019-05-26 by ascillato2