arendst/Tasmota #5863: Display not working alongside WS2812 8-led strip

demian85, 2019-05-26:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
While using a Wemos d1 mini with the following status output:
```
12:24:08 CMD: status 0
12:24:08 SRC: WebConsole from 192.168.0.11
12:24:08 RSL: Group 0, Index 1, Command STATUS, Data 0
12:24:08 MQT: stat/wemos/STATUS = {"Status":{"Module":18,"FriendlyName":["wemos","Sonoff2"],"Topic":"wemos","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[1,1,4,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
12:24:08 MQT: stat/wemos/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:01:59","StartupUTC":"2019-05-26T15:22:09","Sleep":50,"CfgHolder":4617,"BootCount":325,"SaveCount":5639,"SaveAddress":"F8000"}}
12:24:08 MQT: stat/wemos/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019-05-26T12:04:13","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:24:08 MQT: stat/wemos/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Garchatel WiFi",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["10008229","280500000100000000000000000000000000","00000000"]}}
12:24:09 MQT: stat/wemos/STATUS4 = {"StatusMEM":{"ProgramSize":511,"Free":492,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1625C2","FlashMode":3,"Features":["00000809","0FAAE394","000103A0","23B617CC","0000BBC0"]}}
12:24:09 MQT: stat/wemos/STATUS5 = {"StatusNET":{"Hostname":"wemos-2497","IPAddress":"192.168.0.12","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"BC:DD:C2:81:09:C1","Webserver":2,"WifiConfig":5}}
12:24:09 MQT: stat/wemos/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"wemos","MqttClient":"wemos","MqttUser":"wemos-7wmaw9ta8h932o8p","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:24:09 MQT: stat/wemos/STATUS7 = {"StatusTIM":{"UTC":"Sun May 26 15:24:09 2019","Local":"Sun May 26 12:24:09 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"-03:00","Sunrise":"07:47","Sunset":"17:53"}}
12:24:09 MQT: stat/wemos/STATUS10 = {"StatusSNS":{"Time":"2019-05-26T12:24:09","Switch1":"ON","Switch2":"ON","Switch3":"OFF","ANALOG":{"A0":2},"AM2301":{"Temperature":21.1,"Humidity":65.1},"BMP280":{"Temperature":22.2,"Pressure":1009.9},"BH1750":{"Illuminance":32},"PressureUnit":"hPa","TempUnit":"C"}}
12:24:09 MQT: stat/wemos/STATUS11 = {"StatusSTS":{"Time":"2019-05-26T12:24:09","Uptime":"0T00:02:00","SleepMode":"Dynamic","Sleep":0,"LoadAvg":1002,"POWER1":"ON","Dimmer":32,"Color":"515151","HSBColor":"0,0,32","Channel":[31,31,31],"Scheme":7,"Width":1,"Fade":"OFF","Speed":2,"LedTable":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Garchatel WiFi","BSSId":"CC:2D:E0:A6:11:71","Channel":1,"RSSI":98,"LinkCount":2,"Downtime":"0T00:00:06"}}}
12:24:09 SRC: MQTT
```

Oled display does not work and ignores all the DisplayText commands. It was working before adding the WS2812 to the configuration. Led strip works okay with or without the Display connected.
Is this the expected behavior? What can I do to make them work together?
I don't even want them to be powered on at the same time.

Thanks.

ascillato, 2019-05-26:
Hi, the version you upload is sonoff.bin
The version with display support is sonoff-display.bin. Please update to latest Tasmota with display support. Thanks

demian85, 2019-05-26:
I thought the status output would say what kind of build I'm using. I
compiled latest version myself and enabled display support alongside other
features.
I was using version 6.4 before and upgraded just to check if issue has been
fixed.

ascillato, 2019-05-26:
Ok. So, in order to have more debug information, please complete the troubleshooting template. Thanks

demian85, 2019-05-26:
Okay, not sure what else can I do for you. This is what's missing:

### TO REPRODUCE
Flash Wemos d1 mini with Tasmota 6.5 including display support. Attach a ws2812 led strip to some gpio. Add an OLED 128x64 I2C display and try to display some text using DisplayText cmd.

### EXPECTED BEHAVIOUR
Text is displayed in the screen

demian85, 2019-05-26:
Configuration:
[image]

ascillato2, 2019-05-26:
Please, update to http://thehackbox.org/tasmota/sonoff-display.bin in order to exclude any compilation issue. That version has display and ws2812 support.

demian85, 2019-05-27:
Somehow changing DisplayDimmer from 1 to 15 did the trick... Has this behavior changed in the latest version somehow?

closed 2019-05-27 by demian85

ascillato, 2019-05-27:
No, there was no change on that.

demian85, 2019-05-27:
Nope, After almost 6 months using this configuration. I just found out that the display needs to be powered on like a relay to use it. Why is the display considered as a relay? Now I understand the purpose of the TOGGLE button in the web interface...