arendst/Tasmota #5865: Switch status not read at boot

valepe, 2019-05-26:
### BUG DESCRIPTION
The status of two external switches are not read at startup

### REQUESTED INFORMATION

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : SONOFF DUAL R2
- [ ] Tasmota binary firmware version number used : 6.5.0 / pre-compiled
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :

### TO REPRODUCE
I solder two wires to GPIO 4 and 14. To these wires are attached two LDRs (connected in digital divide mode) to control the leds of a remote control panel. I set them as SWITCH1 and SWITCH2 with SwitchMode 1 for both. All works fine except during the boot up of the Sonoff.

### EXPECTED BEHAVIOUR
I expect to see the relays status updated just after the bootup by reading the status of the respective LDR (HIGH or LOW) .
But if the relays were on when Sonoff was turned off, they will put on at boot even if LDR are in off mode (with default PowerOnState set).
I changed the PowerOnState to 0 but they remain off even if LDRs are on.
I should cover/uncover the LDR to update the relay status.

ascillato, 2019-05-26:
Hi,

Not a bug. It is a misconfiguration of the switch. Please, try command switchmode 1 (follow)

If you need further assistant, please address this to the Tasmota support chat. Thanks

ascillato2, 2019-05-26:
[Tasmota Support Chat](https://discord.gg/Ks2Kzd4).

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-26 by ascillato2

valepe, 2019-05-26:
Sorry but as I wrote I have already setup both switches with switchmode 1. It doesn't work

ryancasler, 2020-10-13:
I am having the same problem.  I have the switchMode as 1 and setOption0 as 1 as well, to make sure that the sensor state is restored (otherwise it defaults to off).  If the switch is changed while the device is powered off (such as would happen with a contact sensor) the state is not read on boot.  The state is restored from memory but no reading of the pin is done.

erhapp, 2021-11-04:
I noticed the same behaviour. A switch in switchmode 1 does not influence the relay state upon reboot. It appears that the switch state is not evaluated or that the evaluation occurs before the state of the switch is set to pre-reboot state.

If this is not a bug, how can Tasmota be forced to update the relay to the state of the switch upon reboot?

ascillato, 2021-11-04:
Hi, Tasmota evaluates the change in the inputs and responds upon changes. This is by design.

If you want Tasmota to change the state of an output at boot time depending on the state of a switch, you must use a rule:

Assuming you have a switch, configured in a GPIO as a SWITCH1 with `switchmode1 1` (follow) that gives you 0 when the switch is in OFF position and 1 when it is in ON position:
```js
Rule1 1
Rule1 ON Switch1#Boot DO Power1 %value% ENDON
```

erhapp, 2021-11-04:
Perfect. Thank you very much for the detailed explanation.