arendst/Tasmota #5867: sdm230 shutdown does not reset readings

servim, 2019-05-28:
Hi!
When disconnecting sdm230 from electricity, the latest data from cdm230 continue to be published on the Tasmot web page, which is misleading. Is this a limitation of the protocol or can it be fixed so that when the sdm230 is disconnected, the readings are cleared?
Thanks :)

Jason2866, 2019-05-28:
It is by design, since SDM230 are build to be always connected to AC.

ascillato2, 2019-05-28:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-28 by ascillato2

Jason2866, 2019-05-28:
@ascillato2 and @servim 
@curzon01 will do a PR to change behaviour. We have discussed and doing as suggested is possible and a better behaviour.
Thx for your suggestion!

servim, 2019-05-28:
pzem004t the same situation.

PR #5874 (2019-05-29): SDM120/230 reset readings when offline
## Description:
Reset readings from Modbus device if offline

make also Modbus device address configurable within my_user_config.h/user_config_override.h
using `#define SDM120_ADDR <int>`

**Related issue (if applicable):** fixes #5867

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched (Also remember to update _changelog.ino_ file)
  - [ ] Only one feature/fix was added per PR.
  - [x] The code change is tested and works.
  - [x] The code change pass travis tests. **Your PR cannot be merg [... cut]