arendst/Tasmota #5868: Issues with LED control on Tuya switch

mesab0ogie, 2019-05-28:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
I recently purchased a variant of the Tuya switches in a 1,2 and 3 gang config. After a couple days of tinkering and reading through the available documentation i managed to flash the switches via Tuya-Convert with no issues(albeit after bricking one 3 gang switch in the process).

Now after that was done, i was faced with another can of worms, which is the configuration of the switches. Again after many hours of reading and digging i managed to make some progress. Firstly i would like to say that i do not think that these switches are the ones which are controlled via serial from the MCU as i have read issue #5815 and followed it and it does not work. What i did next was the setting gpios as relays trick. With that i managed to find out the relevant relays, buttons and leds. The thing is i thought i got this resolved as the switch i worked on first was a 3 gang. After some configuration and setting rules to link the leds to the relays, that switch worked flawlessly. 

However when i went on to configure the 2 gang switch i faced an issue. I managed to trace the respective buttons, relays and leds. But the problem is the leds are abit wonky. turning on relay 1 will turn on led 2, and turning on relay 2 will also turn on led 2. i have attached some videos and pictures so you can see what i mean. 
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [* ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [* ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ *] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [* ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [* ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : Generic Tuya Switch with TYWE3s chip
- [ ] Tasmota binary firmware version number used : 6.5.0.12____ / (pre-compiled with only wifi information added)
- [ ] Development IDE - Compiler / Upload tools used : Atom
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

15:39:25 SRC: WebConsole from 192.168.1.105
15:39:25 RSL: Received Topic /status, Data Size 1, Data 0
15:39:25 RSL: Group 0, Index 1, Command STATUS, Data 0
15:39:25 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:39:25 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:14:47","StartupUTC":"2019-05-28T14:24:38","Sleep":50,"CfgHolder":4617,"BootCount":42,"SaveCount":417,"SaveAddress":"F5000"}}
15:39:25 RSL: STATUS2 = {"StatusFWR":{"Version":"6.5.0.12(sonoff)","BuildDateTime":"2019-05-24T08:46:17","Boot":4,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
15:39:25 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mesab0ogie_2.4G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A001","280500000100000000000000000000000000","00000000"]}}
15:39:25 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":571,"Free":432,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0FDAE394","041783A0","23B617CC","01003BC0"]}}
15:39:25 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-5372","IPAddress":"192.168.1.138","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"EC:FA:BC:96:F4:FC","Webserver":2,"WifiConfig":4}}
15:39:25 RSL: STATUS7 = {"StatusTIM":{"UTC":"Tue May 28 14:39:25 2019","Local":"Tue May 28 15:39:25 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:54","Sunset":"20:40"}}
15:39:25 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-05-28T15:39:25"}}
15:39:25 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-05-28T15:39:25","Uptime":"0T00:14:47","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"mesab0ogie_2.4G","BSSId":"40:16:7E:32:AD:78","Channel":1,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:04"}}}
15:39:42 WIF: Checking connection...
15:39:42 WIF: Connected

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

this is the rule that i used
![20190528_222505](https://user-images.githubusercontent.com/10762974/58487259-dbda3180-8199-11e9-892f-e2f7499aab0d.jpg)

rule1 on power1#state do ledpower1 %value% endon on power2#state do ledpower2 %value% endon

15:26:28 APP: Button1 immediate
15:26:28 SRC: Button
15:26:28 RSL: RESULT = {"POWER1":"ON"}
15:26:28 RSL: POWER1 = ON
15:26:28 RUL: POWER1#STATE performs "ledpower1 1"
15:26:28 SRC: Rule
15:26:28 RSL: Received Topic /ledpower1, Data Size 1, Data 1
15:26:28 RSL: Group 0, Index 1, Command LEDPOWER, Data 1
15:26:28 RSL: RESULT = {"LedPower1":"ON"}
15:26:29 CFG: Saved to flash at F8, Count 414, Bytes 3584
15:26:32 APP: Button1 immediate
15:26:32 SRC: Button
15:26:32 RSL: RESULT = {"POWER1":"OFF"}
15:26:32 RSL: POWER1 = OFF
15:26:32 RUL: POWER1#STATE performs "ledpower1 0"
15:26:32 SRC: Rule
15:26:32 RSL: Received Topic /ledpower1, Data Size 1, Data 0
15:26:32 RSL: Group 0, Index 1, Command LEDPOWER, Data 0
15:26:32 RSL: RESULT = {"LedPower1":"OFF"}
15:26:33 CFG: Saved to flash at F7, Count 415, Bytes 3584
15:26:35 APP: Button2 immediate
15:26:35 SRC: Button
15:26:35 RSL: RESULT = {"POWER2":"ON"}
15:26:35 RSL: POWER2 = ON
15:26:35 RUL: POWER2#STATE performs "ledpower2 1"
15:26:35 SRC: Rule
15:26:35 RSL: Received Topic /ledpower2, Data Size 1, Data 1
15:26:35 RSL: Group 0, Index 2, Command LEDPOWER, Data 1
15:26:35 RSL: RESULT = {"LedPower1":"ON"}
15:26:35 CFG: Saved to flash at F6, Count 416, Bytes 3584
15:26:39 APP: Button2 immediate
15:26:39 SRC: Button
15:26:39 RSL: RESULT = {"POWER2":"OFF"}
15:26:39 RSL: POWER2 = OFF
15:26:39 RUL: POWER2#STATE performs "ledpower2 0"
15:26:39 SRC: Rule
15:26:39 RSL: Received Topic /ledpower2, Data Size 1, Data 0
15:26:39 RSL: Group 0, Index 2, Command LEDPOWER, Data 0
15:26:39 RSL: RESULT = {"LedPower1":"OFF"}
15:26:40 CFG: Saved to flash at F5, Count 417, Bytes 3584
15:26:42 WIF: Checking connection...
15:26:42 WIF: Connected

```
**(Please, remember to close the issue when the problem has been addressed)**

<iframe width="560" height="315" src="https://www.youtube.com/embed/8v2s-PwOgQM" frameborder="0" allow="accelerometer; autoplay; encrypted-media; gyroscope; picture-in-picture" allowfullscreen></iframe>

<iframe width="560" height="315" src="https://www.youtube.com/embed/wpZ94nruyuk" frameborder="0" allow="accelerometer; autoplay; encrypted-media; gyroscope; picture-in-picture" allowfullscreen></iframe>

Jason2866, 2019-05-28:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

mesab0ogie, 2019-05-28:
apologies for the trouble will close the issue and get on the Tasmota Support Chat.

closed 2019-05-28 by mesab0ogie