arendst/Tasmota #5869: SK6812 RGBW - real white does not work

Vujagig, 2019-05-28:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
The command 000000FF results in 00000000, the white LED is not activated.
All other colors work correctly

my_user_config setting:
#define USE_WS2812_CTYPE     NEO_GRBW 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (i.e. Sonoff Basic) : WeMos D1 Mini Pro
- [ x] Tasmota binary firmware version number used : 6.5.0.13 / self-compiled 
- [x ] Development IDE - Compiler / Upload tools used : Arduino IDE / OTA
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
17:16:45 CMD: status 0
17:16:45 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:16:45 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:01","StartupUTC":"2019-05-28T15:15:44","Sleep":50,"CfgHolder":4617,"BootCount":11,"SaveCount":63,"SaveAddress":"3F4000"}}
17:16:45 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0.13(sonoff)","BuildDateTime":"2019-05-28T16:16:44","Boot":31,"Core":"2_5_1","SDK":"2.2.1(cfd48f3)"}}
17:16:45 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SAW-AP",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
17:16:45 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":601,"Free":3472,"Heap":21,"ProgramFlashSize":4096,"FlashSize":16384,"FlashChipId":"1840EF","FlashMode":3,"Features":["00000809","0FFAE394","001783A0","23BE37CC","01003BC0"]}}
17:16:45 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3973","IPAddress":"192.168.10.99","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"2C:3A:E8:1F:6F:85","Webserver":2,"WifiConfig":4}}
17:16:45 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1F6F85","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:16:45 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue May 28 15:16:45 2019","Local":"Tue May 28 17:16:45 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+02:00","Sunrise":"05:14","Sunset":"21:30"}}
17:16:45 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-05-28T17:16:45"}}
17:16:45 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-05-28T17:16:45","Uptime":"0T00:01:01","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":56,"Color":"8F8F8F00","HSBColor":"0,0,56","Channel":[56,56,56,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"SAW-AP","BSSId":"44:4E:6D:39:96:E6","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
17:25:50 CMD: color 000000ff
17:25:50 SRC: WebConsole from 192.168.10.101
17:25:50 RSL: Received Topic /color, Data Size 8, Data 000000ff
17:25:50 RSL: Group 0, Index 1, Command COLOR, Data 000000ff
17:25:50 RSL: stat/sonoff/RESULT = {"POWER":"OFF","Dimmer":0,"Color":"00000000","HSBColor":"0,0,0","Channel":[0,0,0,0]}

17:26:58 CMD: color
17:26:58 SRC: WebConsole from 192.168.10.101
17:26:58 RSL: Received Topic /color, Data Size 0, Data 
17:26:58 RSL: Group 0, Index 1, Command COLOR, Data 
17:26:58 RSL: stat/sonoff/RESULT = {"Color":"00000000"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-05-29:
Nasty bug... The light_type is initialized with `light_type = 11` (+WS2812) with a `light_subtype = 3` (RGB). But later it is updated to `light_subtype = 4` (RGBW). Unfortunately the Light engine does not take into account the update.
I'm preparing a PR to correct this.

s-hadinger, 2019-05-29:
PR #5878.
@Vujagig can you please check it solves your issue and close it?

closed 2019-05-29 by arendst

Vujagig, 2019-05-30:
It Works

commit 6d7d8363 (2019-05-29): Merge pull request #5878 from s-hadinger/fix/ws2812_rgbw

Fix #5869, missing white channnel for WS2812
commit 74685f12 (2019-05-29): Fix #5869 typo
commit 86d14fe2 (2019-05-29): Fix #5869