arendst/Tasmota #5872: Rotary Encoder ky-040

craigueh, 2019-05-28:
Would it be possible to add support for a rotary encoder like ky-040?

I see there are guides with successful support for this encoder on a sonoff basic running espEasy - https://community.home-assistant.io/t/analogue-style-rotary-dimmer-switch-sonoff-espeasy-mqtt/29448.

Could Tasmota also support this encoder in a similar fashion?

blakadder, 2019-05-29:
It could if there were someone who codes the support and submits a PR

Jason2866, 2019-05-29:
Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-05-29:
Thanks for sharing your ideas 

It is already opened a request for encoders support in #288

closed 2019-05-29 by ascillato2

james-fry, 2019-06-11:
> I see there are guides with successful support for this encoder on a sonoff basic running espEasy - https://community.home-assistant.io/t/analogue-style-rotary-dimmer-switch-sonoff-espeasy-mqtt/29448.

If you are a HA user then you can also consider ESPHome.
I am the OP of that post on the HA forums and I now changed from ESP Easy to ESP Home.
I continue to use Tasmota for switch/relay applications, but ESPHome has a lot of very flexible and flash image efficient sensor support.