arendst/Tasmota #5873: how to get a switch rule to work with switch_topic="" and NOT control the relay?

mikep1998, 2019-05-29:
Hello, I have been working on this for days. Finally found the issue switch_topic was set to "my switch topic" so that the relay1 is not controlled by switch1. 

What is the correct way to get a rule to execute for switch1 and not have relay1 change?

thanks

blakadder, 2019-05-29:
Everything is easily findable in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches
You have not read the provided message when opening the issue. This is not the place to discuss issues like this.

---
Read the support [Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).
Please consult the [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information or seek help in [Discord Chat](https://discord.gg/Ks2Kzd4).).

closed 2019-05-29 by blakadder

meingraham, 2019-06-13:
@mikep1998 Are you mikep_1998 on Discord? If so, please check you private messages there. Thanks.