arendst/Tasmota #5875: Problem with Pzem-016

tizius79, 2019-05-29:
I connected the PZEM to the wemos d1 mini but I noticed that at every power-up the values of Wh day and total are always set at 42949.672 KWh.
What can I do to eliminate this problem?
Before I took the wifi module out of the house, I also had problems with all the displayed values that continued to change randomly.

00:00:00 Project sonoff Sonoff Version 6.5.0(sonoff)-2_5_0
00:00:00 WIF: Connecting to AP1 NETGEAR in mode 11N as contatore-2589...
00:00:02 RSL: tele/contatore/SENSOR = {"Time":"1970-01-01T00:00:02","ENERGY":{"TotalStartTime":"2019-05-29T12:42:41","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":133468016,"ApparentPower":6209134080,"ReactivePower":0,"Factor":105.49,"Frequency":2194,"Voltage":4281,"Current":1450291.875}}
00:00:05 RSL: tele/contatore/SENSOR = {"Time":"1970-01-01T00:00:05","ENERGY":{"TotalStartTime":"2019-05-29T12:42:41","Total":42949.672,"Yesterday":0.000,"Today":42949.672,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":1.00,"Frequency":50,"Voltage":233,"Current":0.000}}
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on contatore-2589 with IP address 192.168.43.219
12:46:11 MQT: Attempting connection...
12:46:16 MQT: Connect failed to 192.168.0.120:9001, rc -2. Retry in 10 sec

[image]

ascillato2, 2019-05-29:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

joba-1, 2019-05-29:
looks like a 32bit integer overflow

tizius79, 2019-05-29:
> looks like a 32bit integer overflow

It seems but the strange thing is that it is on all variables.
It could be that the IC of the pzem answers you have random characters starting the esp8266.

tizius79, 2019-05-29:
As I imagined.
The pzem passes him the wrong values.

00:00:00 CFG: Loaded from flash at 3F5, Count 68
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.5.0(sonoff)-2_5_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 NETGEAR in mode 11N as contatore-2589...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 DMP: 01 04 14 A7 2D B2 33 56 71 9A 64 4F 8C D3 6B C6 F3 55 B6 29 35 00 00 5B 94
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:02 RSL: tele/contatore/SENSOR = {"Time":"1970-01-01T00:00:02","ENERGY":{"TotalStartTime":"2019-05-29T20:11:44","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":133461464,"ApparentPower":6206813696,"ReactivePower":0,"Factor":105.49,"Frequency":2194,"Voltage":4280,"Current":1450291.750}}
00:00:03 DMP: 01 04 14 A7 2D B2 33 56 71 9A 64 4F 8C D3 6B C6 F3 55 B6 29 35 00 00 5B 94
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 DMP: 01 04 14 08 DC 00 00 00 00 00 08 00 00 00 02 00 00 01 F4 00 64 00 00 68 B8
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on contatore-2589 with IP address 192.168.0.83
00:00:04 APP: (UTC) Wed May 29 19:13:56 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
20:13:57 RSL: tele/contatore/SENSOR = {"Time":"2019-05-29T20:13:57","ENERGY":{"TotalStartTime":"2019-05-29T20:11:44","Total":42949.672,"Yesterday":0.000,"Today":42949.672,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":1.00,"Frequency":50,"Voltage":227,"Current":0.000}}
20:13:57 DMP: 01 04 14 08 D6 00 00 00 00 00 08 00 00 00 02 00 00 01 F4 00 64 00 00 F3 61
20:13:58 MQT: Attempting connection...
20:13:58 MQT: Connected

ascillato2, 2019-05-30:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you? Also, show how your hardware is connected and powered up. 

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

tizius79, 2019-05-30:
I solved it by putting a 5 second delay on power up. What do you say?

In the function "PzemAcEverySecond" I changed the line 41 from
"static uint8_t send_retry = 0;" to "static uint8_t send_retry = 5;"

[image]

ascillato2, 2019-06-01:
Cool!!!

Great to know that it is working for you now.

Thanks for sharing your solution. Very appreciated :+1:

closed 2019-06-01 by ascillato2

arendst, 2019-06-02:
Did some research and alas the wrong data does contain a valid crc :-(

So for now the only possible fix is to provide a delay of four seconds just after restart.

tizius79, 2019-06-02:
> Did some research and alas the wrong data does contain a valid crc :-(
> 
> So for now the only possible fix is to provide a delay of four seconds just after restart.

The same thing that I thought too.
I think the PZEM module needs a few seconds before being read after power.

andker87, 2020-02-12:
Hi @tizius79, I'm having several problems with pzem-016 and wemos d1 mini like you had. Live monitoring values are ok, but Today Energy randomly lowers or resets, or sometimes jumps to 42949.672 (and consequently the Total Energy Value). Did you had the same problem of lower values like me? This is the issue I opened https://github.com/arendst/Tasmota/issues/7674
Thank you

tizius79, 2020-02-12:
Hi @andker87,
Have you set gpio1 as Tx and gpio3 as Rx?

andker87, 2020-02-12:
No, gpio1 is RX and gpio3 is TX. If I switch via software data is 0 for all values, so I think I need ti switch also cables, but for now is as your last photo

tizius79, 2020-02-12:
Invert the pin configuration and invert the wires of the RX eTX.  You will see that it works later.

andker87, 2020-02-12:
I’ll try ad soon as I can and I’ll let you know. Have you seen my issue? Thank you again

tizius79, 2020-02-12:
Yes, I also had the same problem, but analyzing the code I saw that everything was linked to the wrong configuration of the pins.  If the rx and tx pins are inverted the code will create a serious software and it is precisely this that can create incorrect pzem readings.

andker87, 2020-02-12:
So you also had random value resets? Great, I hope to have time to modify the wirings soon! Do you use tasmota “classic” or tasmota sensor?

tizius79, 2020-02-12:
Yes, I had the same number and the voltage also changed at random.  I only used tasmota in this case.  Usually I prefer to write the code.  In what sense "classic" tasmota?

andker87, 2020-02-12:
I mean which version of tasmota, if the “classic”/“full” or the sensor version. But I desume you meant the classic one

tizius79, 2020-02-12:
I usually use the one to be compiled with Arduino which I think is the classic version

andker87, 2020-02-12:
yes, it is. so, I switched cables and gpios config... hope it works! I'll let you know in hours, just let me monitor the situation. thanks again

andker87, 2020-02-27:
Hi @tizius79 , I switched cables and gpios and it's working fine since 9 days. Thank you so much!

tizius79, 2020-02-27:
You're welcome