arendst/Tasmota #5877: change start mode when no wifi

prankousky, 2019-05-29:
Hi,

I have been using `Tasmota` for quite a while now. Mostly with `sonoff` wifi outlets, then replaced many of those with the `gosund` ones because power measuring functions. 

Recently while re-wiring, I have had a bunch of power outages (or rather, I had to turn of all electricity in the house in order to do the wiring). **Most** devices automatically connected to the wifi once it was back on. However, **some** of my devices did not! 

I remember that there was a way to configure this behavior when editing some configuration file before building the `.bin` file manually and flashing it. However, it has been quite a while that I did all this. I am just a hobbyist, and at the moment don't have too much spare time to work myself into all this (because I forgot how to do some things due to it "just working"™ and me not having to do anything with my hard- or software (which is a big thumbs up to tasmota, which just works without any trouble!)).

So now I'd like to **not** flash / upgrade my devices, but just send some command via mqtt (or the browser console) and tell **all** devices to keep searching for wifi when (re)booting. I believe some devices go into AP mode instead of continuously searching for their assigned wifi networks. ***Is this possible?***

I also use some sonoff light switches; those were some of the first things I flashed, so back then I definitely did not include the continuous wifi search; after the power outage, my bedroom light switch is stuck on "whatever it does" (not sure); I can control it manually, but if I want to get it working, I have to turn off the fuse for the bedroom while wifi is already up, then turn it on again (which is *usually* fine). However, just to save myself some work for the next time I'll do anything that requires me to power off everything and/or some tasmota devices **and** the wifi, I'd like to work around this.

According to `TasmoAdmin`, I am currently using versions

* 6.3.0.14
* 6.2.1.3
* 6.2.1.18
* 6.3.0.1 

While I'd prefer to just build fresh bins with the current version, I simply don't have the time to; or rather, I don't want to risk doing something incorrectly and then having to spend way too much time to fix it. So if I can avoid doing this **for now**, I'd prefer that. I will upgrade to current version once I have enough time that "allows me to mess up" (so if something goes wrong, I can spend hours trying to see what went wrong), but at the moment, if I re-flash and something goes wrong, I'll be stuck without working outlets until I have spare time to fix this.

## tl;dr

Is there a way to tell my "outdated" version devices to **always** keep searching for wifi until found when starting or power cycling *without* having to flash/re-flash anything? If so, how?

Thank you in advance for your help :)

meingraham, 2019-05-29:
tl;dr - this is a question better asked on Discord instead of GitHub.

You don't need to build your own firmware binaries. All of these 6.x devices should upgrade to the latest Tasmota and retain your configuration. You can ask TasmoAdmin to perform a bulk OTA upgrade (just make sure your OTAURL points to the right hackbox pre-compiled binary repository).

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

Mike

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-05-29:
Duplicated issue. Please see command wificonfig at https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#Wi-Fi Thanks

closed 2019-05-29 by ascillato2