arendst/Tasmota #5881: pzem004 shutdown does not reset readings

servim, 2019-05-30:
Hi!
When disconnecting pzem004  from electricity, the latest data from pzem004 continue to be published on the Tasmota web page, which is misleading. Is this a limitation of the protocol or can it be fixed so that when the pzem004 is disconnected, the readings are cleared?
PS It is not necessary that the device is permanently connected to electricity, it can have a plug to connect and when it is disconnected from electricity, the automation system to which the "tasmota" transmits data will think that the device is still on the network and consumes energy, it is, IMHO wrong.
Thanks :)

curzon01, 2019-05-30:
Similarly to your SDM230 issue but slightly different code change - I can do a PR to change the behaviour but I'm off next days so you have to wait...

servim, 2019-05-30:
Thank you for your responsiveness! I'll wait :)

arendst, 2019-05-30:
@curzon01 Norbert, I'll do it.

ascillato2, 2019-05-30:
Closing this issue as it is fixed.

Thanks everyone for working on this :+1:

closed 2019-05-30 by ascillato2