arendst/Tasmota #5883: core 2.5.2 fatal exeption

Supermicha81, 2019-05-30:
After updating a Sonoff-4ch Rev1 (not pro) to 6.5.0.13 with core 2.5.2 the device start every several hours new...
Restart Reason is:

Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000

while the device starts new, severals channels turn on.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff 4ch Rev1 (not Pro)
- [x] Tasmota binary firmware version number used : 6.5.0.13 / (compiled with benzino77/tasmocompiler)
- [x] Development IDE - Compiler / Upload tools used : uploaded with TasmoAdmin
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
11:42:27 CMD: status 0
11:42:27 MQT: stat/sonoff/STATUS = {"Status":{"Module":7,"FriendlyName":["weiße Lampe","Fernsehbeleuchtung","Pyramide","DVD-Player"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:42:27 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.10.11:80/data/firmwares/sonoff.bin","RestartReason":"Exception","Uptime":"0T03:51:17","StartupUTC":"2019-05-30T05:51:10","Sleep":250,"CfgHolder":4617,"BootCount":46,"SaveCount":135,"SaveAddress":"F7000"}}
11:42:27 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0.13(sonoff)","BuildDateTime":"2019.05.29 20:28:27","Boot":6,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
11:42:27 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Peace",""],"TelePeriod":300,"Resolution":"55C18100","SetOption":["00000009","280500000100060000000000000000000000","00000001"]}}
11:42:27 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":451,"Free":552,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","06886380","041683A1","210217C4","010028C0"]}}
11:42:28 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"TV-Schrank links","IPAddress":"192.168.10.103","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"60:01:xx:xx:xx:xx","Webserver":2,"WifiConfig":5}}
11:42:28 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.10","MqttPort":1883,"MqttClientMask":"TV-Schrank links","MqttClient":"TV-Schrank links","MqttUser":"Micha","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:42:28 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu May 30 09:42:28 2019","Local":"Thu May 30 11:42:28 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99}}
11:42:28 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-05-30T11:42:28"}}
11:42:28 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-05-30T11:42:28","Uptime":"0T03:51:18","SleepMode":"Dynamic","Sleep":250,"LoadAvg":4,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"C8:0E:xx:xx:xx:xx","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
I can't do this, because the Restarts are unpredictable
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Nothing

Supermicha81, 2019-05-30:
edit

ascillato2, 2019-05-30:
Hi,

Sorry, I can't reproduce your issue. 

Exception 29 means that a part of the program had tried to write in a protected address of the memory.

Seems that the problem is a bad or incomplete compilation of the core 2.5.2 or a bad flashing.

So, for solving your issue, please, erase all flash with esptool.py as explained in the wiki and flash with esptool.py the precompiled bins provided in thehackbox.org/tasmota. There you have up to core 2.5.1. Soon, the core 2.5.2 will be added there.

closed 2019-05-30 by ascillato2

ratsputin, 2019-12-01:
An interesting footnote to this...

I have several Lohas RGBW lights that I just upgraded to sonoff 7.1 and one of them is crashing with the exact same exception:

```
00:00:36 MQT: LSLight3/tele/INFO1 = {"Module":"AiLight","Version":"7.1.0(tasmota)","FallbackTopic":"cmnd/LSLight3_fb/","GroupTopic":"LSLights/cmnd/"}
00:00:36 MQT: LSLight3/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"LSLight3-7748","IPAddress":"192.168.1.212"}
00:00:36 MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
```

Not that I really expect anything from this; just wanted to add some information that might help address this in the future.  All lights were upgraded from 6.7 at the same time; this is the only light that's having this issue.

s-hadinger, 2019-12-02:
@ratsputin Thanks for the information. Indeed there's nothing we can do without more logs. I thought we got rid of the last crash source. Tell us if it happens again. 

Can you do `Reset 3` just in case?

`epc1:0x4000df64` is actually `memcpy` which does not tell much.

ratsputin, 2019-12-02:
@s-hadinger thank you for the reply.

Yesterday, I reflashed the device (not really expecting a change) and it did continue to crash.  Just now I performed a `reset 3`.  I'll post here with the results once I have something concrete/

I'm happy to provide additional logs if that would be of value; however, I have to assume that this is, in fact, some sort of flash corruption issue or hardware error as I have eight other devices that are ostensibly identical to this one running the same firmware without issue.

Thank you for the information regarding `memcpy`, by the way.

ratsputin, 2019-12-02:
@s-hadinger, it is continuing to crash after the `reset 3`.  There doesn't seem to be any particular pattern to the crashing; it happens anywhere from once to twice per hour.

I did, however, just realize that I may be incorrect in assuming that none of my other lights are crashing as well.  I only noticed this one, as it turns itself on when it reboots.  I need to check my logs to see if any of the other lights are crashing...

Okay, none of my other lights are logging any `"RestartReason":"Exception"` messages so it's just this one.

Would you like for me to gather further information on this one or should I just erase the config and start over with it?

s-hadinger, 2019-12-02:
Let's put in on hold for now. We need the stacktrace but currently you can only get it through serial.

I found this project to save the stacktrace in Flash: https://github.com/krzychb/EspSaveCrash

@arendst Unless you have another solution, I would like to give it a try: adding a command to capture the next stacktrace. Making it not automatic will protect the Flash in case of bootloop.

What about takeing taking the 4K block just below Settings. That should be bank `SETTINGS_LOCATION - CFG_ROTATES`, i.e. `SPIFFS_END - 8`, right?

arendst, 2019-12-02:
-8 is used for power cycle detection. Take -9

slipx06, 2019-12-02:
Maybe not related but upgraded my POW R2 to 7.1.1 and getting the same.

```
00:21:14 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Pow R2","Version":"7.1.1(tasmota)","FallbackTopic":"cmnd/DVES_178159_fb/","GroupTopic":"cmnd/sonoffs/"}
00:21:14 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0345","IPAddress":"192.168.1.106"}
00:21:15 MQT: tele/sonoff/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
```

ascillato, 2019-12-02:
@slipx06 did your device sync with NTP?

slipx06, 2019-12-02:
> @slipx06 did your device sync with NTP?

No NTP server. Web console is also very slow and unresponsive now.

ascillato, 2019-12-02:
Ok. So, that seems to be your issue. I had a problem few days ago where my ISP blocked some NTP servers. One of those was the one used by one of my cameras and another was the one used by my router.  So, I start having that exception 29. I set another NTP server in my router and all issues gone.

ascillato, 2019-12-02:
I also use Tasmoadmin and when it is querying my Tasmotas, I don't have any issue. So. Try to fix your NTP first.

slipx06, 2019-12-03:
Checked and its connecting to pool.ntp.org without any issues. It's only my POW R2 that has this performance problem.

s-hadinger, 2019-12-03:
I experienced slow-downs of the Web interface in two occasions:

- `Sleep 0` when bulb is On, fixed #6980 
- UDP packets not read fast enough, fixed #6832 

@slipx06 Can you check your `Sleep` value and make sure it's not `0`.

Do you have Emulation (Hue or Wemo) turned on?

@ascillato Good lead, it may also be a UDP issue because of absence of NTP server. I need to investigate.

ratsputin, 2019-12-03:
@s-hadinger I just did a bit more digging around and discovered that I was seeing the same crash under v6.6.0.20; however, it was on a different light (LSLight2).

For posterity's sake, here are all of the logs I have with "Fatal exception" messages in them:
```
messages:Dec  1 14:04:44 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
messages:Dec  1 14:37:43 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x40232eda epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000004 depc:0x00000000"}
messages:Dec  1 14:39:40 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
messages:Dec  2 10:17:37 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:3 flag:2 (EXCEPTION) epc1:0x4010133d epc2:0x00000000 epc3:0x00000000 excvaddr:0x400354b0 depc:0x00000000"}
messages:Dec  2 10:44:42 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
messages:Dec  2 11:04:46 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
messages:Dec  2 11:05:51 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
messages:Dec  2 11:19:18 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
messages:Dec  2 11:29:04 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
messages:Dec  2 12:14:21 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
messages:Dec  2 20:08:07 LSLight3-7748 ESP-MQT: LSLight3/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"LSLights","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000","Uptime":"0T07:53:56","StartupUTC":"2019-12-02T18:14:11","Sleep":50,"CfgHolder":4617,"BootCount":39,"SaveCount":268,"SaveAddress":"F6000"}}
messages-20191110:Nov  4 18:36:27 LSLight2-4009 ESP-MQT: LSLight2/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
messages-20191201:Nov 29 18:20:25 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:3 flag:2 (EXCEPTION) epc1:0x40101345 epc2:0x00000000 epc3:0x00000000 excvaddr:0x40034930 depc:0x00000000"}
```
The LSLight2 exception would have been under v6.6.  Here are the related INFO messages for it:
```
messages-20191110:Nov  4 18:36:27 LSLight2-4009 ESP-MQT: LSLight2/tele/INFO1 = {"Module":"AiLight","Version":"6.6.0.20(6bcfcf1-sonoff)","FallbackTopic":"cmnd/LSLight2_fb/","GroupTopic":"sonoffs"}
messages-20191110:Nov  4 18:36:27 LSLight2-4009 ESP-MQT: LSLight2/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"LSLight2-4009","IPAddress":"192.168.1.211"}
messages-20191110:Nov  4 18:36:27 LSLight2-4009 ESP-MQT: LSLight2/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
```
I realize this isn't of a great deal of value without stack dumps but I wanted to be sure I tracked any related information here.

One thing to note, the `exception:3` on Dec 2 10:17:37 on LSLight3 is something different.  I don't know if that might shed some light into what's going on with that lamp?

Here are the related logs around that exception:
```
Dec  2 10:17:37 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO1 = {"Module":"AiLight","Version":"7.1.0.1(8cf2d15-tasmota)","FallbackTopic":"cmnd/LSLight3_fb/","GroupTopic":"LSLights/cmnd/"}
Dec  2 10:17:37 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"LSLight3-7748","IPAddress":"192.168.1.212"}
Dec  2 10:17:37 LSLight3-7748 ESP-MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:3 flag:2 (EXCEPTION) epc1:0x4010133d epc2:0x00000000 epc3:0x00000000 excvaddr:0x400354b0 depc:0x00000000"}
```
As I said, this may be of no value without capturing the stack trace.  If the above info provides any insight right now, I'm happy to provide more information, otherwise, I'll await a stack capture mechanism.

s-hadinger, 2019-12-05:
@ratsputin I submitted PR ##7114. When it's merged, you can try again on your bulb and we'll have a better view of what's going on.

ratsputin, 2019-12-05:
@s-hadinger Thank you very much.  I've subscribed to that PR and will keep an eye on it.

As a side-note, I want to extend my thanks to you and the team for your work on Tasmota.  You're providing an extremely valuable service to the community and should know that it's appreciated.

s-hadinger, 2019-12-07:
@ratsputin The PR was #7134 merged. It should be in the pre-compiled binaries in less than two hours. If you self-compile, please keep a copy of the `firmware.map` file.

Then you can flash the device that is rebooting. After a reboot, please paste results from `Status 12`here.

ratsputin, 2019-12-07:
@s-hadinger Thank you.  I have downloaded the latest pre-compiled binary version of tasmota.bin and flashed it into the problematic light.  To test, I performed a `Status 12` right after the upgrade and received the following:
```
15:34:36 MQT: LSLight3/stat/STATUS12 = {"StatusSTK":{"call_chain":[,"40227c70"]}}
```
I'm assuming this is an indication that I have the right version.  I'll follow up with the same command when it next crashes.

s-hadinger, 2019-12-07:
Yes. The version is working. Can you download the map file too?

s-hadinger, 2019-12-07:
Actually you may have the version with the stuck OTA process. Anyways the stack trace will be good.

ratsputin, 2019-12-08:
Hmm...  It appears the console is truncating the stack dump.  Here's what I got on a Status 12:
```
19:50:15 MQT: LSLight3/stat/STATUS12 = {"StatusSTK":{"call_chain":["40101440","401016cb","402501a4","402501a4","40248b05","40248c00","4000050c","40249687","401016cb","402496b0","4024b5cd","40101702","4024f8cd","401016cb","4026158c","402489c1","402487eb","4026ba3a","4026ba4b","402611b7","4026724f","40000f49","40000f49","40000e19","40001878","4010585d","40105863","4010000d","4026009c","4026004d","40262313","401056a5","401009b4","4010528a","40104af5","401035af","4010518d","401056a5","401035af","40262313","4010528a","4000050c","40103400","40104bd9","401009b4","4000d9c4","4010518d","40105465","4010372e","4010464f","4010528a","4000050c","40103400","402464d8","402464b1","4027e2c6","4000050c","40245fae" ...
```
Ah, I did just notice that console messages also go to syslog.  Here's a bit more from syslog:
```
Dec  7 19:50:16 LSLight3-7748 ESP-MQT: LSLight3/stat/STATUS12 = {"StatusSTK":{"call_chain":["40101440","401016cb","402501a4","402501a4","40248b05","40248c00","4000050c","40249687","401016cb","402496b0","4024b5cd","40101702","4024f8cd","401016cb","4026158c","402489c1","402487eb","4026ba3a","4026ba4b","402611b7","4026724f","40000f49","40000f49","40000e19","40001878","4010585d","40105863","4010000d","4026009c","4026004d","40262313","401056a5","401009b4","4010528a","40104af5","401035af","4010518d","401056a5","401035af","40262313","4010528a","4000050c","40103400","40104bd9","401009b4","4000d9c4","4010518d","40105465","4010372e","4010464f","4010528a","4000050c","40103400","402464d8","402464b1","4027e2c6","400005
```
Ugh...  I don't think this crash is related to what started this whole discussion.  After I flashed the lamp it was stuck on white; I played with the various commands trying to get it into `scheme 4` and it was behaving strangely.  After a bit, it strated acting normally.  Looking at the logs, it was likely this exception that cleared up its behavior.

I'll keep an eye on it and look for the exception that started this.  You can disregard the above if you wish.

ratsputin, 2019-12-08:
@s-hadinger Well, it crashed again.  Under 7.1.2.2(dd6cc32-tasmota) running Scheme 4, it's crashing with the following:
```
20:56:39 MQT: LSLight3/tele/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4000df64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
```
Here's the `status 12` dump:
```
21:42:13 MQT: LSLight3/stat/STATUS12 = {"StatusSTK":{"call_chain":["4025af0b","4025aea0","4025ae47","40259fa4","40259fcd","40257a38","401036de","40253d9d","40248e7d","4025872f","40000f58","402581a2","402618f7","402611b7","4023f9a3","40000f49","40000f49","40000e19","40001878","4010585d","40105863","4010000d","4026009c","4026004d","40262313","40262362","401009b4","401024a4","4026141f","402549bb","40105da7","401009b4","401028d5","4010464c","401024a4","4010573f","4010491a","40248636","40104d8b","4000050c","40103400","40246198","40104853","40103718","401009b4","40103400","401056a5","4010321c","4010528a","4010518d","40245fae","40246845","40245fae","40246178","40282100","40246845","40245ee4","40246944" ...
```
Do I need to get you any more information?

As a side note, under this version, even though it's showing running `scheme 4`, the light is frozen with just white turned on after it crashes.

s-hadinger, 2019-12-08:
Hmmm. It's hard to get a reliable result without the corresponding map file. I recompiled on my side to generate a local map file.

If the my map file matches yours, it seems that the crash happened during a NTP request, maybe due to a Wifi reconnect (that's unclear and this is mostly SDK closed-source code).

Do you see Wifi reconnects? (look for `LinkCount` in `Status 1`).
Is your NTP correctly configured and working?

ratsputin, 2019-12-08:
Interesting.

Well, I searched my syslogs for that particular light and checked all of the LinkCounts and the number never exceeded 1.  Also, the lowest RSSI I observed was 68.  Generally, it's in the 78-80 range.

Given that, though, I do see a lot of disconnect/reconnects with MQTT.  Here's an example from just yesterday:
```
messages-20191208:Dec  7 05:38:23 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 07:55:17 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 08:58:53 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 09:19:27 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 11:31:50 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 11:44:51 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 11:55:53 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 13:29:12 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 13:49:19 LSLight3-7748 ESP-WIF: Connected
messages-20191208:Dec  7 13:49:20 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 13:50:40 LSLight3-7748 ESP-WIF: Connected
messages-20191208:Dec  7 13:50:42 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 15:09:59 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 15:32:23 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 15:32:36 LSLight3-7748 ESP-WIF: Connected
messages-20191208:Dec  7 15:32:37 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 15:33:47 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 15:34:03 LSLight3-7748 ESP-WIF: Connected
messages-20191208:Dec  7 15:34:05 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 15:51:39 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 16:57:13 LSLight3-7748 ESP-WIF: Connected
messages-20191208:Dec  7 16:57:14 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 17:15:17 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 18:47:10 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 19:56:49 LSLight3-7748 ESP-WIF: Connected
messages-20191208:Dec  7 19:56:51 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 20:26:12 LSLight3-7748 ESP-WIF: Connected
messages-20191208:Dec  7 20:26:13 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 20:28:50 LSLight3-7748 ESP-WIF: Connected
messages-20191208:Dec  7 20:28:52 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 20:56:37 LSLight3-7748 ESP-WIF: Connected
messages-20191208:Dec  7 20:56:39 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 21:57:50 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 22:00:07 LSLight3-7748 ESP-MQT: Connected
messages-20191208:Dec  7 23:15:16 LSLight3-7748 ESP-MQT: Connected
```
Here's the same search from an identically-configured light connected to the same access-point with a lower RSSI (about five points) that isn't crashing about 6' away from it:
```
messages-20191208:Dec  7 13:49:00 LSLight2-4009 ESP-WIF: Connected
messages-20191208:Dec  7 13:49:01 LSLight2-4009 ESP-MQT: Connected
messages-20191208:Dec  7 13:50:40 LSLight2-4009 ESP-WIF: Connected
messages-20191208:Dec  7 13:50:42 LSLight2-4009 ESP-MQT: Connected
messages-20191208:Dec  7 21:57:51 LSLight2-4009 ESP-MQT: Connected
messages-20191208:Dec  7 22:00:08 LSLight2-4009 ESP-MQT: Connected
```
I have another light whose RSSI is generally in the 45-50 range (that's not crashing) and here's what I see with it:
```
messages-20191208:Dec  7 13:48:21 LSLight8-7804 ESP-WIF: Connected
messages-20191208:Dec  7 13:48:23 LSLight8-7804 ESP-MQT: Connected
messages-20191208:Dec  7 13:50:40 LSLight8-7804 ESP-WIF: Connected
messages-20191208:Dec  7 13:50:42 LSLight8-7804 ESP-MQT: Connected
messages-20191208:Dec  7 21:57:59 LSLight8-7804 ESP-MQT: Connected
```
Judging from the frequency of the disconnects, it looks as though your thought that it's an NTP-related crash due to the network stack is a reasonable explanation (although I find it odd that not being able to talk to an NTP server causes a crash but that's a totally different conversation).  I will say that all of my lights are configured identically with static IP addresses and all point to the same NTP server so I'm inclined to believe this issue is internal to this lamp.

Now that we have a likely explanation, would you say it would be appropriate to erase the flash and set this lamp up from scratch again to see if the behavior continues?  Or, is there other troubleshooting that would be of value?

s-hadinger, 2019-12-08:
@ratsputin I would like still to get a stack trace with the corresponding map file. I compiled the latest tasmota-DE.bin. Can you please flash your faulty device with this one, and report `Status 12` again. Let's keep `Scheme 4` out of the way for now.

Firmware: https://file.hadinger.fr/q1q4cr
Map file: https://file.hadinger.fr/3pqe9l  (links valid 7 days only)

ratsputin, 2019-12-08:
@s-hadinger Okay, I flashed the lamp with the bin file you provided.  I've switched it to `scheme 0` for the time being.

Unfortunately, I don't speak Deutsch so I'm at a bit of a disadvantage here.  I'm hoping the firmware accepts English commands.

I did try a `status 12` and received the following, which is different from what I've seen in the past:
```
13:58:24 CMD: status 12
13:58:24 MQT: LSLight3/stat/STATUS = {"Status":{"Module":27,"FriendlyName":["LSLight3"],"Topic":"LSLight3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
```
I'm concerned that this image doesn't support the `status 12` command or that I need to do something different, possibly?

arendst, 2019-12-08:
That's ok. Only on an exception restart you'll be greeted with the expected data

s-hadinger, 2019-12-08:
@ratsputin Lol, I thought you wanted a German version, because the issue was opened with German names. But it wasn't you :-o

Indeed `Status 12` is disabled until a crash occurs.

Here is an English tasmota.bin version: https://file.hadinger.fr/is8s5q
Map file: https://file.hadinger.fr/k0ucfc

ratsputin, 2019-12-08:
@s-hadinger That's hilarious!  Nope, I'm a 'murican (from Texas, no less) and only speak English; I'll load up the English version.

Kind regards

ratsputin, 2019-12-09:
@s-hadinger I haven't forgotten about this.  The lamp has been stable for 17 hours so far; I think the longest I've seen it stay up is 36 hours.  I'm going to wait that long then turn `scheme 4` back on to see if that is the source of the stability issue unless you want me to try something different.

On that specific topic, I leave the lamp in `scheme 4` mode 24x7 and just toggle power off/on.  I don't know if the scheme code runs in the background even when the lamp is off or not.

s-hadinger, 2019-12-09:
I would prefer to leave the lamp untouched for a more day. I'm not very familiar with the code path of Scheme 4, I know far better what is happening in Scheme 0.

ratsputin, 2019-12-09:
OK.  I'll leave it as-is, then.

ratsputin, 2019-12-09:
@s-hadinger Okay, so the behavior has changed with this version of the firmware--we're no longer seeing crashes.  Now, what I'm seeing is simply that the lamp disconnects and reconnects, setting itself back to a set of (apparently) saved settings.  Here is one example of what's in my syslog:
```
Dec  9 13:09:17 LSLight3-7748 ESP-MQT: LSLight3/tele/HASS_STATE = {"Version":"7.1.2.3(tasmota)","BuildDateTime":"2019-12-08T21:19:13","Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Module":"AiLight","RestartReason":"Software/System restart","Uptime":"0T22:40:18","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:07","MqttCount":5,"BootCount":78,"SaveCount":385,"IPAddress":"192.168.1.212","RSSI":"68","LoadAvg":19}
Dec  9 13:09:28 LSLight3-7748 ESP-MQT: LSLight3/tele/STATE = {"Time":"2019-12-09T13:09:28","Uptime":"0T22:40:29","UptimeSec":81629,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"OFF","Dimmer":100,"Color":"255,255,255,255","HSBColor":"151,0,100","Channel":[100,100,100,100],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ORION_E","BSSId":"98:DA:C4:F8:DE:C3","Channel":11,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:07"}}
Dec  9 13:14:17 LSLight3-7748 ESP-MQT: LSLight3/tele/HASS_STATE = {"Version":"7.1.2.3(tasmota)","BuildDateTime":"2019-12-08T21:19:13","Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Module":"AiLight","RestartReason":"Software/System restart","Uptime":"0T22:45:18","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:07","MqttCount":5,"BootCount":78,"SaveCount":385,"IPAddress":"192.168.1.212","RSSI":"66","LoadAvg":19}
Dec  9 13:14:28 LSLight3-7748 ESP-MQT: LSLight3/tele/STATE = {"Time":"2019-12-09T13:14:28","Uptime":"0T22:45:29","UptimeSec":81929,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"OFF","Dimmer":100,"Color":"255,255,255,255","HSBColor":"151,0,100","Channel":[100,100,100,100],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ORION_E","BSSId":"98:DA:C4:F8:DE:C3","Channel":11,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:07"}}
Dec  9 13:14:39 hass mosquitto: 1575918879: Socket error on client LSLight3, disconnecting.
Dec  9 13:14:39 LSLight3-7748 ESP-MQT: Attempting connection...
Dec  9 13:14:39 hass mosquitto: 1575918879: New client connected from 192.168.1.212 as LSLight3 (p2, c1, k30, u'DVES_USER').
Dec  9 13:14:39 LSLight3-7748 ESP-MQT: Connected
Dec  9 13:14:39 LSLight3-7748 ESP-MQT: LSLight3/tele/LWT = Online (retained)
Dec  9 13:14:39 LSLight3-7748 ESP-MQT: LSLight3/cmnd/POWER =
Dec  9 13:14:40 LSLight3-7748 ESP-MQT: LSLight3/stat/RESULT = {"Speed":1}
Dec  9 13:14:40 LSLight3-7748 ESP-MQT: LSLight3/stat/RESULT = {"Fade":"ON"}
Dec  9 13:14:40 LSLight3-7748 ESP-MQT: LSLight3/tele/STATE = {"Time":"2019-12-09T13:14:40","Uptime":"0T22:45:41","UptimeSec":81941,"Heap":28,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":19,"MqttCount":6,"POWER":"ON","Dimmer":100,"Color":"255,255,255,255","HSBColor":"151,0,100","Channel":[100,100,100,100],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ORION_E","BSSId":"98:DA:C4:F8:DE:C3","Channel":11,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:07"}}
Dec  9 13:14:40 LSLight3-7748 ESP-MQT: LSLight3/stat/RESULT = {"POWER":"ON","Dimmer":100,"Color":"255,255,255,255","HSBColor":"151,0,100","Channel":[100,100,100,100]}
Dec  9 13:14:40 LSLight3-7748 ESP-MQT: LSLight3/tele/STATE = {"Time":"2019-12-09T13:14:41","Uptime":"0T22:45:42","UptimeSec":81942,"Heap":28,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":21,"MqttCount":6,"POWER":"ON","Dimmer":100,"Color":"255,255,255,255","HSBColor":"151,0,100","Channel":[100,100,100,100],"Scheme":4,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ORION_E","BSSId":"98:DA:C4:F8:DE:C3","Channel":11,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:07"}}
Dec  9 13:14:40 LSLight3-7748 ESP-MQT: LSLight3/stat/RESULT = {"Scheme":4}
Dec  9 13:19:16 LSLight3-7748 ESP-MQT: LSLight3/tele/HASS_STATE = {"Version":"7.1.2.3(tasmota)","BuildDateTime":"2019-12-08T21:19:13","Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Module":"AiLight","RestartReason":"Software/System restart","Uptime":"0T22:50:18","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:07","MqttCount":6,"BootCount":78,"SaveCount":386,"IPAddress":"192.168.1.212","RSSI":"66","LoadAvg":29}
```
I only noticed this because the light kept resetting itself to `scheme 4` and turning itself on without me doing anything.  You'll note that it still says `"RestartReason":"Software/System restart"`, so it's no longer crashing; however, it is not behaving correctly as far as I can tell.

Here's a quick dump of all of the disconnects I've seen in the last 24 hours from all eight of my lights (all AiLight profiles from the same manufacturer).  The clusters you're seeing at 11:30 and 21:50 were due to work I was doing on the network at the time:
```
[root@hass log]# grep ESP-MQT\:\ Connected messages
Dec  8 06:02:44 LSLight3-7748 ESP-MQT: Connected
Dec  8 07:01:53 LSLight3-7748 ESP-MQT: Connected
Dec  8 07:13:37 LSLight3-7748 ESP-MQT: Connected
Dec  8 07:14:11 LSLight3-7748 ESP-MQT: Connected
Dec  8 11:30:02 LSLight4-7931 ESP-MQT: Connected
Dec  8 11:30:02 LSLight1-3927 ESP-MQT: Connected
Dec  8 11:30:04 LSLight7-6389 ESP-MQT: Connected
Dec  8 11:30:06 LSLight5-3490 ESP-MQT: Connected
Dec  8 11:30:08 LSLight2-4009 ESP-MQT: Connected
Dec  8 11:30:09 LSLight3-7748 ESP-MQT: Connected
Dec  8 11:30:09 LSLight6-7848 ESP-MQT: Connected
Dec  8 11:39:59 LSLight8-7804 ESP-MQT: Connected
[... cut ...]
Note that this one light (LSLight3) is the only one that consistently and repeatedly disconnects.  Another piece of information here: I have another light by a different manufacturer that's situated right in the middle of these landscape lights (it's not an AiLight; I'm just using a generic setup for a plain, white lamp) and it's seeing none of this behavior.

As we're I'm longer getting an exception, is there really anything more that can be done to troubleshoot this issue?  I'm happy to continue to work with you to try to get to the bottom of this.

s-hadinger, 2019-12-09:
It's not totally reassuring that the one light that used to crash no longer crashes... I agree that there's not much to be done for now.

ratsputin, 2019-12-09:
That's fair.  If you concur, I'm going to downgrade the firmware in this lamp to 7.1.0.1 (the version the other lamps are running and it was previously running--so I have a consistent baseline) and erase the flash.  If that doesn't fix the issue, I have to assume this is a hardware issue with this particular lamp.

ratsputin, 2019-12-10:
Would you believe that the lamp hasn't had a crash since the version of the firmware you provided?  As mentioned above, it had stopped crashing once I updated to 7.1.2.3.  I have since just downgraded to 7.1.0.1 and it remains stable.  Actually, it's not even disconnecting like what we were seeing with 7.1.2.3.

Note, I _did not_ erase the flash.  The only thing I did was downgrade.

I'll keep a close eye on it; however, it almost looks as though something in 7.1.2.3 addressed whatever was causing earlier versions to crash.

I will follow up in the next day or two with what I see.

s-hadinger, 2019-12-10:
Let's call it 'magic'

Anyways we now have the crash recorder implemented.

itarika2, 2020-04-10:
Hello, I have the same as previous users problem with few differents devices for led managment and not only. Of course, before write here i read all pages like issues, wiki, simillar topics and so on and so forth...

I tried different hardware with different versions of firmware, but **still have devices that constantly reboots with extention 29** in the log. Here you are a few examples:

**AriLux AL-LC06**

Firmware 8.2.0

```00:00:06 MQT: tele/Light1/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["40235070","00000000","40100450"],"EXCVADDR":"00000004","DEPC":"00000000","CallChain":["4023506c","402350f8","4024e5be","40101b06","40252c4d","40101acf","40264948","4024bce5","4024bb0b","4026edf6","4026ee07","40264573","4026a60b","40000f49","40000f49","40000e19","40105891","40105897","4010000d","40263458","40263409","402491e2","40249b19","402491e2","40249118","40249c44","40281695","40249118","40284ae8","40284ae7","40249617"]}}

00:00:06 MQT: tele/Light1/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","40100450"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["401013f0","4025e2c3","4025e258","4025e1ff","4025d360","4025d389","4025adad","4025ada2","40103700","40257159","4025baeb","40000f58","4025b55e","40264cb3","40264573","40000f49","40000f49","40000e19","40105891","40105897","4010000d","40263458","40263409","402491e2","40249b19","402656cf","40100cb0","40258317","40100cb0","40102909","401024d8"]}}```

Firmware 8.1.0

`00:00:06 MQT: tele/Light1/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4025491f","00000000","40101d2c"],"EXCVADDR":"00000004","DEPC":"00000000","CallChain":["40259078","40262693","40261f53","40000f49","40000f49","40000e19","40001878","40105865","4010586b","4010000d","40260e38","40260de9","401009b4","401024ac","4010575f","4010575f","40103043","40103224","401056ad","401009b4","401024ac","40105292","40103043","40104afd","40103224","40105195","401036e6","4010540e","40104657","40104654","401035b7"]}}`

Firmware 7.2.0

`00:00:06 MQT: tele/Light1/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["40234f8e","00000000","40101d2c"],"EXCVADDR":"00000004","DEPC":"00000000","CallChain":["40235018","4024c21e","4010170a","40250851","401016d3","40262510","40249945","4024976f","4026c9ce","4026c9df","4026213b","402681e3","40000f49","40000f49","40000e19","40001878","40105865","4010586b","4010000d","40261020","40260fd1","401009b4","40106005","4000050c","4000da1a","4027f2b5","4000050c","40103408","4000e21c","4000da01","4010575f"]}}`

Firmware 6.7.1

`00:00:06 MQT: tele/Light1/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4025413b epc2:0x00000000 epc3:0x40101ca4 excvaddr:0x00000004 `depc:0x00000000”}

Firmware 7.1.0

`00:00:06 MQT: tele/Light1/INFO3 = {"RestartReason":"Fatal exception:29 flag:2 (EXCEPTION) epc1:0x40232eda epc2:0x00000000 epc3:0x40101d2c excvaddr:0x00000004 depc:0x00000000”}`

**NodeMCU v2** ESP8266 (CP2102) with WS2812

Firmware 8.2.0

```
00:00:06 MQT: tele/Light1A/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["40235070","00000000","00000000"],"EXCVADDR":"00000004","DEPC":"00000000","CallChain":["4023506c","402350f8","4024e5be","40101b06","40252c4d","40101acf","40264948","4024bce5","4024bb0b","4026edf6","4026ee07","40264573","4026a60b","40000f49","40000f49","40000e19","40105891","40105897","40105a78","4010000d","40263458","40263409","402491e2","402491e2","40100cb0","40102909","401024d8","40105773","40100cb0","40104dbf","401024d8"]}}

00:00:06 MQT: tele/Light1A/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["40256f3b","00000000","00000000"],"EXCVADDR":"00000004","DEPC":"00000000","CallChain":["40256f30","4025b694","40264cb3","40264573","40242fe8","40000f49","40000f49","40000e19","40105891","40105897","4010000d","40263458","40263409","402491e2","402491e2","401056d9","40100cb0","40102909","401024d8","4010578b","401024d8","4010306f","40100cb0","401024d8","40101e22","40100cb0","40248ecb","40103416","40103491","40100cb0","4000050c"]}}
```

```
00:00:06 MQT: tele/Light1B/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["40256f3b","00000000","00000000"],"EXCVADDR":"00000004","DEPC":"00000000","CallChain":["40256f30","4025b694","40264cb3","40264573","40242fe8","40000f49","40000f49","40000e19","40105891","40105897","4010000d","40263458","40263409","40100cb0","40102909","401024d8","40105773","4010494e","40104dbf","40104887","4010374c","4000050c","40103434","402491e2","4000e264","4000d4df","40281695","40000000","40249118","402491e2","40100cb0"]}}

00:00:06 MQT: tele/Light1B/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["40256f3b","00000000","00000000"],"EXCVADDR":"00000004","DEPC":"00000000","CallChain":["40256f30","4025b694","40264cb3","40264573","40000f49","40000f49","40000e19","40105891","40105897","4010000d","40263458","40263409","402491e2","402491e2","401056d9","401024d8","4010578b","401024d8","4010306f","40103250","40100cb0","401024d8","4010578b","4010306f","40103250","40103250","40103712","4000050c","40103434","40248ece","402491e2"]}}
```

**NodeMCU v2** ESP8266 (CP2102) with sensors

Firmware 8.2.0

`00:00:06 MQT: tele/AirMonitor/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4023a7d0","00000000","00000000"],"EXCVADDR":"00000004","DEPC":"00000000","CallChain":["4023a7cc","4023a858","4025267a","40101a26","40256d01","401019ef","402689f8","4024fd95","4024fbb7","40272ea6","40272eb7","40268623","4026e6b7","40000f49","40000f49","40000e19","4010580d","40105813","4010000d","40267508","402674b9","40269783","40100bd0","40102454","40105707","40102454","40100bd0","40102454","401056ef","401048ca","40104d3b"]}}`