arendst/Tasmota #5885: PZEM (017 an others) Voltage accuracy

gigi9000, 2019-05-30:
I have a PZEM DC (017) allo is ok with the last TASMOTA version (6.5).
The amperage is fairly accurate but the voltage lacks precision (rounded to the integer). however, the PZEM DCs have a measurement accuracy of 0.01 v. Is it possible to have better accuracy for the voltage please.
Thanks

arendst, 2019-05-30:
See wiki: Voltres

ascillato2, 2019-05-30:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-30 by ascillato2

gigi9000, 2019-05-30:
I 'm debuting  command via web is like this but after i'm not sure
http:// xxxx/cm?cmnd=VoltRes